zephyrproject-rtos/zephyr #13340: NRF52 pca10040 boards open the "Flash hardware support" option, the BT Mesh example does not work properly

xiaoliang314, 2019-02-13:
**Describe the bug**
NRF52 pca10040 boards open the "Flash hardware support" option, the BT Mesh example does not work properly. The UART console does not see any output, return to normal after turning off this option.

nvlsianpu, 2019-02-15:
@xiaoliang314 
At last samples/bluetooth/mesh/ works well at my desk: I was able to saw loot of loggs on the console.
Please point what example it was (pleas check before with current master). 

> "Flash hardware support" 

what do you mean?

xiaoliang314, 2019-02-16:
@nvlsianpu This problem occurred when my project was upgraded from HEAD(2caecb5) to the latest version. My chip may still keep the old version of the flash record. When I use the example (samples/bluetooth/mesh) to download to the chip, it doesn't work. This example works when I closed "drivers/flash hardware support" in menuconfig and erased all flash.

nvlsianpu, 2019-02-18:
Can you full erase your chip and test it with newest master?

Backward compatibility is not guaranteed for both `settings` and `fcb` which might cause problem with old records you have.
Disabling `drivers/flash hardware support` disables `settings` module as well - that why the example started working.

So It is possible that after changes your storage content is incompatible with newest code-base. This might be either fcb or settings.

xiaoliang314, 2019-02-20:
@nvlsianpu thanks!

closed 2019-02-20 by xiaoliang314