zephyrproject-rtos/zephyr #13347: tests/drivers/watchdog/wdt_basic_api fails on test_wdt_callback_1() for Quark SE / arc

cinlyooi-intel, 2019-02-13:
**Describe the bug**
`test_wdt: test_wdt_callback_1() == TC_PASS is false`

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=quark_se_c1000_devboard -DARCH=x86
3. ninja
4. ninja run

**Screenshots or console output**
```
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
 Testcase: test_wdt_callback_1
 Assertion failed at /home/cinlyooi/tests/zephyr.git/tests/drivers/watchdog/wdt_basic_api/src/test_wdt.c:308: test_wdt: test_wdt_callback_1() == TC_PASS is false

 FAIL - test_wdt
===================================================================

```
**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr
 - Commit SHA or Version used: 9072d34f306078acdc9d950c9beae5cc0738ea19

**Additional context**

Please note it is the same as top case of #10623

SavinayDharmappa, 2019-03-04:
sensor subsystem is not getting reset , because of which interrupts are not cleared hence we are observing assertion for test_wdt_callback_1.

SavinayDharmappa, 2019-03-05:
after further analysis it is clear that the wdt_qmsi driver present currently in zephyr only for x86 part of quark_se_c1000. In case of  ARC part of quark_se_c1000 time 0 / 1 is used for  watchdog reset generation. Based on this analysis i would be bypassing the test case for arc part of quark_se_c1000 @nashif  @ManojSubbarao

SavinayDharmappa, 2019-03-06:
#14067

rljordan, 2019-03-25:
@SavinayDharmappa  Any update on this?

SavinayDharmappa, 2019-03-25:
@rljordan  I have already raised PR related to this issue #14067

closed 2019-08-19 by nashif