zephyrproject-rtos/zephyr #13353: z_timeout_remaining should substract z_clock_elapsed

alenxyz, 2019-02-13:
**Describe the bug**
Applications that calls z_timeout_remaining would be interested in the actual remaining time at that moment. Right now the return value does not consider elapsed time, which means the timer will expire sooner than the returned value. Maybe the current behaviour is indented but at least for me it was unexpected and not documented.

**Expected behavior**
z_timeout_remaining returns the actual remaining time

**Impact**
Workaround: substract elapsed time in the application.

**Environment (please complete the following information):**
Custom toolchain with Zephyr 1.10 due to C99 only
1.14 seems to show same behaviour from code review

**Additional context**

andyross, 2019-02-13:
Yeah, that's a bug.    Horrifyingly, we don't test this API **at all**, anywhere.  It's actually dead code.

alenxyz, 2019-02-21:
Thanks Andy for confirmation. Btw, I also noticed when the head timer is removed before expiry, the HW timer can be set to a later time, or cleared when it was the only timer. It does not seem to cause any trouble though.

PR #13994 (2019-03-01): kernel/timeout: fix k_timer_remaining_get() when tickless
In some circumstances (e.g., a tickless kernel), k_timer_remaining_get()
would not account for time passed that didn't involve clock interrupts.
This adds a simple fix for that, and adds a test case.  In addition, the
return value of k_timer_remaining_get() is clamped at 0 in the case of
overdue timers and the API description is adjusted to reflect this.

Fixes: #13353
commit 0ad4022e (2019-03-01): kernel/timeout: fix k_timer_remaining_get() when tickless

In some circumstances (e.g., a tickless kernel), k_timer_remaining_get()
would not account for time passed that didn't involve clock interrupts.
This adds a simple fix for that, and adds a test case.  In addition, the
return value of k_timer_remaining_get() is clamped at 0 in the case of
overdue timers and the API description is adjusted to reflect this.

Fixes: #13353
closed 2019-03-01 by andrewboie