zephyrproject-rtos/zephyr #13361: nucleo_f103rb blinky example cannot run

zl3one6, 2019-02-13:
**Describe the bug**
Following the document, build and flash blinky example into ST Nucleo-F103RB development board, cannot watch LED working. Jumpers on the board are not changed. ST official demos downloaded by Keil MDK are working correctly. Debugger is ST-Link.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nucleo_f103rb ..
3. ninja
    ninja flash
4. See error

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04, Windows 10
 - Zephyr SDK on Ubuntu, GCC arm embedded on Windows
 - Zephyr 1.13.0 release

gon1332, 2019-02-13:
With a quick look on Nucleo's .dts the onboard LED is aliased to LED0 as needed by the blinky sample.

I do not have this Nucleo. So, for starters, could you please check the returning value of `device_get_binding` against `NULL` in the blinky code?

erwango, 2019-02-13:
It is strange as it is working on my side, on top of master.
Can you check it is working on recent zephyr?

zl3one6, 2019-02-14:
In version 1.13.0, the nucleo_f103rb.dts contains the following LED port definition:

	leds {
		compatible = "gpio-leds";
		green_led_2: led@0 {
			gpios = <&gpioa 5 GPIO_INT_ACTIVE_HIGH>;
			label = "User LD2";
		};
	};

It seems that the program does not hit the main function

erwango, 2019-02-14:
So I tested back on 1.13, and it also works on my side.
What is the version of your nucleo? Mine is MB1136 rev c. (I don't  think this can be the issue anyway, but ...)
You mention you tested on windows and Ubuntu and you have same issue on both environments?
Have you tested hello_world sample?
Finally, can you try to flash by .bin drag and drop on drive mounted when plugging?

zl3one6, 2019-02-19:
It seems the clock configuration issue. It works now, thanks

closed 2019-02-19 by zl3one6

grybouilli, 2025-05-12:
For any further reference, here's how I got it all working:
in the file `${zephyrproject_directory}/zephyr/boards/st/nucleo_f103rb/nucleo_f103rb.dts`, i put the following clock configuration:

```dts
&clk_lsi {
    status = "okay";
};

&clk_hsi {
    clock-frequency = <DT_FREQ_M(8)>;
    status = "okay";
};

&pll {
    mul = <2>;
    clocks = <&clk_hsi>;
    status = "okay";
};

&rcc {
    clocks = <&pll>;
    clock-frequency = <DT_FREQ_M(8)>;
    ahb-prescaler = <1>;
    apb1-prescaler = <1>;
    apb2-prescaler = <1>;
    adc-prescaler = <2>;
};
```