zephyrproject-rtos/zephyr #13378: Missing Documentation for Networking subsystem

laperie, 2019-02-13:
This is part of #12955 and in a way another incarnation of #2210 
In the newly organized API reference documentation we need to put content in the networking page (https://docs.zephyrproject.org/latest/reference/networking/index.html) 
Per joint agreement, the approximate structure of the high-level topic is as follows: 

- Overview
- Architecture/composition
- Breakdown of a simple example (commented code snippets) 
- List of the key examples 
- References to important how-to's and BKMS
- API structure (pointers to the next level  doc)

ulfalizer, 2019-02-14:
Might be a good idea to have documentation in plain English too. Auto-generated API docs seldom put stuff in the order that they make the most sense.

jukkar, 2019-02-19:
Sent couple of PRs that update the missing documentation. More PRs will follow.

#13505 doc: net: Add more info for network connectivity with host
#13507 doc: net: Enhance DHCPv4 documentation
#13509 doc: net: Enhance DNS resolver documentation
#13516 doc: net: Enhance network configuration library documentation
#13517 doc: net: context: Add comment to use BSD socket API instead
#13518 Network core helpers documentation fixes

rveerama1, 2019-02-20:
https://github.com/zephyrproject-rtos/zephyr/pull/13554 Add high-level documentation for CoAP library.

carlescufi, 2019-03-07:
@jukkar can this be closed now?

jukkar, 2019-03-07:
> @jukkar can this be closed now?

One issue in my list is still pending (#13714). Also net-pkt documentation needs to be done and we can track that in this PR if needed. So lets keep this open for time being.

jukkar, 2019-03-20:
Closing this as the last remaining network document update for net-pkt is in #14613 and is tracked via #13114

closed 2019-03-20 by jukkar