zephyrproject-rtos/zephyr #13387: CI should enforce that Kconfig entries do not accidentally pollute the toplevel menu

SebastianBoe, 2019-02-14:
**Is your feature request related to a problem? Please describe.**
We are continuously accidentally polluting the toplevel Kconfig menu with entries that do long belong in this menu.

**Describe the solution you'd like**
CI should enforce that there are at most X entries in the top level menu. Increasing the number of entries would then not be possible to do by accident, as it would require tweaking a constant in a test somewhere.

**Additional context**
[image]

ulfalizer, 2019-02-14:
@SebastianBoe 
Could you look into who added the entries you don't think belong there, and talk to them? It's easy to forgot to check the top-level menu, and it might not be obvious that things are added there.

I think that'd be the most efficient solution in this case. A CI check could be added as well.

Thanks.

SebastianBoe, 2019-02-14:
I did for some of the entries, but was not able to figure it out for all. Anyway, this issue is not about particulars.

ulfalizer, 2019-02-14:
Simple to make a CI check for at least. I'll add one tomorrow.

ulfalizer, 2019-02-15:
@SebastianBoe Here's a PR: https://github.com/zephyrproject-rtos/ci-tools/pull/16

Couldn't add you as a reviewer for some reason.

nashif, 2019-02-21:
merged now

closed 2019-02-21 by nashif