zephyrproject-rtos/zephyr #13388: settings: fcb-backend: how to assign storage partition properly

nvlsianpu, 2019-02-14:
**The problem we have **
Currently it is possible via kconfig to assign any partition for FCB using its number.
Partitions flash_area_id becomes non predefined (are auto-generated since #13081).
So it is possible only to ques which number will be signed to certain area.

Unfortunately it is not possible to transfer FLASH_AREA_XXX_ID label via Kconfig. It is possible to make some run-time binding operation - but this seams like using a tank to kill an ant.

**solution**
Let assign settings to storage partition and remove SETTINGS_FCB_FLASH_AREA property from settings Kconfig. Optionally we can decide for use a dedicated partition name.

In mine opinion each flash memory user module should use dedicated partition. Optionally some partition might group into one biger partition (I think it is not forbidden to declare such flash_map - but this is independent topic, not important for this one)

nvlsianpu, 2019-02-14:
@galak @anangl @carlescufi @pdunaj @Laczen @cvinayak : your opinions.

Laczen, 2019-02-14:
My honest opinion: **try to remove flash_map**.

Flash_map serves two purposes:
1. Getting a unique mapping between id and location in flash.
2. Protecting against writing/reading the wrong location given 1.

If 1. is not possible anymore, because of how it is generated, 2. has no meaning at all.

But if you really need to keep flash_map (think about how you will handle multiple images), it should be done at runtime.

nvlsianpu, 2019-02-14:
@Laczen 
If you wana't to operate on more than one device then you need always to know which device should you use to operate on certain flash-area. So you need a way to describe relations. So flash map is an answer to that. Otherwise you still need something else.

Flash_map is also generic component - otherwise we need dedicated solution for any user module.

There is also PR pending which introduce API for get flash device form flash_map - so it will be possible to use flash map as only information source (https://github.com/zephyrproject-rtos/zephyr/pull/13303).

Predefined ID was lost - but it was not needed - we still can use area we want because each area has very deterministic area ID (like `DT_FLASH_AREA_IMAGE_SCRATCH_ID`. We can provide a way to assign ID number manually (using DTS) at some point - but this is not needed right now, and I believe not very soon (that's why I didn't protest against).

I believe you maybe have got better idea than flash_map is. We can discuss such a idea in other topic.

Laczen, 2019-02-14:
> @Laczen
> If you wana't to operate on more than one device then you need always to know which device should you use to operate on certain flash-area. So you need a way to describe relations. So flash map is an answer to that. Otherwise you still need something else.
>

No you don't, this information is already available as you are using it to define the default flash_map. If each dts entry generates the device name there is no need to put it in a array for looking it up.
 
> Flash_map is also generic component - otherwise we need dedicated solution for any user module.
> 
> There is also PR pending which introduce API for get flash device form flash_map - so it will be possible to use flash map as only information source (#13303).
>
This is definitively a bad idea, lets force everyone into using a unnecessary intermediate level.

> Predefined ID was lost - but it was not needed - we still can use area we want because each area has very deterministic area ID (like `DT_FLASH_AREA_IMAGE_SCRATCH_ID`. We can provide a way to assign [...]

Flash_map only makes sense if this is generated at the same time as the bootloader, the dts would not need any definition of regions whatsoever, the flash_map routines would read this from flash. A change in how your flash_map would look like would be a upgrade of the flash_map definition in flash.

nvlsianpu, 2019-02-14:
cc @lemrey

nvlsianpu, 2019-02-19:
@carlescufi, @pdunaj Any thought?

nvlsianpu, 2019-02-19:
cc @SebastianBoe

lemrey, 2019-02-20:
I support the idea of removing `SETTINGS_FCB_FLASH_AREA` from the options. The point is, Kconfig is able to set it correctly and automatically; the user doesn't have a simple way of doing that.

carlescufi, 2019-03-01:
Like @lemrey, I also Agree with the proposed solution as a first step. @nvlsianpu could you propose a PR?

Laczen, 2019-03-01:
@nvlsianpu, @carlescufi, yesterday I created a PR #13954 that generates the DEV name for each of the parts in the dts definition. This would allow you to create a default_flash_map easily without using the area_id. The definitions that are generated are:
`DT_FLASH_AREA_IMAGE_0_DEV="NRF_FLASH_DRV_NAME"`
Using these definitions you can generate a default flash_map that has known definitions (linking the id again to the definition). This also allows the removal of flash_map completely but this is something else.

PR #14031 (2019-03-04): settings: always uses the storage partition for FCB
It was possible via Kconfig to assign any partition for FCB using
its number. Partitions flash_area_id becomes non predefined
(are auto-generated). So it is possible only to guess which
number will be signed to certain area.

Unfortunately it is not possible to transfer FLASH_AREA_XXX_ID
label via Kconfig.

Patch assigns settings to the storage partition and remove
SETTINGS_FCB_FLASH_AREA property from settings Kconfig.

fixes #13388

I Added myself as settings tests codeowner - I'm already settings codeowner, so this makes sense.
commit 15159ddd (2019-03-04): settings: always uses the storage partition for FCB

It was possible via Kconfig to assign any partition for FCB using
its number. Partitions flash_area_id becomes non predefined
(are auto-generated). So it is possible only to guess which
number will be signed to certain area.

Unfortunately it is not possible to transfer FLASH_AREA_XXX_ID
label via Kconfig.

Patch assigns settings to the storage partition and remove
SETTINGS_FCB_FLASH_AREA property from settings Kconfig.

fixes #13388
closed 2019-03-04 by carlescufi