zephyrproject-rtos/zephyr #13389: bluetooth: controller: Check on done queue performed too late

mped-oticon, 2019-02-14:
`ull_event_done` checks `MFIFO_DEQUEUE(done)` for emptiness.
(I'd say it is unlikely that the done queue is empty, but still, the check exists.)

If empty, `ull_event_done` returns NULL. Fine.
I noticed that currently, Nordic's and Oticon's LLL throw away the return code of `ull_event_done`.
However, it makes little sense for the callers to check this return code -- the damage has already been done: We have completed an event, yet ULL can not process it because we can not obtain a done-event. This could lead to LSTO in case of conn upd.

Proposed fix: At prepare time we have to check that `mfifo_done` is not empty. If empty, close the event.

carlescufi, 2019-03-19:
@mped-oticon in order not to affect the current release, and give that this is on the non-default split arch, turning this into enhancement.

joerchan, 2019-06-04:
Got assert bd1e8fbf826f3e45f98c7c39edd044fad8c965bb (PR: 16499)
```west build -b nrf52840_pca10056 samples/bluetooth/central_hr -DCONFIG_BT_LL_SW_SPLIT=y -DCONFIG_BT_DATA_LEN_UPDATE=y```

lll_conn.c: 637:
```
	e = ull_event_done_extra_get();
->	e->type = EVENT_DONE_EXTRA_TYPE_CONN;
	e->trx_cnt = trx_cnt;
	e->crc_valid = crc_valid;
```

```
>>> p/x mfifo_done
$3 = {
  s = 0x4, 
  n = 0x6, 
  f = 0x1, 
  l = 0x1, 
  m = {[0x0] = 0xa8, [0x1] = 0x8, [0x2] = 0x0, [0x3] = 0x20, [0x4] = 0xa8, [0x5] = 0x8, [0x6] = 0x0, [0x7] = 0x20, [0x8] = 0xe0, [0x9] = 0x8, [0xa] = 0x0, [0xb] = 0x20, [0xc] = 0x18, [0xd] = 0x9, [0xe] = 0x0, [0xf] = 0x20, [0x10] = 0x50, [0x11] = 0x9, [0x12] = 0x0, [0x13] = 0x20, [0x14] = 0x88, [0x15] = 0x9, [0x16] = 0x0, [0x17] = 0x20}
}
>>> frame 4
#4  0x00011732 in isr_done (param=0x20000fa0 <conn_pool+28>) at /home/joakim/repos/zephyrproject/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c:637
637		e->type = EVENT_DONE_EXTRA_TYPE_CONN;
>>> p e
$4 = (struct event_done_extra *) 0x0 <z_finalize_fd>
```

Happened just after connection:
```
[DEVICE]: d4:7e:61:d1:1b:ed (random), AD evt type 0, AD data len 24, RSSI -83
[AD]: 1 data_len 1
[AD]: 9 data_len 19
[DEVICE]: d9:ce:62:67:35:92 (random), AD evt type 0, AD data len 11, RSSI -65
[AD]: 1 data_len 1
[AD]: 3 data_len 6
Connected: d9:ce:62:67:35:92 (random)
***** MPU FAULT *****
  Data Access Violation
  MMFAR Address: 0x0
***** Hardware exception *****
Current thread ID = 0x20001694
Faulting instruction address = 0x11732
Fatal fault in ISR! Spinning...
```

carlescufi, 2021-12-22:
@mtpr-ot Can you please take a look at this?

mtpr-ot, 2021-12-22:
> 	e = ull_event_done_extra_get();
> 	e->type = EVENT_DONE_EXTRA_TYPE_CONN;

This code was changed a long time ago to assert if unable to obtain a done event for the MFIFO. In cases where that happens there is nothing else to do than assert (as now implemented). Without a done-node, the prepare/done "machine" grinds to a halt, and there's no way to handle that.
If a problem like the this happens it's either due to some system getting stuck, or more likely, the allocation of done elements was insufficient.

mtpr-ot, 2021-12-22:
I disagree with the proposed fix, as we cannot just "close the event" without a done-node to clean up states. I think we should close this enhancement.

cvinayak, 2021-12-22:
Closing based on current implementation state.

closed 2021-12-22 by cvinayak