zephyrproject-rtos/zephyr #13396: Cannot connect to Galaxy S8 via BLE

pdunaj, 2019-02-14:
**Describe the bug**
Start device with settings erased. Device advertises. Try connecting to GalaxyS8. Connection fails with reason 42 (`BT_HCI_ERR_DIFF_TRANS_COLLISION`).

Connection can be established using other devices: Windows 7 (Intel driver), Windows 10, Moto G5 (Android).

Zephyr was able to connect to this device not so long ago.

**To Reproduce**
Simply try to connect.

**Expected behavior**
Device should connect.

**Impact**
Showstopper. Galaxy S8 is a high end phone. If Zephyr cannot connect to it it's bad.

**Screenshots or console output**
```
[00320098] <inf> battery: Battery level: 100% (4223 mV)
[00320099] <inf> event_manager: e: battery_level_event level=100
[00608001] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem taken
[00608001] <dbg> bt_ctlr_hci_driver.prio_recv_thread: RX node enqueue
[00608002] <dbg> bt_ctlr_hci_driver.prio_recv_thread: RX node enqueue
[00608003] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem take...
[00608003] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem taken
[00608003] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem take...
[00608004] <dbg> bt_ctlr_hci_driver.recv_thread: unblocked
[00608005] <dbg> bt_ctlr_hci_driver.recv_thread: Packet in: type:1 len:21
[00608008] <dbg> bt_ctlr_hci_driver.hci_driver_send: enter
[00608008] <dbg> bt_ctlr_hci_driver.cmd_handle: Replying with event of 6 bytes
[00608010] <dbg> bt_ctlr_hci_driver.hci_driver_send: exit: 0
[00608010] <dbg> bt_conn.bt_conn_prepare_events:
[00608012] <dbg> bt_conn.bt_conn_set_state: disconnected -> connected
[00608012] <dbg> bt_conn.bt_conn_ref: handle 0 ref 2
[00608017] <inf> ble_state: Connected to 56:b7:18:80:8c:f1 (random)
[00608018] <inf> ble_state: Set security level
[00608021] <dbg> bt_conn.bt_conn_prepare_events:
[00608021] <dbg> bt_conn.bt_conn_prepare_events: Adding conn 0x2000251c to poll list
[00608024] <inf> event_manager: e: ble_peer_event id=0x2000251c CONNECTED
[00608025] <inf> ble_adv: Advertising stopped
[00608025] <inf> hid_state: Subscriber 0x2000251c connected
[00608025] <inf> hid_state: Active subscriber 0x2000251c
[00608030] <inf> event_manager: e: led_event led_id:0 mode:ON period:0 color: < 250 250 250 >
[00608035] <inf> event_manager: e: led_event led_id:1 mode:ON period:0 color: < 200 0 15 >
[00608055] <dbg> bt_conn.bt_conn_send_cb: conn handle 0 buf len 6 cb 0x00000000
[00608057] <dbg> bt_conn.bt_conn_unref: handle 0 ref 1
[00608058] <dbg> bt_ctlr_hci_driver.hci_driver_send: enter
[00608059] <dbg> bt_ctlr_hci_driver.cmd_handle: Replying with event of 6 bytes
[00608060] <dbg> bt_ctlr_hci_driver.hci_driver_send: exit: 0
[00608060] <dbg> bt_conn.bt_conn_process_tx: conn 0x2000251c
[00608060] <dbg> bt_conn.send_buf: conn 0x2000251c buf 0x20007154 len 6
[00608061] <dbg> bt_conn.send_frag: conn 0x2000251c buf 0x20007154 len 6 flags 0x00
[00608061] <dbg> bt_conn.bt_conn_notify_tx: conn 0x2000251c
[00608061] <dbg> bt_conn.add_pending_tx: conn 0x2000251c cb 0x00000000
[00608061] <dbg> bt_ctlr_hci_driver.hci_driver_send: enter
[00608062] <dbg> bt_ctlr_hci_driver.hci_driver_send: exit: 0
[00608062] <dbg> bt_conn.bt_conn_prepare_events:
[00608063] <dbg> bt_conn.bt_conn_prepare_events: Adding conn 0x2000251c to poll list
[00608064] <dbg> bt_ctlr_hci_driver.recv_thread: blocking
[00608065] <dbg> bt_ctlr_hci_driver.recv_thread: unblocked
[00608065] <wrn> bt_ctlr_hci: handle: 0x0000, CSA: 0.
[00608066] <dbg> bt_ctlr_hci_driver.recv_thread: blocking
[00610489] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem taken
[00610490] <dbg> bt_ctlr_hci_driver.prio_recv_thread: Num Complete: 0x0000:1
[00610490] <dbg> bt_conn.bt_conn_ref: handle 0 ref 2
[00610491] <dbg> bt_conn.bt_conn_unref: handle 0 ref 1
[00610492] <dbg> bt_ctlr_hci_driver.prio_recv_thread: RX node enqueue
[00610492] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem take...
[00610493] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem taken
[00610493] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem take...
[00610494] <dbg> bt_conn.bt_conn_notify_tx: conn 0x2000251c
[00610494] <dbg> bt_conn.bt_conn_prepare_events:
[00610496] <dbg> bt_conn.bt_conn_prepare_events: Adding conn 0x2000251c to poll list
[00610497] <dbg> bt_ctlr_hci_driver.recv_thread: unblocked
[00610498] <dbg> bt_ctlr_hci_driver.recv_thread: Packet in: type:1 len:14
[00610498] <dbg> bt_conn.bt_conn_ref: handle 0 ref 2
[00610499] <dbg> bt_conn.bt_conn_unref: handle 0 ref 1
[00610501] <dbg> bt_ctlr_hci_driver.hci_driver_send: enter
[00610501] <dbg> bt_ctlr_hci_driver.cmd_handle: Replying with event of 6 bytes
[00610502] <dbg> bt_ctlr_hci_driver.hci_driver_send: exit: 0
[00610502] <dbg> bt_conn.bt_conn_prepare_events:
[00610503] <dbg> bt_conn.bt_conn_prepare_events: Adding conn 0x2000251c to poll list
[00610504] <dbg> bt_ctlr_hci_driver.recv_thread: blocking
[00612072] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem taken
[00612072] <dbg> bt_ctlr_hci_driver.prio_recv_thread: RX node enqueue
[00612073] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem take...
[00612074] <dbg> bt_ctlr_hci_driver.recv_thread: unblocked
[00612075] <dbg> bt_ctlr_hci_driver.recv_thread: Packet in: type:3 len:15
[00612075] <dbg> bt_conn.bt_conn_ref: handle 0 ref 2
[00612075] <dbg> bt_conn.bt_conn_recv: handle 0 len 11 flags 02
[00612075] <dbg> bt_conn.bt_conn_recv: First, len 11 final 7
[00612075] <dbg> bt_conn.bt_conn_recv: rx_len 0
[00612076] <dbg> bt_conn.bt_conn_recv: Successfully parsed 11 byte L2CAP packet
[00612077] <dbg> bt_conn.bt_conn_send_cb: conn handle 0 buf len 11 cb 0x00000000
[00612080] <dbg> bt_conn.bt_conn_unref: handle 0 ref 1
[00612081] <dbg> bt_conn.bt_conn_process_tx: conn 0x2000251c
[00612081] <dbg> bt_conn.send_buf: conn 0x2000251c buf 0x20007154 len 11
[00612082] <dbg> bt_conn.send_frag: conn 0x2000251c buf 0x20007154 len 11 flags 0x00
[00612082] <dbg> bt_conn.bt_conn_notify_tx: conn 0x2000251c
[00612082] <dbg> bt_conn.add_pending_tx: conn 0x2000251c cb 0x00000000
[00612082] <dbg> bt_ctlr_hci_driver.hci_driver_send: enter
[00612083] <dbg> bt_ctlr_hci_driver.hci_driver_send: exit: 0
[00612084] <dbg> bt_conn.bt_conn_prepare_events:
[00612084] <dbg> bt_conn.bt_conn_prepare_events: Adding conn 0x2000251c to poll list
[00612085] <dbg> bt_ctlr_hci_driver.recv_thread: blocking
[00615231] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem taken
[00615232] <dbg> bt_ctlr_hci_driver.prio_recv_thread: Num Complete: 0x0000:1
[00615233] <dbg> bt_conn.bt_conn_ref: handle 0 ref 2
[00615234] <dbg> bt_conn.bt_conn_unref: handle 0 ref 1
[00615234] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem take...
[00615235] <dbg> bt_conn.bt_conn_notify_tx: conn 0x2000251c
[00615236] <dbg> bt_conn.bt_conn_prepare_events:
[00615236] <dbg> bt_conn.bt_conn_prepare_events: Adding conn 0x2000251c to poll list
[00615239] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem taken
[00615240] <dbg> bt_ctlr_hci_driver.prio_recv_thread: RX node enqueue
[00615241] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem take...
[00615242] <dbg> bt_ctlr_hci_driver.recv_thread: unblocked
[00615242] <dbg> bt_ctlr_hci_driver.recv_thread: Packet in: type:1 len:6
[00615243] <dbg> bt_conn.bt_conn_ref: handle 0 ref 2
[00615243] <dbg> bt_conn.bt_conn_set_state: connected -> disconnected
[00615248] <inf> ble_state: Disconnected from 56:b7:18:80:8c:f1 (random) (reason 42)
```

**Environment (please complete the following information):**
replicated on PCA20041
Zephyr ncs: a9d9dc78760a7751576987401fde7263eaa6c094 (Zephyr: aeff0ee4ac)
nrf: 322e76fe11cffefaf4df2aece1982f104e549ee4

**Additional context**
N/A

pdunaj, 2019-02-14:
Error is thrown by `conn_update` `subsys/bluetooth/controller/ll_sw/ctrl.c:9547`

pdunaj, 2019-02-14:
Hi @cvinayak , do you have any guess what may be wrong?

cvinayak, 2019-02-26:
@pdunaj Last I knew that S8 does not correctly handle slave initiated procedure collisions correctly, it accepts a PHY update request from peripheral and performs PHY update procedure while S8 itself is initiating a connection update procedure, Zephyr will detect such violations and as per spec disconnect. The work around is to disable auto PHY update initiation in Zephyr (Kconfig option, present in Bluetooth section). 

@sjanc please confirm that the work around is accepted by customers using Zephyr in products.

pdunaj, 2019-02-27:
Hi @cvinayak , you are right and disabling `BT_AUTO_PHY_UPDATE` made S8 connectable. What are the disadvantages of having `BT_AUTO_PHY_UPDATE` disabled?

pdunaj, 2019-02-27:
@cvinayak , is this needed to trigger switch to 2Mb PHY?

pdunaj, 2019-03-01:
workaround works so I am closing this issue

closed 2019-03-01 by pdunaj

dhananjaygj, 2020-11-09:
Recently we are also facing this issue in a lot of latest Android devices and the workaround proposed in this ticket fixes the issue. Probably related to one Bluetooth chip vendor for these latest phones. 
What are the implications of disabling this BT_AUTO_PHY_UPDATE option? 
@cvinayak @joerchan

joerchan, 2020-11-09:
> What are the implications of disabling this BT_AUTO_PHY_UPDATE option?

The zephyr side will not initiate changing the PHY to 2M automatically.