zephyrproject-rtos/zephyr #13397: Function documentation is missing for BSD sockets

jukkar, 2019-02-14:
The BSD socket API function documentation is missing in include/net/socket.h

pfalcon, 2019-02-15:
That's true but BSD Socket API is intended to implement standard POSIX API, as described in http://pubs.opengroup.org/onlinepubs/9699919799/ . Beyond that, it's unclear where exactly (as belonging to  which part) it should be described. I first would like sockets API to be integrated with POSIX subsystems, then we can decide where docs belong.

Thus, changing this to "enhancement".

jukkar, 2019-02-15:
We have a section for BSD sockets in the documentation and currently the zsock_ xxx() API functions are empty. This is definitely a bug and not enhancement so putting the proper label back.

pfalcon, 2019-02-18:
https://github.com/zephyrproject-rtos/zephyr/pull/13495 at least updates the intro section.

nashif, 2019-02-18:
The API documentation can refer to whatever posix standard we are implementing, right now it is empty and it is not clear how those should be used:

https://docs.zephyrproject.org/latest/reference/networking/sockets.html

laperie, 2019-03-14:
@pfalcon -- Any updates?

jukkar, 2019-03-26:
@pfalcon are you going to work on this as we are running out of time soon?

pfalcon, 2019-03-26:
> @pfalcon are you going to work on this as we are running out of time soon?

Any work requires planning to be done beforehand, and the plan for 1.14 was to **implement as much as possible of BSD sockets**. If you think about it, it makes perfect sense: the documentation for BSD sockets is already written (as part of the POSIX standard, linked above). So, if we implement BSD sockets rightfully and faithfully, we can just use existing documentation, whereas otherwise, we need to write "documentation" along the lines of ... what exactly? "This function was intended to work like posix xxxx() function, but we didn't put enough time into implementing it, instead we decided to put time into documenting the fact that we didn't put enough time into implementing it." Something like that?

Good news is that, once we spent as much as possible effort into implementing that API into 1.14 (and there're still patches unmerged, and I have more in queue if those get merged), we can add documentation in 1.14.1. 1.14.2, or whatever. We can do that, after we decide what and how we'll document. Example of decision points to make:

1. Do we document getaddrinfo() or zsock_getaddrinfo()?
2. getaddrinfo() is a POSIX function, and POSIX requires it to be declared in `<netdb.h>`, that's part of tha API. But we don't have `<netdb.h>`. So again, what do we document? "This function is supposed to be in `<netdb.h>`, but so far we didn't do enough homework on that, and hereby we documented this fact of us not having done the mentioned homework." I'm sure we can do better than that, by deciding on the `<netdb.h>` and all other header matters first (https://github.com/zephyrproject-rtos/zephyr/issues/13787, etc.)

If you think that now is the right time to decide on the points above, let's do it now, but I thought now is the time to do the release testing??

jukkar, 2019-03-26:
Just see https://docs.zephyrproject.org/latest/reference/networking/sockets.html and it is missing documentation for `ZSOCK_xxx` macros, various other standard defines and `zsock_xxx()` functions. Please spend couple of hours to add the missing stuff there.

pfalcon, 2019-03-26:
I'll try to, but it's likely will be wasted effort (or more specifically, an effort which will likely need to be largely redone later, effectively doing the same thing twice).

jukkar, 2019-03-26:
> an effort which will likely need to be largely redone later, effectively doing the same thing twice

Why would we do this twice? If the missing API docs are written, why would we need to do it second time, I do not follow your logic here.

pfalcon, 2019-03-26:
Well, see my detailed reply above - it's unclear how to do these docs in the best way, so something done in haste might need to be reworked later. Anyway, on it, based on your suggestions ("just to fill blanks in https://docs.zephyrproject.org/latest/reference/networking/sockets.html")

pfalcon, 2019-03-26:
So, we're just starting and already hit issues: https://github.com/zephyrproject-rtos/zephyr/issues/14907 . I wish all this stuff was done in due time and not out of haste.

PR #14912 (2019-04-04): net: sockets: Add docstrings for BSD Sockets API
The current idea is that we document zsock_* prefixed symbols, refer
to Open Group POSIX website
(http://pubs.opengroup.org/onlinepubs/9699919799/) for normative
descriptions, and explicitly mention bare POSIX name of a function
too (so e.g. users could find it via search).

Fixes: #13397
commit bd10c72b (2019-04-04): net: sockets: Add docstrings for BSD Sockets API

The current idea is that we document zsock_* prefixed symbols, refer
to Open Group POSIX website
(http://pubs.opengroup.org/onlinepubs/9699919799/) for normative
descriptions, and explicitly mention bare POSIX name of a function
too (so e.g. users could find it via search).

Fixes: #13397
closed 2019-04-04 by galak