zephyrproject-rtos/zephyr #13403: USBD event handling rework

pawelzadrozniak, 2019-02-14:
USB device stack - power state tracking.
In a real life application with USB device stack, we need to keep track of power state. if the cable is connected or not and if the bus is suspended or active. I see it as a fundamental feature, so it could be embedded in USBD to simplify application.

Currently, USBD acts as a proxy passing driver events to the class implementation and the only way knowing the bus state in application is to use class callback - classes forward the event from driver. To make things worse, we will get multiplied events (i.e. "suspended") when using a multiple classes/instances (composite device).
Power / bus events are related to the stack itself and IMO should not be coming from class.

This is how see it:
A main dispatcher in usb_device.c (just like the one for composite device) which will forward driver events to every class instance + application callback (only necessary events, like cable connect/disconnect, etc.).
Classes would pass only related events in a callback, i.e. HID events.
In such case - do we still need to split implementation info composite and non-composite device?
I would prefer to avoid #ifdef CONFIG_COMPOSITE_DEVICE everywhere.

Please let me know what you think.

jfischer-phytec-iot, 2019-02-14:
>In such case - do we still need to split implementation info composite and non-composite device?
I would prefer to avoid #ifdef CONFIG_COMPOSITE_DEVICE everywhere.

No, this should go away. 
I started working on it, but I do not have time to finish it. @aurel32 has opened a PR that goes in this direction https://github.com/zephyrproject-rtos/zephyr/pull/11041

emob-nordic, 2019-12-19:
Closed by #20375

closed 2019-12-19 by emob-nordic