zephyrproject-rtos/zephyr #13408: DT_FLASH_AREA generated seems to be different for Zephyr and MCUBootloader

rgundi, 2019-02-14:
The latest Zephyr HEAD generates the below DT symbols (tests/boards/intel_s1000_crb/build/zephyr/include/generated/generated_dts_board.conf)

# DT_FLASH_AREA
DT_FLASH_AREA_0_DEV="MX25UM512"
DT_FLASH_AREA_0_LABEL=IMAGE_SCRATCH
DT_FLASH_AREA_0_OFFSET_0=3997696
DT_FLASH_AREA_0_SIZE_0=196608
DT_FLASH_AREA_1_DEV="MX25UM512"
DT_FLASH_AREA_1_LABEL=MCUBOOT
DT_FLASH_AREA_1_OFFSET_0=0
DT_FLASH_AREA_1_SIZE_0=196608
DT_FLASH_AREA_2_DEV="MX25UM512"
DT_FLASH_AREA_2_LABEL=IMAGE_1
DT_FLASH_AREA_2_OFFSET_0=2097152
DT_FLASH_AREA_2_SIZE_0=1900544
DT_FLASH_AREA_3_DEV="MX25UM512"
DT_FLASH_AREA_3_LABEL=IMAGE_0
DT_FLASH_AREA_3_OFFSET_0=196608
DT_FLASH_AREA_3_SIZE_0=1900544
DT_FLASH_AREA_IMAGE_0_ID=3
DT_FLASH_AREA_IMAGE_1_ID=2
DT_FLASH_AREA_IMAGE_SCRATCH_ID=0
DT_FLASH_AREA_MCUBOOT_ID=1
DT_FLASH_AREA_NUM=4
DT_FLASH_AREA_0_OFFSET=3997696
DT_FLASH_AREA_0_SIZE=196608
DT_FLASH_AREA_1_OFFSET=0
DT_FLASH_AREA_1_SIZE=196608
DT_FLASH_AREA_2_OFFSET=2097152
DT_FLASH_AREA_2_SIZE=1900544
DT_FLASH_AREA_3_OFFSET=196608
DT_FLASH_AREA_3_SIZE=1900544

The latest MCUBootloader HEAD generates the below DT symbols (boot/zephyr/build/zephyr/include/generated/generated_dts_board.conf)

# DT_FLASH_AREA
DT_FLASH_AREA_0_DEV="MX25UM512"
DT_FLASH_AREA_0_LABEL=IMAGE_1
DT_FLASH_AREA_0_OFFSET_0=2097152
DT_FLASH_AREA_0_SIZE_0=1900544
DT_FLASH_AREA_1_DEV="MX25UM512"
DT_FLASH_AREA_1_LABEL=IMAGE_0
DT_FLASH_AREA_1_OFFSET_0=196608
DT_FLASH_AREA_1_SIZE_0=1900544
DT_FLASH_AREA_2_DEV="MX25UM512"
DT_FLASH_AREA_2_LABEL=MCUBOOT
DT_FLASH_AREA_2_OFFSET_0=0
DT_FLASH_AREA_2_SIZE_0=196608
DT_FLASH_AREA_3_DEV="MX25UM512"
DT_FLASH_AREA_3_LABEL=IMAGE_SCRATCH
DT_FLASH_AREA_3_OFFSET_0=3997696
DT_FLASH_AREA_3_SIZE_0=196608
DT_FLASH_AREA_IMAGE_0_ID=1
DT_FLASH_AREA_IMAGE_1_ID=0
DT_FLASH_AREA_IMAGE_SCRATCH_ID=3
DT_FLASH_AREA_MCUBOOT_ID=2
DT_FLASH_AREA_NUM=4
DT_FLASH_AREA_0_OFFSET=2097152
DT_FLASH_AREA_0_SIZE=1900544
DT_FLASH_AREA_1_OFFSET=196608
DT_FLASH_AREA_1_SIZE=1900544
DT_FLASH_AREA_2_OFFSET=0
DT_FLASH_AREA_2_SIZE=196608
DT_FLASH_AREA_3_OFFSET=3997696
DT_FLASH_AREA_3_SIZE=196608

Please note that the IDs assigned are different in both the cases.
Zephyr:
DT_FLASH_AREA_IMAGE_0_ID=3
DT_FLASH_AREA_IMAGE_1_ID=2
DT_FLASH_AREA_IMAGE_SCRATCH_ID=0
DT_FLASH_AREA_MCUBOOT_ID=1

MCUBoot:
DT_FLASH_AREA_IMAGE_0_ID=1
DT_FLASH_AREA_IMAGE_1_ID=0
DT_FLASH_AREA_IMAGE_SCRATCH_ID=3
DT_FLASH_AREA_MCUBOOT_ID=2

I am not quite sure if this is going to have any effect on mcubootloader. However, it is surprising how/why they can be different. Thought of bringing this up as some things have changed in this space in both zephyr and mcuboot.

galak, 2019-02-14:
> I am not quite sure if this is going to have any effect on mcubootloader. However, it is surprising how/why they can be different. Thought of bringing this up as some things have changed in this space in both zephyr and mcuboot.

I dont think anything changed, its just that the order we parse the list of partitions varies build to build right now.  I wanted to leave that artifact in place as it means no one will depend on the order being a certain way.

closed 2019-02-14 by galak

ulfalizer, 2019-02-14:
It's actually non-deterministic. The areas get shuffled around between runs, though with the same values in them. I've noticed the same thing while working on https://github.com/zephyrproject-rtos/zephyr/pull/13385 and earlier.

The Python code might be iterating through a `set()` somewhere. Get an arbitrary order there.

rgundi, 2019-02-14:
Sounds good. Thanks for the clarification @galak and @ulfalizer.