zephyrproject-rtos/zephyr #13417: tests/drivers/watchdog/wdt_basic_api/testcase.yaml: test_wdt_no_callback() failed

cinlyooi-intel, 2019-02-15:
**Describe the bug**
After restart, test_wdt_no_callback() output stop at "Waiting to restart MCU"

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=arduino2 ..
3. make
4. make run

**Screenshots or console output**
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Watchdog setup error
Waiting to restart MCU
```

**Environment (please complete the following information):**
 - OS: Fedora29
 - Toolchain: zephyr
 - Commit SHA or Version used: 9072d34f306078acdc9d950c9beae5cc0738ea19

**Additional context**
Add any other context about the problem here.

cinlyooi-intel, 2019-03-04:
Repeated  in Commit ID fd7c4430e1398374f2f5b49bcd6ec95166e52e9a for sam_e70_xplained

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-704-gfd7c4430e139 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Watchdog setup error
Waiting to restart MCU
```

aurel32, 2019-03-08:
The problem is that on the SAM WDT can only be enabled or disabled once (the mode register is write once). By default the `CONFIG_WDT_SAM_DISABLE_AT_BOOT` option is enabled and disables the watchdog. Then it's not possible to enable it again.

After disabling this option, the test passes. I wonder how this option is different from the generic `CONFIG_WDT_DISABLE_AT_BOOT` option which is NOT enabled by default.

aurel32, 2019-03-08:
> After disabling this option, the test passes.

Well I am wrong there. The MCU reboots correctly when the WDT triggers, however it also reset the RAM, while the test assumes that the RAM is left untouched by the WDT reset.

overheat, 2019-04-09:
Fixed through #14968

ioannisg, 2019-11-26:
> > After disabling this option, the test passes.
> 
> Well I am wrong there. The MCU reboots correctly when the WDT triggers, however it also reset the RAM, while the test assumes that the RAM is left untouched by the WDT reset.

So this test, as is, cannot pass on the SAM WDT @aurel32 and we should just exclude the platform from the test for now, until we extend the test to cover the SAM watchdog.

@nashif any thoughts against?

PR #21008 (2019-11-27): tests: drivers: watchdog: exclude platforms with SAM WDT from testing
On watchdog-triggered reboot, the SAM platforms
reset RAM, so the wdt_basic_api cannot be completed
successfully, as it relies on RAM retention (relies
on variables stored in RAM retaining their values
across different boot cycles). Exclude the platforms
with the SAM Watchdog from this test.

Fixes #13417
closed 2019-11-27 by nashif