zephyrproject-rtos/zephyr #13421: tests/drivers/watchdog/wdt_basic_api: test_wdt_no_callback() repeats indefinitely

cinlyooi-intel, 2019-02-15:
**Describe the bug**
We are seeing the same second console log output for https://github.com/zephyrproject-rtos/zephyr/issues/10623 for the same board type, i.e. quark_d2000:x86

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=quark_d2000_crb -DARCH=x86
3. make
4. See error

**Screenshots or console output**
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-211-g95e8a582b4 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
```

cinlyooi-intel, 2019-02-15:
@nashif : I believe you will be closing this bug as the same as #8097 just like #10623. 

When you close this as duplicate can you put a brief explanation of your reasoning please? It is difficult for me to see how this is the same as #8097 because I am not as experience as you.

cinlyooi-intel, 2019-03-04:
More work done and it is not an automation failure

Very occasionally, this test has been known to pass.

SavinayDharmappa, 2019-03-05:
https://github.com/zephyrproject-rtos/zephyr/pull/13659

SavinayDharmappa, 2019-03-05:
@cinlyooi-intel  as stated in #13659 patch exclude the watchdog test case for quark_d2000_crb
,as it seen that upon watchdog reset the conents of ram are lost.
Hence the existing testcase fails.

closed 2019-03-05 by SavinayDharmappa