zephyrproject-rtos/zephyr #13422: Can't use GPIO 2, 3 and 4

paultjepils1999, 2019-02-15:
**Describe the bug**
Can't use GPIO 2, 3 and 4. When I compile/run the application it shows errors.
I have tried to manually ad the defines to the generated_dts_board.conf but that doesn't work for me. please see image below. Are there any other settings I have to configure or is this a bug?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1050_evk ..
3. sudo ninja debug
4. See error

**Expected behavior**
My expection was that the application would compile and run.

**Impact**
Can't power on a simple LED from GPIO_2_10 (D9), GPIO_1 and GPIO_5 are working (the built in LED can be powered on and off)

**Screenshots or console output**
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu 16.04

**Additional context**
prj.conf:
CONFIG_PRINTK=y
CONFIG_KERNEL_SHELL=y
CONFIG_OBJECT_TRACING=y
CONFIG_THREAD_MONITOR=y
CONFIG_INIT_STACKS=y
CONFIG_BOOT_BANNER=n

CONFIG_GPIO=y

galak, 2019-02-15:
@paultjepils1999 Guessing you need to update the DTS to enable those GPIO blocks and probably do proper pin muxing/cfg as well.

paultjepils1999, 2019-02-15:
@galak Do you mean by downloading the newest zephyr project? because that doesn't work. I configured the gpio pin mux the same as the the built in LED like this:

IOMUXC_SetPinMux(IOMUXC_GPIO_B0_10_GPIO2_IO10, 0);

IOMUXC_SetPinConfig(IOMUXC_GPIO_B0_10_GPIO2_IO10,
			    IOMUXC_SW_PAD_CTL_PAD_PKE_MASK |
			    IOMUXC_SW_PAD_CTL_PAD_SPEED(2) |
			    IOMUXC_SW_PAD_CTL_PAD_DSE(6));

PR #13438 (2019-02-16): soc: nxp_imx: Add missing dts fixups for gpio 2,3,4 instances
The imx rt dts fixup was missing gpio 2,3,4 instances. This caused a
build error in the gpio driver when an application enabled:

CONFIG_GPIO_MCUX_IGPIO_2=y
CONFIG_GPIO_MCUX_IGPIO_3=y
CONFIG_GPIO_MCUX_IGPIO_4=y

Fixes #13422
closed 2019-02-16 by galak