zephyrproject-rtos/zephyr #13423: Default logger stack size insufficient for various samples

rlubos, 2019-02-15:
Stack sentinel reports that logger stack overflows in default configuration. I've originally obesrved this problem in `echo_client` and `echo_server` samples running on `nrf52840_pca10056`, but this problem can be reproduced in other samples as well (verified with `logger` sample):
```
Current thread ID = 0x2000055c
Faulting instruction address = 0x4aee
```

nordic-krch, 2019-03-07:
@rlubos can you check echo_client and echo_server now after #14036 is merged. I've checked logger sample and it has no issues now. #14036 saved ~250 bytes on logger thread stack.

rlubos, 2019-03-07:
@nordic-krch Thank you, it seems that #14036 fixed the problem - I no longer observe a crash. Therefore I believe we can close the issue.

closed 2019-03-07 by rlubos