zephyrproject-rtos/zephyr #13432: ARM: Fault Handling Mechanism and how to correctly report Stack Corruptions

ioannisg, 2019-02-15:
We seem to  have technical challenges for both the MPU-based stack corruption detection, as well as for the detection based on the ARMv8-M stack pointer LIMIT registers.

First, about MPU. The way we implement the guard area, for stack corruption is somewhat problematic for SUPERVISOR threads. 
The reason is simple; when there is a stack overflow, and the area is violated by Push instructions, the stack frame is _NOT_ loaded in the guard area.
This means, that we cannot "by design" extract safely the instruction address that caused the error! Now, we might be lucky enough,
and there was an (at least) 8-byte "slack" between SP and Guard at the time we did the Push that overflowed the stack.
In that case, the "assumed" stack context will have the upper 8-byte part of the stack frame correct (and we will be able to read the address)!
This will be the PC and XPSR. But, in general, we cannot be sure about the faulty instruction address, and, of course, we can not do any "decoding" to see what type of instruction it was.
This is not a tragedy, but we must state this clearly. I think, the same applies for USER threads,
because the processor MODE changes after we branch to the MPU exception; the stacking occurs earlier, and... same problem.

Fortunately, the processor does report "Precise data access violation" _and_ "Stacking Error", _and_ the violated address.
It is safe to report a stack corruption in this scenario, because it could only have been the following:
- The "normal" case: thread tried to push to stack, violated the guard and stacking also (naturally) failed
- The "crazy" case: thread tried to explicitly write to the stack guard area (for some crazy reason), MPU fault raised, and (only) its Stacking caused the overflow.
  But it _was_ an overflow, and it was due to thread's operations, so it is ok to report it.

Things we WON'T report, now, as Stack corruptions:
- Standard stacking error due to interrupt entry. This IS actually a minor problem, cause, practically, the thread stack did run out of space, but
  we only caught it through an un-related exception. We can decide if we want to report this as a Stack Fail. Note, however, that we won't have the instruction address, of course,
  but, also, we won't have the violated address in this case (it is not a Direct Address Violation). **_So we need to decide what to do with that._**
  
- The "crazy" scenario, where thread tried to explicitly write to the stack guard area, so an MPU fault raised but not together with Stacking error (i.e. there was lots of stack area remaining,
  to stack the MPU fault context frame). I think this is OK to not report it as Stack Limit Check fail.
  

Second, about ARMv8-M and the built-in stack overflow checking:
Here the things are slightly different. First, we do not have a guard; just a limit on the PSP.
If a stack overflow occurs, we definitely know it was the right stack overflow (due to STMDB sp! that made the SP try to go beyond the preset limit).
So we are safe to report Stack Limit Check Fail, but we won't have a stack frame to present to the user at all!

---

My proposal is to just report to the user that the instruction address may be trash, and report Stacking && Direct Violation errors as stack corruptions (if memory addresses are within the guard areas)

agross-linaro, 2019-02-15:
In the case of the IRQ causing a stack overflow due to the stack frame push, we can still at least determine that it was an overflow.  We know the stack address and the size of the stack frame, so we can backup the address to determine which thread stack it was that overflowed.  We won't have the instruction, but we would at least know it ran out of stack space.

ioannisg, 2019-02-20:
Raising this priority to high - we need to understand what is going on with the tail-chaining behavior in ARMv8-M, under faults with stacking error.

ioannisg, 2019-02-21:
We filed a case for the tail-chaining with ARM - the details are below:

So, briefly, the situation is as follows:

1.	Our aim is to detect stack corruption due to stack overflow in un-privileged mode. We are aware of having SPLIM registers in ARMv8, but we want a solution to be shared with ARMv7, that’s why we want an MPU-based solution.
2.	So we configure the unprivileged thread stack with nPRIV RW permissions – all other SRAM is set to PRIV RW. Our PSP is initially pointing inside the unprivileged stack.
3.	As soon as we overflow the unprivileged stack, we get MemManage “Stacking Error” and, possibly, also “Data Access Violation”. That’s as expected. According to the spec, the “created” stack frame may be corrupted (that’s what we observe), but the Stack Pointer has actually descent by -0x20 (that’s also what we observe).
4.	Inside the MemManage handler we inspect the error and “pend” the pendSV exception, to do the required RTOS re-scheduling after the MPU fault  handler returns.
5.	Upon the return of the MemManage handler, we experience MemManage “Unstacking Errors”. The specification says that this is due to the fact that the restoration of the stack frame is performed with the privilege mode where we return to. And since we were in nPRIV mode, and try to pop from a PRIV RW stack, we get errors. So, this is also expected, however:
6.	According to the specification, we would expect the Cortex-M33 to perform tail-chaining of the pendSV exception. This does not seem to occur in ARMv8-M. It does occur in ARMv7-M. Tail chaining would help to skip the Unstacking Error, as we would normally unstack only after the pendSV return, and that would probably move the stack pointer elsewhere. So our inquiry concerns the experienced skipping of tail-chaining of the pendSV exception, after the MPU stacking error; should we expect this? Where could we find the explanation in the spec?

closed 2019-02-28 by andrewboie

ioannisg, 2019-03-01:
> Raising this priority to high - we need to understand what is going on with the tail-chaining behavior in ARMv8-M, under faults with stacking error.

We have now understood what happens here.

This has been closed automatically. I am just adding here the resolution regarding tail-chaining.
Interrupt tail chaining is an optimization and, in general, software design shall not be based on the assumption that it definitely occurs, even if the conditions, according to the specification, are met. 

Therefore, the software is responsible to always ensure that Un-stacking errors do not occur, and not rely on tail-chaining. We have implemented this in #13334 .

PR #13334 (2019-02-28): Arch arm core fix fault report nxp mpu
Fixes #7706 
Fixes #13342 
Closes #13432 

This PR
- investigates stack corruptions, not only or precise memory violations, but also on stacking errors
-  refactors fault.c, so the assessment of stack fail is moved to thread.c, as it accesses kernel structures,
- allows reporting thread stack corruption for user threads stacks,
- improves error messages and inline documentation related to ARM memory faults,