zephyrproject-rtos/zephyr #13433: Error when rebooting the frdm_k64f board

chtavares592, 2019-02-15:
Hello, I have a question, if anyone can help me I will be very happy. I have an application on zephyr, that downloads an image and saves it in flash area 1. I use the last mcuboot and the board is frdm_k64f. When I rebooted happens that error bellow:

```
[00:00:00.004,948] <inf> mcuboot: Starting bootloader
[00:00:00.010,967] <inf> mcuboot: Image 0: magic=good, copy_done=0x3, image_ok=0x1
[00:00:00.018,793] <inf> mcuboot: Scratch: magic=unset, copy_done=0x5b, image_ok=0x3
[00:00:00.016,952] <inf> mcuboot: Boot source: slot 0
[00:00:00.014,445] <inf> mcuboot: Swap type: test
***** BUS FAULT *****
  Imprecise data bus error
***** Hardware exception *****
Current thread ID = 0x00000000
Faulting instruction address = 0x3ebe
```
I don't know why it happens.
So, I used the git bisect to find where started the problem, and the commit is:

```
commit dc0713a706e7e2898c172c6c0965b1b784859841
Author: Andy Ross <andrew.j.ross@intel.com>
Date:   Thu Jul 5 12:50:41 2018 -0700

    kernel: Cleanup. Remove redundant test when calling _Swap()
    
    _Swap() must already handle the case where _get_next_ready_thread() is
    the same as _current.
    
    Signed-off-by: Andy Ross <andrew.j.ross@intel.com>
```
Unfortunately, I don't know how to solve it.

MaureenHelm, 2019-02-18:
This is caused by the mcux flash driver using a semaphore while mcuboot/zephyr has multithreading disabled. We can either enable multithreading in mcuboot or modify the mcux flash driver similar to 408ea1464b254972f00469cad4f23c078c2857d4.

chtavares592, 2019-02-18:
@MaureenHelm, I changed that on mcuboot, and that worked fine, thanks for your help.

d3zd3z, 2019-02-19:
How much does this change the code size?

d3zd3z, 2019-02-19:
Wouldn't it be better to fix the semaphore/locking code to not crash if multithreading isn't enabled?

mike-scott, 2019-02-19:
@d3zd3z I put up a PR to fix the flash driver for when CONFIG_MULTITHREADED=n

d3zd3z, 2019-02-19:
Aside from the fact that semaphores != mutexes, but that is a deeper issue throughout much of the code.

mike-scott, 2019-02-19:
Ah nm, I see what you're saying.  I bet that code is likely used for several (and sometimes the wrong reasons).  It might be hard to generically "fix" it for non-multithreaded usecase.

agross-linaro, 2019-02-19:
Same issues with PSOC6 when I call the CM4 enable.  Multithreading is enabled.  This is a SMP system as well.  But it crashes during hello world if I add a line to release the CM4 from the CM0.  This is just register writes and a busy wait.

nashif, 2019-02-19:
> Normally, we'd really like to be able to build MCUboot without multi-threading, as it isn't needed..

That depends on what you want to do with your bootloader, not needed for many cases maybe, but needed in some. For example I am interested in having USB support in the bootloader, and that requires multi-threading.
Also, some hardware can afford the extra bytes...

d3zd3z, 2019-02-19:
Bootloader without multithreading is absolutely required for small targets. We need to fix anything that keeps that from working or people won't be able to use Zephyr on these targets.

d3zd3z, 2019-02-19:
>> Normally, we'd really like to be able to build MCUboot without multi-threading, as it isn't needed..
> That depends on what you want to do with your bootloaders.

Please realize this is a non-sequitur. "Some targets may want multithreading, therefore it is ok if multithreading=n is broken." We don't avoid bugs because some code doesn't provoke that bug.

nashif, 2019-02-20:
> Bootloader without multithreading is absolutely required for small targets.

Multithreading in itself is orthogonal to the whole discussion, the only reason we are talking about threading here is because we managed to save a few kilobytes when we disabled this "feature". Bootloaders do not really care if they have multi-threading or not.

> therefore it is ok if multithreading=n is broken

Where is it broken? 
Drivers being enabled in bootloader that require multi-threading are broken in this case, not threading it self. 

This feature had 1 single purpose, to save a few bytes, Zephyr was never designed and IMO should not be changed to support single-threaded applications in a way that things would work just out of the box, also those that require multi-threading such as USB.

So, in bugs like this one, instead of saying that multithreading=n is a must for mcuboot the reporter should be aware that if they have enough flash they should be able to just enable threading and move on without having to go and change drivers and subsystems to become single-threaded.

andyross, 2019-02-20:
Yeah, as @nashif points out CONFIG_MULTITHREADING is fundamentally just a size optimization, and not the only way to skin this cat.  It was the result of a really clever afternoon of hackery by Ben Walsh where he would stare at the linker output and add #ifdefs as needed to get needless references to drop out.  It's fragile, and over time has bitrotten to the point where it doesn't actually provide the size benefits it did originally, anyway.

In theory, most ("most") of the work should be doable via other means.  The timer driver can be disabled in the current tree, and if you do (and also don't call any APIs that need timeouts) the whole subsystem should disappear.   Likewise all of the scheduler should drop out if unused except for a few bits that are referenced by the ISR exit paths.

But that too requires "unused" to be true -- here we had a driver calling into semaphore IPC code, which is going to suck stuff into the build that Just Plain Can't Work without "multithreading".  Maybe it used to by accident, but it can't in general (in this case specifically, you can't block on a semaphore, the kernel literally has no way of doing that without another thread to switch to!)

Somewhere there's an issue filed where I talk about reworking the semaphore APIs (which drivers tend to use for "wake up userspace") into a less general thing where the thread code would spin waiting on the driver.

I mean, I agree that minimal size restricted mode bootloaders (with tiny, supportive drivers like flash and uart) are a valuable target and we should work to support them.  But the current CONFIG_MULTITHREADING is just being asked to do too much.  Someone needs to come back from scratch and rethink the scheme.

MaureenHelm, 2019-02-22:
> How much does this change the code size?

CONFIG_MULTITHREADING=n
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       40488 B         1 MB      3.86%
            SRAM:       23836 B       192 KB     12.12%
        IDT_LIST:         120 B         2 KB      5.86%
```

CONFIG_MULTITHREADING=y
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       42444 B         1 MB      4.05%
            SRAM:       24172 B       192 KB     12.29%
        IDT_LIST:         120 B         2 KB      5.86%
```

andyross, 2019-02-23:
Wow, that's... big.  What's in that 40k?  For reference, when CONFIG_MULTITHREADING was implemented the first time it took a minimal Zephyr build of around 8k down to 4k or so (very vague memory, but it was roughly in that range).  And it seems like we've eroded away about half the savings, so only get about 2k these days.

The bottom line is I wouldn't be shocked if, with a little work, we could recover that missing 2k.  But that's just not going to make much difference on a 40k binary.  There's only so much kernel code to begin with.

andyross, 2019-02-23:
Yeah.  Just tried a hello_world build on qemu_cortex_m3 (i.e. the closest build we have to "just the ARM kernel, please").  It's 8944 bytes of flash by default, shrinking to 6688 with MULTITHREADING=n.  That's pretty much the kind of delta we're looking at.

Is it really such a hardship to force your 40k bootloader to waste an extra 2256 bytes?

nashif, 2019-02-23:
Not that CONFIG_DEBUG is enabled by default in mcuboot, so just disabling that will give you

```
Memory region         Used Size  Region Size  %age Used
           FLASH:       37104 B         1 MB      3.54%
            SRAM:       24172 B       192 KB     12.29%
        IDT_LIST:         120 B         2 KB      5.86%
```

with multithreading on.

Then you have logging enabled, keeping this and applying

CONFIG_LOG_DISABLE_FANCY_OUTPUT_FORMATTING=y gives:

```
Memory region         Used Size  Region Size  %age Used
           FLASH:       34904 B         1 MB      3.33%
            SRAM:       24172 B       192 KB     12.29%
        IDT_LIST:         120 B         2 KB      5.86%
```

you can go and disable printk and logging, which are enabled by default and seem to be more important than multithreading and get:

```
Memory region         Used Size  Region Size  %age Used
           FLASH:       28332 B         1 MB      2.70%
            SRAM:       24092 B       192 KB     12.25%
        IDT_LIST:         120 B         2 KB      5.86%
```

and there is lots more spaces to save still.

nashif, 2019-02-24:
closing given the original reporter got things to work and multi-threading is not the culprit here. there are many places where space can be saved and if there is a need to make drivers single-thread aware, then this is another issue.

closed 2019-02-24 by nashif