zephyrproject-rtos/zephyr #13434: Aliases inside dts leads to warnings

tautologyclub, 2019-02-15:
Having something like 

	aliases {
		apds9306 = &apds9306;
	};

inside your dts now triggers the `__DEPRECATED_MACRO` warning because the dts generation script doesn't prefix alias-based defines with `DT_`. I doubt this was the intention. This was introduced in 
 commit 561d6dd1ed0f1d3a04916646dfcdb6082fb89155 with absurdly arbitrary exemptions for \#defines that start with `LED`, `SW` and `PWM_LED`. 

So the commit should probably be reverted until the dts-alias generation has been fixed so it prefixes its defines with `DT_`.

ulfalizer, 2019-02-15:
Looks like there might be two different macros generated for aliases - one with a `DT_` prefix.

Check  `new_alias_prefix` in `add_prop_aliases()` in `scripts/dts/extract/globals.py`, and commit a7c430f ("scripts: dts: Use DT_ prefix in generated labels").

galak, 2019-02-16:
When we introduced the alias generation, we added DT_ prefix as @ulfalizer mentioned.  So the non-DT_ one's are concerned deprecated.

closed 2019-02-16 by galak

zerotri, 2019-04-10:
@galak @ulfalizer As the non-DT_ prefixed macros are being deprecated, is there any plan for generating macro labels that are less hardware specific?

In a prior release I could use an alias to define a macro `SERIAL_RS232_LABEL` that could translate to either `DT_NS16550_SERIAL_RS232_LABEL` or `DT_ST_STM32_USART_SERIAL_RS232_LABEL` depending on which board I was building for. The same code could potentially run unchanged on either. What is the recommended way of achieving this going forward without maintaining multiple configurations per board per application?