zephyrproject-rtos/zephyr #13436: Enabling cooperative scheduling via menuconfig may result in an invalid configuration

mnkp, 2019-02-15:
**Bug description**
When developing an application user may want to test different kernel scheduling algorithms. By default both cooperative and preemptive scheduling are enabled. To enforce cooperative scheduling only user is likely to use `ninja menuconfig` or equivalent command to set `CONFIG_NUM_PREEMPT_PRIORITIES=0`. Unfortunately this will result in an invalid configuration. There are two other Kconfig options `CONFIG_MAIN_THREAD_PRIORITY` and `CONFIG_SYSTEM_WORKQUEUE_PRIORITY` which also need to be modified to ensure cooperative scheduling works correctly.

**To Reproduce**
1. build hello_world example
2. run `ninja menuconfig` and set `CONFIG_NUM_PREEMPT_PRIORITIES` to 0.
3. recompile and flash the firmware
4. the console reports `***** USAGE FAULT *****` error. 

The zephyr/.config file contains the following settings
```
CONFIG_MAIN_THREAD_PRIORITY=0
CONFIG_SYSTEM_WORKQUEUE_PRIORITY=-1
```
event though this is not a valid configuration for cooperative scheduler.

**Expected behavior**
While having `ninja menuconfig` automatically configure required dependencies does not seem to be possible in this case we should:
- better document `CONFIG_NUM_PREEMPT_PRIORITIES`, `CONFIG_NUM_COOP_PRIORITIES` Kconfig options mentioning the dependencies and the recommended settings.
- generate a compile time error, e.g. via `BUILT_ASSERT_MSG` if an invalid set of Kconfig options is detected

**Impact**
This issue is relatively easy to run into and difficult to debug.

**Environment**
 - Version used v1.14.0-rc1

ulfalizer, 2019-02-19:
Not an expert on these symbols, but if `MAIN_THREAD_PRIORITY` and `SYSTEM_WORKQUEUE_PRIORITY` should be fixed whenever `NUM_PREEMPT_PRIORITIES` is 0, then this patch will do:

```diff
diff --git a/kernel/Kconfig b/kernel/Kconfig
index d2261067a8..a53b948e25 100644
--- a/kernel/Kconfig
+++ b/kernel/Kconfig
@@ -78,7 +78,7 @@ config NUM_PREEMPT_PRIORITIES
          priority, and be the only thread at that priority.
 
 config MAIN_THREAD_PRIORITY
-       int "Priority of initialization/main thread"
+       int "Priority of initialization/main thread" if PREEMPT_ENABLED
        default -2 if !PREEMPT_ENABLED
        default 0
        help
@@ -368,7 +368,7 @@ config SYSTEM_WORKQUEUE_STACK_SIZE
        default 1024
 
 config SYSTEM_WORKQUEUE_PRIORITY
-       int "System workqueue priority"
+       int "System workqueue priority" if PREEMPT_ENABLED
        default -2 if COOP_ENABLED && !PREEMPT_ENABLED
        default  0 if !COOP_ENABLED
        default -1
```

The problem is that any user-configurable symbol (any symbol with a visible prompt) will get "locked" to its value whenever the configuration is saved and later loaded. It has to work that way, because there's no way to tell whether the value of the symbol came from the user setting it explicitly (in e.g. `menuconfig`) or was derived from `default`s.

To prevent the file from getting locked (and the user changing it), the symbol must be made non-configurable.

I'll make a PR with the patch tomorrow unless someone complains.

mnkp, 2019-02-19:
@andyross and @andrewboie will now better but I believe we want user to be able to configure main thread and work queue priorities when cooperative scheduling is used. So the above patch is probably not the right approach.

ulfalizer, 2019-03-25:
Back after reading the manual a bit to better understand what these actually do.

Was thinking a `range` could be used maybe, but the priorities being up to `CONFIG_NUM_PREEMPT_PRIORITIES - 1` makes it a bit messy (plus `CONFIG_NUM_COOP_PRIORITIES` being negative).

Think a compile-time check somewhere might be best for now.

SebastianBoe, 2020-02-13:
Hi,

I would like to give a go at fixing this, but I was not able to reproduce the runtime error.

It's stated that after setting CONFIG_NUM_PREEMPT_PRIORITIES=0

then

```
CONFIG_MAIN_THREAD_PRIORITY=0
CONFIG_SYSTEM_WORKQUEUE_PRIORITY=-1

```

will be produced and this is invalid. I am able to reproduce that these options are set in this manner.
But what if this is an invalid configuration, then what would be the desired values?

carlescufi, 2021-02-09:
@tejlmand do you have plans to address this?

tejlmand, 2021-02-09:
No concrete plans at the moment, as solving this is not an easy thing to do.
The approach examined in #22800 turned out to have other unintended effects.

closed 2021-04-26 by github-actions[bot]