zephyrproject-rtos/zephyr #13444: Build failure when including both socket.h and time.h

vanti, 2019-02-16:
**Describe the bug**
When including both headers:

```
#include <net/socket.h>
#include <posix/time.h>
```

A build error is seen:
```

In file included from /home/zephyrproject-work/zephyr/samples/basic/blinky/src/main.c:11:0:
/home/zephyrproject-work/zephyr/include/net/socket.h:218:17: error: redefinition of ‘struct zsock_timeval’
 #define timeval zsock_timeval
                 ^
/home/zephyrproject-work/zephyr/include/posix/time.h:23:8: note: in expansion of macro ‘timeval’
 struct timeval {
        ^~~~~~~
/home/zephyrproject-work/zephyr/include/net/socket.h:34:8: note: originally defined here
 struct zsock_timeval {
        ^~~~~~~~~~~~~
/home/zephyrproject-work/zephyr/samples/basic/blinky/src/main.c: In function ‘main’:
/home/zephyrproject-work/zephyr/samples/basic/blinky/src/main.c:24:17: warning: unused variable ‘tv’ [-Wunused-variable]
  struct timeval tv;
                 ^~
CMakeFiles/app.dir/build.make:62: recipe for target 'CMakeFiles/app.dir/src/main.c.obj' failed
make[2]: *** [CMakeFiles/app.dir/src/main.c.obj] Error 1

```

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/basic/blinky
2. include the two headers as shown above
3. enable posix api and networking in prj.conf:
```
CONFIG_POSIX_API=y
CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
```
4. mkdir build; cd build
5. cmake .. -DBOARD=cc3220sf_launchxl
6. make
7. See error

**Expected behavior**
It should be possible to include both header files

**Impact**
Prevents code from building when both headers are needed.
A workaround seems to be to comment out this line in the net/socket.h:
`#define timeval zsock_timeval`
This change allows the code to build.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.9.5
 - Commit SHA or Version used: 10f2bdfc5709bb8780ba41224d4c867eec26c431

pfalcon, 2019-02-18:
BSD Sockets API currently doesn't integrate well with POSIX subsys, and POSIX subsys itself has problems with integrating with many parts of Zephyr: libc, specific architectures, etc. So, what's reported here is just one of many issues. (But I can add that posix/time.h is especially peculiar thingy.)

I'm prototyping some solutions to a bunch of issues in https://github.com/zephyrproject-rtos/zephyr/pull/12984 . It's unclear though if/when/how those improvements can land, given that response to even the most simple and obvious, like https://github.com/zephyrproject-rtos/zephyr/pull/13426 lead to erratic responses.

galak, 2019-03-18:
With SDK 0.10.0 and recent changes is this still an issue, can't seem to reproduce.

vanti, 2019-03-18:
@galak I am still seeing the issue on 87b7bee52c34a9989a8b74128ed76011e2cf9a01 with SDK 0.10.0.

pfalcon, 2019-03-19:
As mentioned above, fixing this properly would require some legwork and actually, discussion of the overall approach first (I now submitted https://github.com/zephyrproject-rtos/zephyr/issues/13787 on that). So, I suggest to move this forward from 1.14.

galak, 2019-03-26:
> @galak I am still seeing the issue on [87b7bee](https://github.com/zephyrproject-rtos/zephyr/commit/87b7bee52c34a9989a8b74128ed76011e2cf9a01) with SDK 0.10.0.

what issue, can you explain further.

vanti, 2019-03-28:
@galak the build error that I encountered by following the procedure in the "To Reproduce" section is still present.

galak, 2019-04-16:
Not able to reproduce with SDK 0.10.0 and current master.

closed 2019-04-16 by galak