zephyrproject-rtos/zephyr #13463: frdm_kl25z samples/basic/threads Kernel Panic

hakehuang, 2019-02-18:
**Describe the bug**
frdm_kl25z samples/basic/threads Kernel Panic

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdm_kl25z  ..
3. make
4. See error

**Expected behavior**
thread switch shall be working

**Impact**
1. the gpio setting is not platform independent

**Screenshots or console output**
```
***** Booting Zephyr OS 1.14.0-rc1 *****

ASSERTION FAIL [gpio_dev != ((void *)0)] @ /build/src/workspace/frdmkl25z_master-32BQPWFMHJHABGVMBAYRTODCAVUIGXJ6WJT3NCXTALMIGKHHKJ6A/samples/basic/threads/src/main.c:50

***** Kernel Panic! *****

Current thread ID = 0x1ffff000

Faulting instruction address = 0x1110
```

**Environment (please complete the following information):**
 - OS:  Linux
 - Toolchain  Zephyr SDK
 - Commit SHA or Version used
v1.14.0_rc1

hakehuang, 2019-02-18:
@MaureenHelm

overheat, 2019-02-22:
Hi hakehuang,

I have the same issue with you, and the only difference is that it happened on my pca_10056 board, which is an nRF52840 chipset on it.

Have gone through the code and document, the problem is in the GPIO driver missing on my Nordic pca_10056 board and maybe same on your board. As a result of this, the gpio driver didn't been linked, and a fault happens when you try to invoke it. 

There are two ways to solve this problem:
1. use " ninja menuconfig " and enable "GPIO Drivers" under "Device Drivers".
2. add "CONFIG_GPIO=y" to the end of  zephyr\samples\basic\threads\prj.conf, and re-build it!

Can you test it? I hope everything goes well!

MaureenHelm, 2019-02-22:
> I have the same issue with you, and the only difference is that it happened on my pca_10056 board, which is an nRF52840 chipset on it.

Thanks @overheat for jumping in on this! Turns out it was a slightly different issue on the frdm_kl25z board, so your patch doesn't quite fix it.

overheat, 2019-02-22:
> > I have the same issue with you, and the only difference is that it happened on my pca_10056 board, which is an nRF52840 chipset on it.
> 
> Thanks @overheat for jumping in on this! Turns out it was a slightly different issue on the frdm_kl25z board, so your patch doesn't quite fix it.

Thank you for your explanation.  Since I don't have frdm_kl25z board, I double checked it with "ninja menuconfig" under frdm_kl25z environmental. 
Yes, you are right. It a different issue even the console output is almost the same, cause the gpio driver already enabled for frdm_kl25z . So it looks like another issue for my Nordic's boards. I will re-submit it with a more proper description.

PR #13691 (2019-02-23): drivers: gpio: Fix mcux driver to init successfully with no irq
Some mcux gpio instances do not have dedicated interrupt vectors and
therefore conditionalize out the IRQ_CONNECT() and irq_enable() calls
during driver initialization. The driver initialization incorrectly
returned an error in this case, when really it just has nothing to do.
The driver can still be used without interrupts, and the gpio configure
function returns an error if an application tries otherwise.

Commit a68120de6d414117ddb67566a7397a5e23556841 introduced a check on
the init return value to prevent applications from usi [... cut]
PR #13658 (2019-03-04): samples: threads: Fix ASSERTION FAIL without gpio driver
Some boards' configuration does not enable GPIO by default,
like nrf52_pca10056, cause ASSERTION FAIL when running this
application. The fix is to add "CONFIG_GPIO=y" to prj.conf

Fixes #13463
closed 2019-02-23 by galak