zephyrproject-rtos/zephyr #13465: tests/lib/mem_alloc/testcase.yaml#libraries.libc.minimal: Bus fault at test_malloc

cinlyooi-intel, 2019-02-18:
**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdm_k64f -DARCH=arm ..
3. make
4. make run

**Screenshots or console output**
```
******** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-248-gd5b2834f58 (delayed boot 1000ms) *****
Running test suite test_c_lib_dynamic_memalloc
===================================================================
starting test - test_malloc
***** BUS FAULT *****
  Precise data bus error
  BFAR Address: 0x20001158
  NXP MPU error, port 3
      Mode: User, Data Address: 0x20001158
      Type: Read, Master: 0, Regions: 0x8800
***** Hardware exception *****
Current thread ID = 0x20000b3c
Faulting instruction address = 0x7ee
Fatal fault in thread 0x20000b3c! Aborting.
```

**Environment (please complete the following information):**
 - OS: fedora29
 - Toolchain: zephyr
 - Commit SHA or Version used: d5b2834f58e11b59c640f00f3f2ca2fb85d3a48c

ioannisg, 2019-02-20:
Thanks, @cinlyooi-intel , I can reproduce this one. Looking at it.

andrewboie, 2019-02-20:
We have a problem in how the thread-specific storage area is being reserved.
The intent of this region is for thread-specific data that the thread can manipulate directly, without having to make a syscall. Currently, we are using this to store errno.

What's happening is that the room for this area is being subtracted from the stack size being passed to _new_thread. With the NXP MPU, this results in this memory not being covered by the MPU region for the stack as it should be.

I'm still studying the code to determine the best solution. The way that stack memory bounds are accounted for differ in subtle ways across ARM, ARC, x86 and this is making this more confusing. My current thinking is that _new_thread needs a 'offset' parameter instead of subtracting the desired offset from the stack size passed in.

andrewboie, 2019-02-20:
@ioannisg if you want to focus on other stuff I can handle addressing this bug.

PR #14194 (2019-03-09): Arch arm fix new thread init
Fixes #13465 

In brief, the patch does the following:
- moves the reservation of userspace thread local data into the arm-specific `_new_thread()`
- corrects the value saved, eventually in `stack_info.size`, which was causing MPU errors, as the user local data were falling outside the MPU-programmed thread stack,
- improves the documentation of _ARCH_THREAD_STACK_SIZEOF, to stress that it gives the area available for the thread operation (i.e. excluding the guard).
- misc stuff, e.g. an assert statement when programming the built-in guard in ARMv8-M.

Priority is  [... cut]
closed 2019-03-09 by andrewboie