zephyrproject-rtos/zephyr #13468: tests/drivers/watchdog/wdt_basic_api/testcase.yaml: Various version of "Waiting to restart MCU"

cinlyooi-intel, 2019-02-18:
**Describe the bug**
Various versions of "Waiting to restart MCU" exists. Unlike #12803 and #13417, this sets of results does not have the "Watchdog setup error" line.

The key here is, we see the test stuck at  "Waiting to restart MCU" line. or after
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-248-gd5b2834f58 (delayed boot 1000ms) *****
which follow the "Wait to restart MCU" line.

At times, we see the test cases keep repeating itself.
In all cases, at the end, test did not return properly.

We will see this when it is running "test_wdt_no_callback" or "test_wdt_callback_1" or  or "test_wdt_callback_2" test cases.

Will be using this bug to store all failure cases at "Watchdog setup error"

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=XXX ... # depends on board
3. make
4. make run

**Screenshots or console output**
Please note that any platform can exhibit the console logs recorded here. The boards listed are for reference to real instance:

_Observed on nrf51_pca10028:_

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-248-gd5b2834f58 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU

```

_Observed on  nrf52_pca10040:_
```
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS v1.14.0-rc1-248-gd5b2834f58 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-248-gd5b2834f58 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Testcase passed
Testcase: test_wdt_callback_1
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS v1.14.0-rc1-248-gd5b2834f58 (delayed boot 1000ms) *****
```

 
_Observed on nrf52840_pca10056:_

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-248-gd5b2834f58 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_callback_1
Testcase passed
Testcase: test_wdt_callback_2
Waiting to restart MCU
```

**Environment (please complete the following information):**
 - OS: fedora29
 - Toolchain: zephyr
 - Commit SHA or Version used: d5b2834f58e11b59c640f00f3f2ca2fb85d3a48c

**Additional context**
Add any other context about the problem here.

cinlyooi-intel, 2019-02-27:
On adruino duo : Commit ID 9072d34f30

overheat, 2019-02-27:
Hi Cinlyooi,

I checked this on my Nordic boards. It works fine with the latest version of Zephyr.
Could you update your zephyr project and test it again?

The attachments are my screenshot.

[image]

overheat, 2019-02-27:
I also add 1000ms boot delay, and it works fine right.
[image]

cinlyooi-intel, 2019-02-28:
@overheat: 

I am still working on understanding the test case execution. 

This is not a straight forward test cases like other unit test because we are expecting test_wdt() to be called repeatedly. My questions are: 

(0) I am expecting the at most 8 instances of "starting test - test_wdt", i.e. a pair for each for 
- zassert_true(test_wdt_no_callback() == TC_PASS, NULL);
- zassert_true(test_wdt_callback_1() == TC_PASS, NULL);
- zassert_true(test_wdt_callback_2() == TC_PASS, NULL);
- zassert_true(test_wdt_bad_window_max() == TC_PASS, NULL);

As for the pair of instances: First one start the test, second one check the result.

However, I am getting more that 8. Is that expected?

(0.1) Will the test case automatically return, or I need to stop the test?

(1) In both console output , I noted that you did not get this line:
`***** Booting Zephyr OS v1.14.0-rc1-623-g9c2c11571670 (delayed boot 1000ms) *****`
after `*****Waiting fo restart MCU ... *****` or `**** delaying boot...*****` as appropriate. Is it important?

(2) Take the "with delay" test case: 
We keep getting alternate of  test with test_wdt_no_callback and test_wdt_callback_1, and, a test with test  test_wdt_callback_1 and _2. Is the alternation expected? I believe it is but would like to confirm.

cinlyooi-intel, 2019-03-04:
Observed in commit  fd7c4430e1398374f2f5b49bcd6ec95166e52e9a for nrf51_pca10028

For nrf51_pca10028:
```
****** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-704-gfd7c4430e139 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
****** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-704-gfd7c4430e139 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-704-gfd7c4430e139 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Testcase passed
Testcase: test_wdt_callback_1
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-704-gfd7c4430e139 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_callback_1
Testcase passed
Testcase: test_wdt_callback_2
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-704-gfd7c4430e139 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Testcase passed
Testcase: test_wdt_callback_1
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-704-gfd7c4430e139 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_callback_1
Testcase passed
Testcase: test_wdt_callback_2
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-704-gfd7c4430e139 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Testcase passed
Testcase: test_wdt_callback_1
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-704-gfd7c4430e139 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_callback_1
Testcase passed
[... 95 lines cut ...]
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Testcase passed
Testcase: test_wdt_callback_1
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-704-gfd7c4430e139 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_callback_1
Testcase passed
Testcase: test_wdt_callback_2
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-704-gfd7c4430e139 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Testcase passed
Testcase: test_wdt_callback_1
Waiting to restart MCU
```

cinlyooi-intel, 2019-03-04:
@overheat 
As part of our  routine we revisited the test cases and we are still seeing the same problem

I am still a novice at zephyr. I think the problem is in `test_wdt_callback_2()`. Main symptom is we never run next test, i.e.,  `test_wdt_bad_window_max()`.

I added a few TC_PRINT() statements to track the progress of the program and I believe the problem is in `test_wdt_callback_2()`. Placing a TC_PRINT statement inside the top of the if-block for `if (m_state == WDT_TEST_STATE_CHECK_RESET) ` shows that it was never executed, instead, it kept going to set the callback, i.e. `m_testvalue=0U` onwards.

Based on my reading of the test case, this is what we should be seeing: (empty lines are mine)
```
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-450-g7f4eb00f47 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Testcase passed

Testcase: test_wdt_callback_1
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-450-g7f4eb00f47 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_callback_1
Testcase passed

Testcase: test_wdt_callback_2
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-450-g7f4eb00f47 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt

Testcase: test_wdt_bad_window_max
...
```

but we never got to 
```
Testcase: test_wdt_callback_2
Testcase passed
```
Instead, we got into an infinite loop of 
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-450-g7f4eb00f47 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Testcase passed
Testcase: test_wdt_callback_1
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-450-g7f4eb00f47 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_callback_1
Testcase passed
Testcase: test_wdt_callback_2
Waiting to restart MCU
```

SavinayDharmappa, 2019-03-05:
@cinlyooi-intel  on which board it looping ?  x86 quark_se_c1000 ?

SavinayDharmappa, 2019-03-05:
@cinlyooi-intel  test_wdt_callback_2 . is specific all back is specific to nrf boards.

cinlyooi-intel, 2019-03-05:
@SavinayDharmappa  NRF boards that  I have at Intel, i.e. pca10028, pca10040,  and pca10056

SavinayDharmappa, 2019-03-05:
I am not aware of nrf architecture ..some who has knowledge on NRF platform should look into it.

carlescufi, 2019-03-06:
@kl-cruz could you please take a look at this failing test?

kl-cruz, 2019-03-08:
Ok, give me some time to reproduce it and dig into it

cinlyooi-intel, 2019-03-13:
Observed on an additional platform today, i.e. sam_e70_xplained (commit 6368ed4edf5c5ffbccf9f904efe836ef5ca5d84a).  Updated description

overheat, 2019-03-13:
@cinlyooi-intel 

Sorry for late, I am working on reviewing technical books, which bought almost ten years ago, for my job interview.
With this patch, everything will go well. No big issue inside kernel or driver, just test flow control issue.
Please help me to review this patch on yours boards. I will raise a pull request if it works.

`diff --git a/tests/drivers/watchdog/wdt_basic_api/src/test_wdt.c b/tests/drivers/watchdog/wdt_basic_api/src/test_wdt.c
index 482bdb48ef..4c529837b3 100644
--- a/tests/drivers/watchdog/wdt_basic_api/src/test_wdt.c
+++ b/tests/drivers/watchdog/wdt_basic_api/src/test_wdt.c
@@ -300,7 +300,7 @@ static int test_wdt_bad_window_max(void)

 void test_wdt(void)
 {
-       if (m_testcase_index != 1) {
+       if (m_testcase_index == 0) {
                zassert_true(test_wdt_no_callback() == TC_PASS, NULL);
        }
        if (m_testcase_index == 1) {
@@ -322,6 +322,7 @@ void test_wdt(void)
                m_testcase_index++;
        }
        if (m_testcase_index > 3) {
+               m_testcase_index = 0;
                m_state = WDT_TEST_STATE_IDLE;
        }
 }`

chiachuanwu, 2019-03-13:
hi @overheat please take note that ,we observed similar issued on adruino_due board as well.

overheat, 2019-03-13:
@chiachuanwu 
Could you test this patch on the adruino_due board?

SavinayDharmappa, 2019-03-13:
@overheat  m_testcase_index is variable which gets memory in noinit section. Default value of variable is not guaranteed to be zero on all platform. You patch assumes m_testcase_index default value will be zero.refer  commit id 020884

overheat, 2019-03-14:
@SavinayDharmappa Thank you for your reminder. I have updated the patch.

overheat, 2019-03-14:
@cinlyooi-intel need you to review this patch  #14466 , thanks.

chiachuanwu, 2019-03-20:
Hi @overheat  i had applied #14466 patch and tested on it, below was the result(2 board passed, 2 boards failed):
**Failed Boards:**
sam_e70_xplained
arduino_due
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1273-g3a8767c1f269 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1273-g3a8767c1f269 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1273-g3a8767c1f269 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1273-g3a8767c1f269 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1273-g3a8767c1f269 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1273-g3a8767c1f269 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1273-g3a8767c1f269 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1273-g3a8767c1f269 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1273-g3a8767c1f269 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1273-g3a8767c1f269 (delayed boot 1000ms) *****
[... 202 lines cut ...]
***** Booting Zephyr OS v1.14.0-rc1-1273-g3a8767c1f269 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
```

Passed Boards:
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1273-g3a8767c1f269 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_callback_2
Testcase passed
Testcase: test_wdt_bad_window_max
PASS - test_wdt
===================================================================
Test suite wdt_basic_test succeeded
===================================================================
RunID: :thak
PROJECT EXECUTION SUCCESSFUL
```

overheat, 2019-03-20:
@chiachuanwu glad to know some boards work. Since I don't have sam_e70_xplained and arduino_due on my hand, I can not reproduce. I will try to find such Atmel MCU based board, and back later.

overheat, 2019-03-27:
@chiachuanwu I have received my arduino_due board and reproduced this failure on it. After a bit of digging, I believe it's related to ATSAMXX chip's watchdog driver because the watchdog callback function "wdt_int_cb0" didn't be invoked. So this failure on arduino_due or sam_e70_xplained is another issue rather than my previous patched one.

@nashif  Hi, do we have any Atmel chips (ATSAMXX ) expert here? Please notice he or she check this failure on arduino_due or sam_e70_xplained board. If not, I can keep digging on it but need a longer time.

chiachuanwu, 2019-03-27:
@overheat thanks for your update.

overheat, 2019-03-28:
Ok, I will take this and try to fix it today.

overheat, 2019-03-28:
@chiachuanwu Could do test this patch on sam_e70_xplained? It works on arduino_due, that I have tested.

chiachuanwu, 2019-03-28:
@overheat , i had tested  PR#14968 , arduino_due board passed, but sam_e70_xplained board still failed, console log as below:

**arduino_due** PASSED
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Testcase passed
Testcase: test_wdt_bad_window_max
PASS - test_wdt
===================================================================
Test suite wdt_basic_test succeeded
===================================================================
RunID: :o0kp
PROJECT EXECUTION SUCCESSFUL
```
**sam_e70_xplained** (FAILED)

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
[... 103 lines cut ...]
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
```

overheat, 2019-03-29:
@cinlyooi-intel @chiachuanwu After digging deeper with GDB, I found that section .noinit be cleared during the boot process on ATSAME70, it's wrong according to line 422 in ```"zephyr\include\arch\arm\cortex_m\scripts\linker.ld"``` and ARM spec.  I think it is probably a bug inside ATSAME70 SoC itself, not Zephyr. I will open another issue to discuss it, but before that could you both give a review on #14968? I need this patch to be merged and ask other people help above that.

cinlyooi-intel, 2019-03-29:
Sorry, still seeing the issue on master  after #14466 is merged into it. Also see this when testing PR #14968 on its own

os: fedora29
toolchain: zephyr
commit: f5a4a698ec17e6a78299966a1a3fe73c3404b78d
board: sam_e70_xplained

screenshot
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
[... 189 lines cut ...]
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1515-g8fdb1b091138 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
***** delaying boot 1000ms (per build configuration) *****
```

overheat, 2019-04-05:
@cinlyooi-intel @SavinayDharmappa @chiachuanwu According to #15109, ATSAME70 don't preserve RAM when watchdog triggers, which is a kind of chip bug and also cause failure in watchdog test. So exclude this series of chips from this test.
Please review again. thanks.

ioannisg, 2019-06-30:
This does not appear to be a generic test failure. I tested that it passes on all Nordic SOCs, 51, 52, 52840 and 9160 and frdm_k64f.

It's a SAMe70 issue only. We still need to do something with the test, but since this is mostly related to the SAMe70 being peculiar, I 'm reducing the priority to low.

Feel free to comment.

PR #14466 (2019-03-28): tests/drivers/watchdog/wdt_basic_api/ behavior wrong.
tests:  watchdog:  Debug for boards enabled TEST_WDT_CALLBACK_2

The original code cannot go to the next step on those boards enabled TEST_WDT_CALLBACK_2 macro, because of the flow control issue. So the test function cannot finish, and the board keeps restart.  As a result, failure on the test.

Fixes #13468
commit f7e7f0f5 (2019-03-28): tests: watchdog: Debug for boards enabled TEST_WDT_CALLBACK_2

The original code cannot go to the next step on those boards enabled
TEST_WDT_CALLBACK_2 macro, because of the flow control issue. So the
test function cannot finish, and the board keeps restart. As a result,
failure on the test.

Fixes #13468
closed 2020-03-11 by nashif