zephyrproject-rtos/zephyr #13481: Regression in CI coverage for (at least) some Ethernet drivers after net_app code removal

pfalcon, 2019-02-18:
48abdc2cb925890213110352372946922e928497, which was part of https://github.com/zephyrproject-rtos/zephyr/pull/12855, removed echo_server (non-socket version) sample app, which was the "default" sample for networking smoke-testing. As such, it included many config overlays test configs (in sample.yaml) to cover various option combinations and drivers. All of them were lost.

Example of lost test configs are those added in commits 7117b73562ea84874929e22ec0b75d3940c7f6fe and e779f91e0cb3c93263654b2ebcb7fd4bf0dbd063 - and those only the configs added literally 10 days before they were blind-removed like that, there're probably more useful configs added over last 2 years.

Bottom line:

1. There were shared concerns that net_app removal from 1.14 is a hasty move.
2. Instead of blind-removing (`rm -rf *`-style) everything like that, test config should have been ported over the other sample(s). (With RFC regarding what becomes a new default "smoke test app").
3. Not done at right time, it needs to be done now.

pfalcon, 2019-02-18:
> Example of lost test configs are those added in commits [7117b73](https://github.com/zephyrproject-rtos/zephyr/commit/7117b73562ea84874929e22ec0b75d3940c7f6fe) and [e779f91](https://github.com/zephyrproject-rtos/zephyr/commit/e779f91e0cb3c93263654b2ebcb7fd4bf0dbd063)

This in turn caused https://github.com/zephyrproject-rtos/zephyr/issues/13484

jukkar, 2019-02-18:
Known issue, there is also #13300 for this.

jukkar, 2019-02-19:
Fixed by #13485

closed 2019-02-19 by jukkar