zephyrproject-rtos/zephyr #13484: net: (At least) eth_smsc911x driver is broken in the master

pfalcon, 2019-02-18:
ebd2edbaafe949626439b4aed6df4dbdcfc60eb5 converted the eth_smsc911x driver to new pkt API, which is great. However, the cause of conversation appears to be the desire to remove the old API, which by all accounts was a hasty act, given that the new API is, ahem, experimental, and "we yet have 2 months to stabilize it". Due to this:

1. eth_smsc911x is broken, as it contains outright typos.
2. The conversion itself is clearly subideal, e.g. it uses function calls to fill each 4 bytes of a packet, where previously one word-sized memory store was used.

p.1 sneaked into the master due to https://github.com/zephyrproject-rtos/zephyr/issues/13481. But overall, instead of hastily convert drivers like eth_smsc911x for the sole purpose of removing the old API, both the driver and old API should have been kept around, to see where the new API is still lacking comparing to the old, and need more work/optimizations.

To reproduce breakage:

1. cd samples/net/sockets/dumb_http_server/
2. mkdir build && cd build
3. cmake -DBOARD=qemu_x86 -DOVERLAY_CONFIG=overlay-smsc911x.conf ..
4. make

tbursztyka, 2019-02-18:
I specifically asked you to review/test that part of the code. You did nothing that time, but I bet you should have now.

pfalcon, 2019-02-18:
@tbursztyka, I did, found that you hosed it, so please fix it. And in general, next time, when I -1 your PR, please take time to address the concerns. Because first you ignore some points, then you remove tests, and finally you push outright broken code to master.

@nashif: FYI.

tbursztyka, 2019-02-20:
@pfalcon about broken code: 

1 - cmake -DBOARD=qemu_x86 -DOVERLAY_CONFIG=overlay-smsc911x.conf ..
2 - make

> In file included from /home/tburszty/Projects/zephyr/drivers/ethernet/eth_smsc911x.c:35:0:
/home/tburszty/Projects/zephyr/drivers/ethernet/eth_smsc911x.c: In function ‘smsc_mac_regread’:
/home/tburszty/Projects/zephyr/drivers/ethernet/eth_smsc911x_priv.h:179:33: error: ‘DT_SMSC_LAN9220_0_BASE_ADDRESS’ undeclared (first use in this function)
 #define SMSC9220_BASE           DT_SMSC_LAN9220_0_BASE_ADDRESS
                                 ^
/home/tburszty/Projects/zephyr/drivers/ethernet/eth_smsc911x_priv.h:182:63: note: in expansion of macro ‘SMSC9220_BASE’
 #define SMSC9220                ((volatile SMSC9220_TypeDef *)SMSC9220_BASE)
                                                               ^~~~~~~~~~~~~
/home/tburszty/Projects/zephyr/drivers/ethernet/eth_smsc911x.c:58:2: note: in expansion of macro ‘SMSC9220’
  SMSC9220->MAC_CSR_CMD = cmd;
  ^~~~~~~~
/home/tburszty/Projects/zephyr/drivers/ethernet/eth_smsc911x_priv.h:179:33: note: each undeclared identifier is reported only once for each function it appears in
 #define SMSC9220_BASE           DT_SMSC_LAN9220_0_BASE_ADDRESS
                                 ^
/home/tburszty/Projects/zephyr/drivers/ethernet/eth_smsc911x_priv.h:182:63: note: in expansion of macro ‘SMSC9220_BASE’
 #define SMSC9220                ((volatile SMSC9220_TypeDef *)SMSC9220_BASE)
                                                               ^~~~~~~~~~~~~
/home/tburszty/Projects/zephyr/drivers/ethernet/eth_smsc911x.c:58:2: note: in expansion of macro ‘SMSC9220’
  SMSC9220->MAC_CSR_CMD = cmd;
  ^~~~~~~~
/home/tburszty/Projects/zephyr/drivers/ethernet/eth_smsc911x.c: In function ‘smsc_mac_regwrite’:
/home/tburszty/Projects/zephyr/drivers/ethernet/eth_smsc911x_priv.h:179:33: error: ‘DT_SMSC_LAN9220_0_BASE_ADDRESS’ undeclared (first use in this function)
 #define SMSC9220_BASE           DT_SMSC_LAN9220_0_BASE_ADDRESS
                                 ^
/home/tburszty/Projects/zephyr/drivers/ethernet/eth_smsc911x_priv.h:182:63: note: in expansion of macro ‘SMSC9220_BASE’
 #define SMSC9220                ((volatile SMSC9220_TypeDef *)SMSC9220_BASE)
                                                               ^~~~~~~~~~~~~
/home/tburszty/Projects/zephyr/drivers/ethernet/eth_smsc911x.c:72:2: note: in expansion of macro ‘SMSC9220’
  SMSC9220->MAC_CSR_DATA = val;
  ^~~~~~~~

commit f01f2781 (2019-02-21): drivers/ethernet: Fix net API usage in smsc911x driver

Fixes #13484
closed 2019-02-21 by nashif