zephyrproject-rtos/zephyr #13489: frdm_k64f test/net/tcp bus fault after test ends

hakehuang, 2019-02-18:
**Describe the bug**
meet bus fault after the test ends

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdm_k64f ..
3. make
4. See error

**Expected behavior**
test ends normally

**Impact**
annoyance

**Screenshots or console output**
```
***** Booting Zephyr OS 1.14.0-rc1 *****

starting test - test TCP init

passed

starting test - test TCP register/unregister port cb

[00:00:00.000,256] [1;33m<wrn> net_if: You have 2 IPv4 net_if addresses but 3 network interfaces[0m

[00:00:00.000,261] [1;33m<wrn> net_if: Consider increasing CONFIG_NET_IF_MAX_IPV4_COUNT value.[0m

[00:00:00.000,266] [1;33m<wrn> net_if: You have 2 IPv6 net_if addresses but 3 network interfaces[0m

[00:00:00.000,270] [1;33m<wrn> net_if: Consider increasing CONFIG_NET_IF_MAX_IPV6_COUNT value.[0m

passed

starting test - test TCP context init

passed

starting test - test IPv6 TCP reset packet creation

passed

starting test - test IPv4 TCP reset packet creation

passed

starting test - test IPv6 TCP syn packet creation

passed

starting test - test IPv4 TCP syn packet creation

passed

starting test - test IPv6 TCP synack packet create

passed

starting test - test IPv4 TCP synack packet create

passed

starting test - test IPv6 TCP fin packet creation

passed

starting test - test IPv4 TCP fin packet creation

passed

starting test - test IPv6 TCP seq check

passed

starting test - test IPv4 TCP seq check

passed

starting test - test TCP seq validity

passed

starting test - test TCP reply context init

passed

starting test - test TCP accept init

passed

starting test - test cleanup

passed

===================================================================

PROJECT EXECUTION SUCCESSFUL

[00:00:02.530,633] [1;31m<err> net_conn: Address families different[0m

***** BUS FAULT *****

  Imprecise data bus error

***** Hardware exception *****

Current thread ID = 0x200042ec

Faulting instruction address = 0xd8d0

Fatal fault in thread 0x200042ec! Aborting.
```

**Environment (please complete the following information):**
 - OS:  Linux
 - Toolchain Zephyr SDK
 - Commit v1.14.0_rc1

hakehuang, 2019-02-18:
@MaureenHelm

ioannisg, 2019-02-20:
I confirm that I can re-produce this one. Looking at it

tbursztyka, 2019-02-20:
@ioannisg can you try with PR #13545 ?

ioannisg, 2019-02-20:
@tbursztyka I will do. Actually, I noticed that I get a few "swap's", (i.e. context-switches) after I get the PROJECT EXECUTION SUCCESSFUL, so I suspected, some thread is still running.

ioannisg, 2019-02-20:
```
[00:00:02.530,645] <err> net_conn: Address families different
***** Booting Zephyr OS v1.14.0-rc1-355-gdb2f8ddae9 *****
starting test - test TCP init
passed
starting test - test TCP register/unregister port cb
[00:00:00.000,256] <wrn> net_if: You have 2 IPv4 net_if addresses but 3 network interfaces
[00:00:00.000,260] <wrn> net_if: Consider increasing CONFIG_NET_IF_MAX_IPV4_COUNT value.
[00:00:00.000,265] <wrn> net_if: You have 2 IPv6 net_if addresses but 3 network interfaces
[00:00:00.000,269] <wrn> net_if: Consider increasing CONFIG_NET_IF_MAX_IPV6_COUNT value.
passed
starting test - test TCP context init
passed
starting test - test IPv6 TCP reset packet creation
passed
starting test - test IPv4 TCP reset packet creation
passed
starting test - test IPv6 TCP syn packet creation
passed
starting test - test IPv4 TCP syn packet creation
passed
starting test - test IPv6 TCP synack packet create
passed
starting test - test IPv4 TCP synack packet create
passed
starting test - test IPv6 TCP fin packet creation
passed
starting test - test IPv4 TCP fin packet creation
passed
starting test - test IPv6 TCP seq check
passed
starting test - test IPv4 TCP seq check
passed
starting test - test TCP seq validity
passed
starting test - test TCP reply context init
passed
starting test - test TCP accept init
passed
starting test - test cleanup
passed
===================================================================
PROJECT EXECUTION SUCCESSFUL
[00:00:02.530,645] <err> net_conn: Address families different

That's what I get after pulling the changes from #13545, @tbursztyka, @hakehuang . There's a logger warning, but the BusFault has disappeared. I suppose you can add a "Fixes #13489" in that PR.

```

tbursztyka, 2019-02-20:
Thanks for verifying this

hakehuang, 2019-02-20:
Hi @ioannisg  after applying #13545 I get the same result as yours.

ioannisg, 2019-02-20:
@hakehuang could you comment, also in #13545 ?

closed 2019-02-21 by nashif

hakehuang, 2019-02-25:
@ioannisg  add comment on #13545.

commit 45df09c0 (2019-02-21): net: tcp: Clean up FIN send on socket shutdown

The sending of the final FIN message was being put on the sent_list
to be handled by the retry logic and being sent directly. But in
the case of a socket that never had a connection, the logic after
the direct send would decrement the reference on the packet which
would cause the retry logic to eventually have a situation where
the pkt is on the sent_list queue but the buffer associated with
the packet is freed back to slab. The code would then get a null
pointer to the tcp header and would fault when setting the crc to
zero on frdm-k64f platform [... cut]