zephyrproject-rtos/zephyr #13502: tests/benchmarks/timing_info: Output only consist of Delay Boot Banner

cinlyooi-intel, 2019-02-19:
**Describe the bug**
There is no output at all for quark_se_c1000_ss_devboard:arc.

Eliminated issue with the board size if I switch the same board to build for  x86 architecture I have the output I am expecting, see "Expected Behaviour"

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=quark_se_c1000_ss_devboard:arc -DARCH=arc ..
3. make
4. make run

**Screenshots or console output**
```
***** delaying boot 1000ms (per build configuration) *****
```

**Expected behavior**
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-266-g4c21bab620 (delayed boot 1000ms) *****
starting test - Time Measurement
Timing Results: Clock Frequency: 32 MHz
Interrupt latency                            :  48 cycles ,  1500 ns
Tick overhead                                : 540 cycles , 16875 ns
Thread Creation                              : 458 cycles , 14312 ns
Thread cancel                                : 274 cycles ,  8562 ns
Thread abort                                 : 478 cycles , 14937 ns
Thread Suspend                               : 479 cycles , 14968 ns
Thread Resume                                : 441 cycles , 13781 ns
Thread Yield                                 : 399 cycles , 12468 ns
Thread Sleep                                 : 666 cycles , 20812 ns
Heap Malloc                                  :1300 cycles , 40625 ns
Heap Free                                    :1436 cycles , 44875 ns
Semaphore Take with context switch           : 706 cycles , 22062 ns
Semaphore Give with context switch           : 584 cycles , 18250 ns
Semaphore Take without context switch        :  96 cycles ,  3000 ns
Semaphore Give without context switch        : 222 cycles ,  6937 ns
Mutex lock                                   : 263 cycles ,  8218 ns
Mutex unlock                                 : 333 cycles , 10406 ns
Message Queue Put with context switch        : 671 cycles , 20968 ns
Message Queue Put without context switch     : 258 cycles ,  8062 ns
Message Queue get with context switch        : 797 cycles , 24906 ns
Message Queue get without context switch     : 256 cycles ,  8000 ns
MailBox synchronous put                      :1126 cycles , 35187 ns
MailBox synchronous get                      : 813 cycles , 25406 ns
MailBox asynchronous put                     : 465 cycles , 14531 ns
MailBox get without context switch           : 719 cycles , 22468 ns
Timing Measurement  finished
PASS - main
===================================================================
```

**Environment (please complete the following information):**
 - OS: Fedora29
 - Toolchain: zephyr
 - Commit SHA or Version used: cf94d6811aeb6cf345c678862729ecb1e91fc272

AdithyaBaglody, 2019-02-19:
@cinlyooi-intel For C1000 the procedure is to flash a hello world in the x86 core and then flash the benchmarking on arc. was this followed?

cinlyooi-intel, 2019-02-20:
@AdithyaBaglody I compiled hello word  using ARCH=x86, then flashed and run it. Then I compiled the benchmark using ARCH=arc, then flashed and run the benchmarking. The result is the same.

Is this equivalent to the sequence you are expecting?

AdithyaBaglody, 2019-02-20:
@cinlyooi-intel 
use cmake -DBOARD=quark_se_c1000_devboard .. to flash hello world
use cmake -DBOARD=quark_se_c1000_ss_devboard .. to flash the timing info

ManojSubbarao, 2019-02-25:
@cinlyooi-intel  please re-check

AdithyaBaglody, 2019-02-25:
Checked this on c1000. and I am able to get the correct output. 
Unable to reproduce this issue.

ManojSubbarao, 2019-02-26:
cannot reproduce, closing the issue

closed 2019-02-26 by ManojSubbarao