zephyrproject-rtos/zephyr #13514: #stm32 creating #gpio #interrupts on 2+ pins with the same pin number failes

StefJar, 2019-02-19:
In my case:

I use PC8 and PA8 as inputs for interrupt generation.

my setup:

```
#define SENSOR_LSM6DS3_INT1	sensor.gpioC,8
#define SENSOR_LSM6DS3_INT2	sensor.gpioA,8

typedef struct sensor_control {
	struct device * gpioA;
	struct device * gpioC;

	struct gpio_callback gpio_cb_acc_int1;
	struct gpio_callback gpio_cb_acc_int2;

} sensor_control_t;

static sensor_control_t sensor;

sensor_error_t sensor_init(void) {
	int ret;

	memset(&sensor, 0, sizeof(sensor));
	// gpio init
	sensor.gpioA = device_get_binding("GPIOA");
	if (!sensor.gpioA) {
		LOG_ERR("Cannot find GPIOA!");
		return sensor_error_sensor_GPIO;
	}

	sensor.gpioC = device_get_binding("GPIOC");
	if (!sensor.gpioC) {
		LOG_ERR("Cannot find GPIOC!");
		return sensor_error_sensor_GPIO;
	}
	// set callbacks

	// int 1
	ret = gpio_pin_configure(SENSOR_LSM6DS3_INT1,
			GPIO_DIR_IN | // input, normal polarity, no pull up/down
			GPIO_INT | GPIO_INT_EDGE | GPIO_INT_ACTIVE_HIGH  // interrupt , edge detection
		);
	if (ret) {
		LOG_ERR("configure MOV_INT1 pin failed");
		return sensor_error_sensor_GPIO;
	}
	// int 2
	ret = gpio_pin_configure(SENSOR_LSM6DS3_INT2,
			GPIO_DIR_IN | // input, normal polarity, no pull up/down
			GPIO_INT | GPIO_INT_EDGE | GPIO_INT_ACTIVE_HIGH // interrupt , edge detection
		);
	if (ret) {
		LOG_ERR("configure MOV_INT2 pin failed");
		return sensor_error_sensor_GPIO;
	}
	gpio_init_callback(&sensor.gpio_cb_acc_int2, sensor_acc_int2, BIT(8));

	ret = gpio_add_callback(sensor.gpioA, &sensor.gpio_cb_acc_int2);
	if (ret) {
		LOG_ERR("adding callback for MOV_INT2 pin failed");
		return sensor_error_sensor_GPIO;
	}

	ret = gpio_pin_enable_callback(SENSOR_LSM6DS3_INT2);
	if (ret) {
		LOG_ERR("enable callback for MOV_INT2 pin failed");
		return sensor_error_sensor_GPIO;
	}

```

the second call at `gpio_pin_configure` fails in `stm32_exti_set_callback`.
this function is called form 'gpio_stm32_config'.

The bug is that pin number is assumed to be the external interrupt line number. 

In this case:
```
gpio_pin_configure(GPIOC, 8, ...)
gpio_stm32_config(GPIOC, .., pin=8, ...)
		if (stm32_exti_set_callback(pin, cfg->port, gpio_stm32_isr, dev) != 0) {
			return -EBUSY;
		}
int stm32_exti_set_callback(line=8, port=2, ..., ...)
{
	struct device *dev = DEVICE_GET(exti_stm32);
	struct stm32_exti_data *data = dev->driver_data;

	if (data->cb[line].cb) {
		return -EBUSY;
	}
...
gpio_pin_configure(GPIOA, 8, ...)
gpio_stm32_config(GPIOA, .., pin=8, ...)
		if (stm32_exti_set_callback(pin, cfg->port, gpio_stm32_isr, dev) != 0) {
			return -EBUSY;
		}
int stm32_exti_set_callback(line=8, port=2, ..., ...)
{
	struct device *dev = DEVICE_GET(exti_stm32);
	struct stm32_exti_data *data = dev->driver_data;

	if (data->cb[line].cb) {
		return -EBUSY; <<< fail because line 8 is already taken by PC8
	}
```
Think somehow the external interrupt line number should be a parameter or the line must be calculated by a formula like line = pin + portOffset[port]

erwango, 2019-02-19:
@StefJar, this is actually the way it works from HW point of view.
If you look at the Reference Manual of your SoC, section EXTI or Interrupt and events, you'll find a Figure named "External interrupt/event GPIO mapping" on which you could see that all pins with the same number (ex PAy, PBy ...PHy) are using the same EXTIy line.
So, at a defined time, you can't program 2 pins with the same number to use different callbacks.

StefJar, 2019-02-19:
True
this means that the scope changes form a pin based approach to an ext intr. based one. Think the intr should not be enabled on gpio base. Instead it should be enabled at the intr. controller.
other possibility is to implement some logic which allows to enable different interrupt sources. This function fails, if different isr handler are defined. Means the  "if (data->cb[line].cb)" is changed to "if ((data->cb[line].cb) && (data->cb[line].cb != cb))"

erwango, 2019-02-19:
What you expect is not fully clear to me, can you clarify?
Anyhow, do you agree to move this point from bug to enhancement (of which we can discuss) ?
We're currently in RC phase of 1.14 delivery, issues tagged as "bugs" are scrutinized..

StefJar, 2019-02-19:
actually I can not enable the interrupts for PA8 after PC8 because gpio_pin_configure fails. it fails because after calling it on PC8 the isr callback is set. This prevents PA8 configured as source of interrupts.
My propose for a quick fix of the "stm32_exti_set_callback" function. This fix is now take into account, that several pin can share one interrupt. We need to make sure that all pin sources are directing to the same callback. This means:
```
int stm32_exti_set_callback(int line, int port, stm32_exti_callback_t cb, void *data)
{
	struct device *dev = DEVICE_GET(exti_stm32);
	struct stm32_exti_data *data = dev->driver_data;

>>>	if ((data->cb[line].cb)&& (data->cb[line].cb != cb)){
		return -EBUSY; <<< fail because line 8 is already taken by PC8
	}
```

erwango, 2019-02-20:
>  that several pin can share one interrupt.

This is not actually possible.
For each line (so group of pins), there is only one port that could be programmed as source of interrupt at a time. So several pins can't share one interrupt at a given moment.
You program them alternatively but you need to call stm32_exti_unset_callback before programming the second pin.

StefJar, 2019-02-20:
to summarize the proposed workaround for the example:

```
gpio_pin_configure(GPIOC, 8, ...)
stm32_exti_unset_callback(8)
gpio_pin_configure(GPIOA, 8, ...)
```

erwango, 2019-02-20:
Exactly. Would this work for your use case?
This being said, we should find a way that fits gpio API to do that. And this might not be available right now.

StefJar, 2019-02-20:
I checked the stm32f412 datasheet. In this case it is only allowed to configure one source of the ext interrupt. In that context the current implementation is correctly reflecting the MCU behavior.
I guess this behavior applies to the other arm/stm MCUs as well.
In my eyes this is not a bug and so I am closing it.

closed 2019-02-20 by StefJar