zephyrproject-rtos/zephyr #13519: BLE Split Link Layer Improvements

carlescufi, 2019-02-19:
This issue lists all the improvements we want to include that are not part of bringing the Split Link Layer up to parity with the existing one.

- [ ] Add documentation for ticker
- [ ] Include the diagrams done by @pskhansen in the documentation

aescolar, 2019-04-25:
"- [ ] Use `atomic_cas` when supported by the arch without disabling interrupts ([see for more info](https://github.com/zephyrproject-rtos/zephyr/pull/12404#pullrequestreview-192102680))"
 was removed due to https://github.com/zephyrproject-rtos/zephyr/pull/13471 being closed

aescolar, 2019-04-25:
"- [ ] Add support for (current) Mesh Extensions (#14678)"
removed as there was no real interest for that, and the ticket was morphed into adding full mesh extensions to the controller.

cfriedt, 2020-02-11:
Ooooo - I like this! It might also be a good idea to add a basic / minimum recommendation and documentation for the radio API. Naturally, most chips will have a different radio, but we are *mostly* using the same interface so far. However, that interface is largely undocumented. Some state diagrams would be nice as well.

carlescufi, 2020-02-12:
> Ooooo - I like this! It might also be a good idea to add a basic / minimum recommendation and documentation for the radio API. Naturally, most chips will have a different radio, but we are _mostly_ [...]

Just added a point at the end, thanks!

joerchan, 2020-07-23:
> Add tests in nrf52_bsim to run on upstream CI for (at least basic) controller functionality

Isn't this done?

kruithofa, 2022-08-25:
@carlescufi we close this issue since realistically no work will be done on this

closed 2022-08-25 by kruithofa

KlemenDEV, 2023-10-04:
> realistically no work will be done on this

Will the split LL ever be "official", or should we use the old system?

cvinayak, 2023-10-04:
> > realistically no work will be done on this
> 
> Will the split LL ever be "official", or should we use the old system?

Split Link Layer is the default upstream since Zephyr v2.0.0-rc1 , see https://github.com/zephyrproject-rtos/zephyr/commit/6cf26b8fe63afdb83817ad551a3f03c40f744a72

KlemenDEV, 2023-10-04:
I see, thanks!