zephyrproject-rtos/zephyr #13522: BT subscribe to characteristic for Indication results in kernel crash

lanvis, 2019-02-19:
After discovery of primary service and characteristics, subscribing to a characteristic with Notify property results in Assert and kernel crash.

I discover this characteristic with Indicate  property.
characteristic discovered attributes:
    [char ATTRIBUTE] handle 19
    char uuid :  a5 4e 6a f8 f1 30 94 c5 1c 53 9e 84 35 35 08 f8
    Properties: RD:WR:I:
    end handle: 21

There is only an example of registration for a notify in the samples but I assume that the Indicate uses the same subscribe structure: bt_gatt_subscribe_params to subscribe Notifies.
   
Code to subscribe:
   uint16_t auth_desc_handle = 0;
   struct bt_gatt_subscribe_params  auth_subscribe_req = {0};

   if ( (auth_desc_handle = get_descriptor_handle(BLE_UUID_DXCM_AUTH)) != 0)
      {
            auth_subscribe_req.notify = cb_indicate_DXCM_AUTH;
            auth_subscribe_req.value = BT_GATT_CCC_INDICATE;
            auth_subscribe_req.ccc_handle = auth_desc_handle;
            auth_subscribe_req.value_handle = get_handle_value(BLE_UUID_DXCM_AUTH);

            if ( (err = bt_gatt_subscribe(conn, &auth_subscribe_req)) >= 0)
            {
              printk(" .. Auth Service Registered\n");
              return(true);
            }
      }

bt_gatt_subscribe() returns success (0) but the kernel Asserts a short time later:
[00:00:50.473,449] <err> bt_ctlr_llsw_ctrl: assert: '_radio.ticker_id_prepare == (7 + conn->handle)' failed

I have traced the bug: att_write_ccc()->gatt_send->bt_att_req_send->att_send_req->bt_l2cap_send_cb..
and the message is successfully sent but I never get the call back.

Faulting instruction is 19bea which via my map file is in rx_enqueue:
 .text.packet_rx_enqueue       0x0000000000011910       0xac 
                                            subsys/bluetooth/controller/libsubsys__bluetooth__controller.a(ctrl.c.obj)
 .text.isr_scan_tgta_check       0x00000000000119bc       0x98 
                                            subsys/bluetooth/controller/libsubsys__bluetooth__controller.a(ctrl.c.obj)

Other info:
writes to attributes work fine.

**To Reproduce**
1. Build Zephyr as a BLE central.
2. Discovery a profile containing an Indicate characteristic.
3. Attempt to subscribe to the Indicate CCC via the bt_gatt_subscribe() API. 
4. See kernel ASSERT after successful bt_gatt_subscribe() return.

**Expected behavior*

**Impact**
I cannot continue to develop my custom profile as it depends on Indicate responses.

**Screenshots or console output**
Attached

**Environment (please complete the following information):**
OS: Ubuntu 16.4
Toolchain: Zephyr SDK
Version: OS 1.13.99

[zephyr.map.txt](https://github.com/zephyrproject-rtos/zephyr/files/2880037/zephyr.map.txt)
[errputty.log.txt](https://github.com/zephyrproject-rtos/zephyr/files/2880039/errputty.log.txt)

lanvis, 2019-02-19:
Board = nrf52840_pca10056

lanvis, 2019-02-20:
Connected:
   interval = 40
   latency  = 0
   dest     = ed:e0:59:fe:70:5a (random)
   src      = c4:4e:36:ef:2f:f8 (random)

lanvis, 2019-02-20:
More Info after another crash with hci debug turned on: 

Subscribe: .notify=0x00000f31, .value=2, .ccc_handle=21, .value_handle=20
 .. Auth Service Registered
[00:00:26.721,008] <dbg> bt_ctlr_hci_driver.hci_driver_send: enter
[00:00:26.721,008] <dbg> bt_ctlr_hci_driver.hci_driver_send: exit: 0
[00:00:26.755,920] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem taken
[00:00:26.755,950] <dbg> bt_ctlr_hci_driver.prio_recv_thread: Num Complete: 0x0000:1
[00:00:26.755,981] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem take...
[00:00:26.805,877] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem taken
[00:00:26.805,908] <dbg> bt_ctlr_hci_driver.prio_recv_thread: RX node enqueue
[00:00:26.805,908] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem take...
[00:00:26.805,938] <dbg> bt_ctlr_hci_driver.recv_thread: unblocked
[00:00:26.805,938] <dbg> bt_ctlr_hci_driver.recv_thread: Packet in: type:3 len:9
***** BUS FAULT *****
  Precise data bus error
  BFAR Address: 0xffffffff
***** Hardware exception *****
Current thread ID = 0x20000cc0
Faulting instruction address = 0x19fae
Fatal fault in thread 0x20000cc0! Aborting.
[00:00:29.005,828] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem taken
[00:00:29.012,908] <dbg> bt_ctlr_hci_driver.prio_recv_thread: RX node enqueue
[00:00:29.020,477] <dbg> bt_ctlr_hci_driver.prio_recv_thread: sem take...

**===traced 19fae in .lst file ::::: fault in net_buf_unref()**

void net_buf_unref(struct net_buf *buf)
#endif
{
   19fa0:       b5f8            push    {r3, r4, r5, r6, r7, lr}
   19fa2:       4d15            ldr     r5, [pc, #84]   ; (19ff8 <net_buf_unref+0x58>)
   19fa4:       4604            mov     r4, r0
                        return;
                }

                if (buf->__buf) {
                        data_unref(buf, buf->__buf);
                        buf->__buf = NULL;
   19fa6:       2600            movs    r6, #0
        while (buf) {
   19fa8:       b904            cbnz    r4, 19fac <net_buf_unref+0xc>
                        net_buf_destroy(buf);
                }

                buf = frags;
        }
}
   19faa:       bdf8            pop     {r3, r4, r5, r6, r7, pc}
                if (--buf->ref > 0) {
   19fac:       7923            ldrb    r3, [r4, #4]
                struct net_buf *frags = buf->frags;
   19fae:       6827            ldr     r7, [r4, #0]
                if (--buf->ref > 0) {
   19fb0:       3b01            subs    r3, #1
   19fb2:       b2db            uxtb    r3, r3
   19fb4:       7123            strb    r3, [r4, #4]
   19fb6:       2b00            cmp     r3, #0

lanvis, 2019-03-14:
Some further investigation revealed that the issue is not only on subscribe but on writes (which makes sense as subscribe is really just a write to a ccc).  Turning on debug and adding additional prints I am seeing null pointers appear in bt_recv() although every failure is not associated with this.  bt_recv() is always the crash point (although I am still testing this). 

In all cases a sniffer trace shows the write sent successfully and the write response received but the application never gets the write response.  I have varied the connection intervals as one mentioned that this may effect the interrupt timing but I still see the same issue.  

Here is a trace from code with HCI_core on.  It starts from the end of discovery, then does subscribe to handle 21, the does a write to handle 20.

ar uuid from userdat:x20003f18
 a5 4e 6a f8 f1 30 94 c5 1c 53 9e 84 37 35 08 f819
Properties:RD: userdat:
buf 0x20004044 (old) ref 1 pool_id 0 35 35 08 f8
;31m---buf = 20003f18
 6 messages dropped ---f 1 pool_id 0
[00:00:14.420,959] <dbg> bt_hci_coCharacterisc Discover Complete: Found 5 characteristics
3533:start13:stop:15dbg> bt_hci_core.bt_recv: buf 0x20003f18 len 31
3534:start16:stop:18
m_auth = 0,15,14,3535

3535:start19:stop:21
3536:start22:stop:24
3537:start25:stop:26
re.process_events: ev->state 4
--- 20 messages dropped ---
[00:00:14.430,084] <dbg> bt_hci_core.hci_num_completed_packets: handle 0 count 1
[00:00:14.430,114] <dbg> bt_hci_core.process_events: count 4
[00:00:14.430,114] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:14.430,145] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:14.430,145] <dbg> bt_hci_core.process_events: ev->state 4
[00:00:14.430,175] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:14.430,175] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 4 events
[00:00:14.436,431] <dbg> bt_hci_core.bt_recv: buf 0x20003f18 len 31
[00:00:14.436,431] <dbg> bt_hci_core.hci_acl: buf 0x20003f18
[00:00:14.436,462] <dbg> bt_hci_core.hci_acl: handle 0 len 27 flags 2
[00:00:14.450,378] <dbg> bt_hci_core.process_events: count 4
[00:00:14.450,408] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:14.450,408] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:14.450,439] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:14.450,439] <dbg> bt_Discovering Descriptors
..start =12, stop =26
buf 0x20004044 (old) ref 1 pool_id 0
..continue
hci_core.process_events: ev->state 4
[00:00:14.450,469buf = 20003f18
] <dbg> bt_hci_core.bt_send: buf 0x20004044 len 15 type 2
[buf 0x20004044 (old) ref 1 pool_id 0
00:00:14.450,592] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 4 events
--- 9 messages dropped ---
[uf = 20003f18
00:00:14.458,801] <dbg> bt_hci_core.hci_acl: buf 0x20003f18
buf 0x20004044 (old) ref 1 pool_id 0
--- 17 messages dropped ---
[0buf = 20003f18
[Descriptor ATTRIBUTE] handle 15re.hci_acl: buf 0x20003f18
..sync desc handle = 15
..exit
buf 0x20004044 (old) ref 1 pool_id 0

--- 17 messages dropped ---
[00:00:14.623,931] uf = 20003f18
0m<dbg> bt_hci_core.hci_acl: buf 0x20003f18
--- 6 messagebuf 0x20004044 (old) ref 1 pool_id 0
s dropped ---
[00:00:14.638,824] buf = 20003f18
<dbg> bt_hci_core.bt_recv: buf 0x20003f18 len 18
--- 17[Descriptor ATTRIBUTE] handle 18
..cgm desc handle = 18
..exit
buf 0x20004044 (old) ref 1 pool_id 0
 messages dropped ---
[00:00:14.661,407] <dbg> bt_hci_core.bt_recv: buf 0x20003f18 len 28
--- 10 messages dropped ---
[... 74 lines cut ...]
buf 0x20004044 (old) ref 1 pool_id 0
buf = 20003f18
[00:00:15.263,366] <dbg> bt_hci_core.process_events: count 4[00:00:15.263,366] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:15.263,366] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:15.263,366] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:15.263,366] <dbg> bt_hci_core.process_events: ev->state 4
[00:00:15.263,397] <dbg> bt_hci_core.bt_send: buf 0x20004044 len 21 type 2
[00:00:15.263,397] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 4 events
[00:00:15.277,709] <dbg> bt_hci_core.hci_num_completed_packets: num_handles 1
[00:00:15.277,709] <dbg> bt_hci_core.hci_num_completed_packets: handle 0 count 1
[00:00:15.277,740] <dbg> bt_hci_core.process_events: count 4
[00:00:15.277,770] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:15.277,770] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:15.277,770] <dbg> bt_hci_core.process_events: ev->state 4
[00:00:15.277,801] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:15.277,801] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 4 events
[00:00:15.283,752] <dbg> bt_hci_core.bt_recv: buf 0x20003f18 len 9
[00:00:15.283,752] <dbg> bt_hci_core.hci_acl: buf 0x20003f18
[00:00:15.283,752] <dbg> bt_hci_core.hci_acl: handle 0 len 5 flags 2
***** MPU FAULT *****
  Instruction Access Violation
***** Hardware exception *****
Current thread ID = 0x20000e9c
Faulting instruction address = 0xe000ed00
Fatal fault in thread 0x20000e9c! Aborting.

lanvis, 2019-03-14:
More info .. looks like the chan->ops->recv(chan, buf)  is the issue as this is where the crash occurs.  It never returns from that call.   I added debugging to show the address of the call and the parameters.  They look identical to the previous call ???  

========

. Auth Service Registered
buf = 20003f18
[00:00:18.392,486] <dbg> bt_hci_core.process_events: count 4
[00:00:18.392,486] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:18.392,486] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:18.392,517] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:18.392,517] <dbg> bt_hci_core.process_events: ev->state 4
[00:00:18.392,517] <dbg> bt_hci_core.bt_send: buf 0x20004044 len 13 type 2
[00:00:18.392,547] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 4 events
[00:00:18.403,411] <dbg> bt_hci_core.hci_num_completed_packets: num_handles 1
[00:00:18.403,411] <dbg> bt_hci_core.hci_num_completed_packets: handle 0 count 1
[00:00:18.403,442] <dbg> bt_hci_core.process_events: count 4
[00:00:18.403,472] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:18.403,472] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:18.403,472] <dbg> bt_hci_core.process_events: ev->state 4
[00:00:18.403,472] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:18.403,503] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 4 events
[00:00:18.409,515] <wrn> bt_hci_core: ..bt_recv

[00:00:18.409,515] <dbg> bt_hci_core.bt_recv: buf 0x20003f18 len 9
[00:00:18.409,515] <dbg> bt_hci_core.hci_acl: buf 0x20003f18
[00:00:18.409,545] <dbg> bt_hci_core.hci_acl: handle 0 len 5 flags 2
[00:00:18.409,545] <wrn> bt_hci_core: ..bt_conn_index
[00:00:18.409,545] <wrn> bt_hci_core: ..bt_conn_receive
[00:00:18.409,545] <wrn> bt_conn: Successfully parsed 5 byte L2CAP packet
**[00:00:18.409,545] <wrn> bt_l2cap: ...chan->ops->recv (0000bcd1)(20000924, 20003f18)**

[00:00:18.421,722] <wrn> bt_l2cap: ...chan->ops recv ret
[00:00:18.421,722] <wrn> bt_hci_core: ..bt_out
challenge_buffer[0] = 0xbe
challenge_buffer[1] = 0x3d
challenge_buffer[2] = 0x63
challenge_buffer[3] = 0x52
challenge_buffer[4] = 0xf6
challenge_buffer[5] = 0x24
challenge_buffer[6] = 0x8b
challenge_buffer[7] = 0x37
..write conn 200007ec
buf 0x20004044 (old) ref 1 pool_id 0
buf = 20003f18
[00:00:18.626,617] <dbg> bt_hci_core.process_events: count 4
[00:00:18.626,617] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:18.626,617] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:18.626,617] <dbg> bt_hci_core.process_events: ev->staa[00:00:18.626,617] <dbg> bt_hci_core.process_events: ev->state 4
[00:00:18.626,678] <dbg> bt_hci_core.bt_send: buf 0x20004044 len 21 type 2
[00:00:18.626,678] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 4 events
[00:00:18.635,986] <dbg> bt_hci_core.hci_num_completed_packets: num_handles 1
[00:00:18.635,986] <dbg> bt_hci_core.hci_num_completed_packets: handle 0 count 1
[00:00:18.636,016] <dbg> bt_hci_core.process_events: count 4
[00:00:18.636,016] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:18.636,047] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:18.636,047] <dbg> bt_hci_core.process_events: ev->state 4
[00:00:18.636,047] <dbg> bt_hci_core.process_events: ev->state 0
[00:00:18.636,077] <dbg> bt_hci_core.hci_tx_thread: Calling k_poll with 4 events
[00:00:18.642,028] <wrn> bt_hci_core: ..bt_recv

[00:00:18.642,028] <dbg> bt_hci_core.bt_recv: buf 0x20003f18 len 9
[00:00:18.642,028] <dbg> bt_hci_core.hci_acl: buf 0x20003f18
[00:00:18.642,028] <dbg> bt_hci_core.hci_acl: handle 0 len 5 flags 2
[00:00:18.642,028] <wrn> bt_hci_core: ..bt_conn_index
[00:00:18.642,028] <wrn> bt_hci_core: ..bt_conn_receive
[00:00:18.642,059] <wrn> bt_conn: Successfully parsed 5 byte L2CAP packet
**[00:00:18.642,059] <wrn> bt_l2cap: ...chan->ops->recv (0000bcd1)(20000924, 20003f18)**

***** MPU FAULT *****
  Instruction Access Violation
***** Hardware exception *****
Current thread ID = 0x20000e9c
Faulting instruction address = 0xe000ed00
Fatal fault in thread 0x20000e9c! Aborting.

lanvis, 2019-03-14:
FYI:  0000bcd1 is bt_att_recv (prev post)

lanvis, 2019-03-14:
A scan of the the transactions is attached.

[image]

joerchan, 2019-03-15:
@lanvis Thank you for all the information. I'm investigating this issue now.

lanvis, 2019-03-15:
Attached is a fresh trace and and associated map file in case you need to look at associated addresses.

[MapTrace0315Bug.zip](https://github.com/zephyrproject-rtos/zephyr/files/2972025/MapTrace0315Bug.zip)

joerchan, 2019-03-19:
@lanvis I'm not able to reproduce this issue with the samples in (tried central_hr)
Can you provide me with something that reproduces the issue? Either using one of the samples or the source code you have.
Also could you please provide the commit SHA you are on (1.13.99 is not a tag).

lanvis, 2019-03-19:
There are only 2 central samples and only 1 with 2-way data traffic.  I too started with central_hr.  There is no indicate example.   It bothers me that the  bt_gatt_subscribe_params struct only has a notify member (no indicate) so I am using the notify member to register my indicate. 

The crash occurs either on the subscribe response and/or a subsequent indicate resulting from a write.  

The peripheral I am connecting too is custom and several characteristics (see trace). 
If you are from Nordic, I can send you a vmware VM with devtree/source and the target BLE peripheral as we have an NDA with Nordic.  If you would like my specific code for subscribe, write, etc, can can post that here.

joerchan, 2019-03-25:
> Code to subscribe:
> uint16_t auth_desc_handle = 0;
> struct bt_gatt_subscribe_params auth_subscribe_req = {0};
>
> if ( (auth_desc_handle = get_descriptor_handle(BLE_UUID_DXCM_AUTH)) != 0)
> { [...]

I think the issue here is that the declaration of `struct bt_gatt_subscribe_params auth_subscribe_req = {0};`, which is on the stack. This parameter needs to be kept valid while you have indications/notifications active.

This probably crashed because the pointer to cb_indication_DXCM_AUTH has been overwritten and the stack is calling a function pointer into invalid memory, causing an MPU fault or BUS fault.

@lanvis Can you confirm that moving ` struct bt_gatt_subscribe_params auth_subscribe_req` to global scope fixes your issue?

carlescufi, 2019-03-25:
@Vudentz could you please take a look at [this comment](https://github.com/zephyrproject-rtos/zephyr/issues/13522#issuecomment-476114079) from @joerchan to see if you agree with his analysis?

Vudentz, 2019-03-25:
> I think the issue here is that the declaration of `struct bt_gatt_subscribe_params auth_subscribe_req = {0};`, which is on the stack. This parameter needs to be kept valid while you have [...]

Yep, that seems to be the problem, the documentation actually states these parameters should stay valid as long as the operation is in progress.

lanvis, 2019-03-25:
Yep … but also in my write.  I just declared them static (not global)
I assumed that the driver copied that data out of the structure before returning …. Need to look at everything now.   
You are the  Mann!!! 😊

I am even getting my indicate.  Still not sure about the format of the indicate data as I get a 

bool gattc_write_request(uint16_t conn_handle, uint16_t attr_handle, const dxcm_buffer_t* data)
{
	struct bt_conn *conn = get_ble_gap_get_conn_ref(conn_handle);
        static struct bt_gatt_write_params write_params = {0};
        bool result = false;
        //uint8_t mrrdat[2]= {1,2};

        //if (!ble_gap_is_connected())
        if (conn != NULL)
	{
  		uint32_t error_code = 0;
                write_params.handle = attr_handle;
                write_params.offset = 0;
		write_params.length = data->len;
		//write_params.data = data->p_buffer;
                write_params.data = data->p_buffer;
                write_params.func = &cb_write_response;
                printm("..write conn %08x\n",conn);
                if ((error_code = bt_gatt_write(conn, &write_params)) == 0)
                {
  
Still not sure of the indicate params as the subscribe structure does not support the indicate parameters service procedure parameters.
I think this is a "bug" because the compiler will always give warning.

It seems that I copied the notify call back function for the indicate so my indicate call back is:
   static u8_t cb_indicate_DXCM_AUTH(struct bt_conn *conn,
                                                   struct bt_gatt_subscribe_params *params,
                                                                const void *data, u16_t length)

But the indicate callback should have these parameters (I believe)
   void cb_indicate_DXCM_AUTH(struct bt_conn *conn,
                            const struct bt_gatt_attr *attr,
                                                    u8_t err)

But how do I get my indicate data?  Do I  pass the same conn and attr structure that I received to read my indicate values with a buffer ptr to get max number of bytes and then it returns the actual number???
..like this??                         

void cb_indicate_DXCM_AUTH(struct bt_conn *conn, const struct bt_gatt_attr *attr,   u8_t err)
{
                                uint8_t my_data[64] ;
                                uint16_t num_in = attr->read(conn, attr, my_data, 64, 0);
                                …
                                …

}

carlescufi, 2019-03-25:
@lanvis thank you for confirming that this is not an issue anymore. Regarding the other questions you have, can you please either:

- Send an email to the [mailing list](https://lists.zephyrproject.org/g/devel)
- Ask us directly on #bluetooth on [Slack](https://tinyurl.com/yarkuemx)

Closing the issue.

closed 2019-03-25 by carlescufi

joerchan, 2019-03-25:
This is correct for subscribing to either notifications and indications
```
static u8_t cb_indicate_DXCM_AUTH(struct bt_conn *conn,
struct bt_gatt_subscribe_params *params,
const void *data, u16_t length)
```
You have a pointer to the data here.

The other one you are looking at is used for sending indications through `bt_gatt_indicate`.