zephyrproject-rtos/zephyr #13549: stm32_min_dev change console port

fghjhuang, 2019-02-20:
i try to change kconfig file "UART_STM32_PORT_1" to "UART_STM32_PORT_3", but i dosen't work, i can not build the helloworld project successfully, how to make it right? i want to use the uart3 to work as console port.

# Kconfig - STM32 Minimum Development Board Configuration
#
# Copyright (c) 2017, embedjournal.com
#
# SPDX-License-Identifier: Apache-2.0
#

if BOARD_STM32_MIN_DEV

config BOARD
	default "stm32_min_dev"

if UART_CONSOLE

config UART_STM32_PORT_3
	default y

endif # UART_CONSOLE

if I2C

config I2C_1
	default y

endif # I2C

if PWM

config PWM_STM32_1
	default y

endif # PWM

endif # BOARD_STM32_MIN_DEV

erwango, 2019-02-20:
You also need to modify "zephyr,console = &usart3" in stm32_min_dev.dts

fghjhuang, 2019-02-21:
`-- Selected BOARD stm32_min_dev
Zephyr version: 1.13.99
Parsing Kconfig tree in /home/david/Desktop/zephyr-master/Kconfig
Loading /home/david/Desktop/zephyr-master/samples/plcnet/build/zephyr/.config as base
CMake Warning at /home/david/Desktop/zephyr-master/cmake/toolchain.cmake:33 (message):
  ZEPHYR_GCC_VARIANT is deprecated, please use ZEPHYR_TOOLCHAIN_VARIANT
  instead
Call Stack (most recent call first):
  /home/david/Desktop/zephyr-master/cmake/app/boilerplate.cmake:260 (include)
  CMakeLists.txt:3 (include)

-- git describe failed: fatal: Not a git repository (or any of the parent directories): .git; "1.13.99" will be used instead
-- Loading /home/david/Desktop/zephyr-master/boards/arm/stm32_min_dev/stm32_min_dev.dts as base
-- Overlaying /home/david/Desktop/zephyr-master/dts/common/common.dts
stm32_min_dev.dts_compiled: Warning (simple_bus_reg): /soc/usbphy: missing or empty reg/ranges property
Traceback (most recent call last):
  File "/home/david/Desktop/zephyr-master/scripts/dts/extract_dts_includes.py", line 758, in <module>
    main()
  File "/home/david/Desktop/zephyr-master/scripts/dts/extract_dts_includes.py", line 749, in main
    defs = generate_node_definitions(yaml_list)
  File "/home/david/Desktop/zephyr-master/scripts/dts/extract_dts_includes.py", line 706, in generate_node_definitions
    extract_string_prop(chosen[k], None, "label", v)
  File "/home/david/Desktop/zephyr-master/scripts/dts/extract_dts_includes.py", line 315, in extract_string_prop
    node = reduced[node_address]
KeyError: '/soc/serial@40004800'
CMake Error at /home/david/Desktop/zephyr-master/cmake/dts.cmake:193 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/david/Desktop/zephyr-master/cmake/app/boilerplate.cmake:295 (include)
  CMakeLists.txt:3 (include)

-- Configuring incomplete, errors occurred!
`

**it still can not build successfully,can you help me?**

fghjhuang, 2019-02-21:
**it works now ,ths! you still need to  config  usart to status ->ok,then you can change the console port**
&usart3 {
	current-speed = <115200>;
	pinctrl-0 = <&usart3_pins_a>;
	pinctrl-names = "default";
	status = "ok";
};

closed 2019-02-21 by fghjhuang