zephyrproject-rtos/zephyr #13560: USB: netusb: kernel crash when testing example echo_server with nucleo_f412zg

Jeepgoing, 2019-02-20:
I have configured board with this command
`cmake -GNinja -DBOARD=nucleo_f412zg -DOVERLAY_CONFIG=overlay-netusb.conf ..`

when i boot up kernel crashed.

```
[1;32muart:~$ [m***** Booting Zephyr OS v1.14.0-rc1-348-gcda3fad5c8 *****
[8D[J[00:00:00.059,036] [0m<inf> usb_net: netusb initialized[0m
[1;32muart:~$ [m[8D[J[00:00:00.059,210] [1;31m<err> usb_net: interface disabled[0m
[1;32muart:~$ [m[8D[J[00:00:00.059,380] [1;31m<err> usb_net: interface disabled[0m
[1;32muart:~$ [m[8D[J[00:00:00.059,532] [1;31m<err> usb_net: interface disabled[0m
[1;32muart:~$ [m[8D[J[00:00:00.059,684] [1;31m<err> usb_net: interface disabled[0m
[1;32muart:~$ [m[8D[J[00:00:00.064,681] [0m<inf> net_config: Initializing network[0m
[1;32muart:~$ [m[8D[J[00:00:00.064,716] [0m<inf> net_config: IPv4 address: 192.0.2.1[0m
[1;32muart:~$ [m[8D[J[00:00:00.064,918] [1;31m<err> usb_net: interface disabled[0m
[1;32muart:~$ [m[8D[J[00:00:00.065,077] [1;31m<err> usb_net: interface disabled[0m
[1;32muart:~$ [m[8D[J[00:00:00.160,084] [0m<inf> net_config: IPv6 address: fe80::200:5eff:fe00:5300[0m
[1;32muart:~$ [m[8D[J[1;32muart:~$ [m[1B[9D[8D[J[00:00:00.166,094] [0m<inf> net_config: IPv6 address: fe80::200:5eff:fe00:5300[0m

***** USAGE FAULT *****
  Division by zero
***** Hardware exception *****
Current thread ID = 0x20003620
Faulting instruction address = 0x80027d4
Fatal fault in ISR! Spinning...

```
so anyone have met this issue?

qianfan-Zhao, 2019-02-24:
an advice not related this issue:

your serial port software doesn't support color so there has something such as [1;

you can try putty

Jeepgoing, 2019-02-24:
@qianfan-Zhao  thanks

finikorg, 2019-02-26:
I have `reel_board` and `frdm_k64f, the sample works fine with them. Do you have other USB samples working on that board?

Jeepgoing, 2019-02-26:
@finikorg  I have tested `hid` sample,it't works good 
`https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/subsys/usb/hid`

finikorg, 2019-03-18:
@Jeepgoing are you using USB Ethernet ECM with Windows? You should use RNDIS. I can verify that RNDIS works with Windows while selecting ECM produces crash (which is also might be an issue, I will check it).

finikorg, 2019-03-18:
There seems to be division by zero bug inside `stm32cube`

```
#3  <signal handler called>
#4  0x080017b8 in USB_EPStartXfer (USBx=0x50000000, ep=0x20000ad8 <usb_dc_stm32_state+532>, 
    dma=0 '\000')
    at /usr/local/src/viper/foss-rtos/ext/hal/st/stm32cube/stm32f4xx/drivers/src/stm32f4xx_ll_usb.c:750
#5  0x080138ec in HAL_PCD_EP_Receive (hpcd=hpcd@entry=0x200008c4 <usb_dc_stm32_state>, 
    ep_addr=ep_addr@entry=1 '\001', pBuf=pBuf@entry=0x20000f18 <usb_dc_stm32_state+1620> "", 
    len=<optimized out>)
    at /usr/local/src/viper/foss-rtos/ext/hal/st/stm32cube/stm32f4xx/drivers/src/stm32f4xx_hal_pcd.c:1040
#6  0x08005f6e in usb_dc_ep_start_read (ep=<optimized out>, data=<optimized out>, 
    max_data_len=<optimized out>)
```

```
(gdb) frame 4
#4  0x080017b8 in USB_EPStartXfer (USBx=0x50000000, ep=0x20000ad8 <usb_dc_stm32_state+532>, 
    dma=0 '\000')
    at /usr/local/src/viper/foss-rtos/ext/hal/st/stm32cube/stm32f4xx/drivers/src/stm32f4xx_ll_usb.c:750
750	      pktcnt = (ep->xfer_len + ep->maxpacket -1U)/ ep->maxpacket;
(gdb) p ep->maxpacket
$2 = 0
```

erwango, 2019-03-18:
@ydamigos , @loicpoulain can you have a check?

ydamigos, 2019-03-19:
I couldn't reproduce the error. I tested echo_server using netusb on my 96b_carbon. It starts without crashing:
```
uart:~$ ***** Booting Zephyr OS v1.14.0-rc1-1291-g94ae61f765d3 *****
[00:00:00.059,000] <inf> usb_net: netusb initialized
[00:00:00.059,000] <err> usb_net: interface disabled
[00:00:00.059,000] <err> usb_net: interface disabled
[00:00:00.059,000] <err> usb_net: interface disabled
[00:00:00.059,000] <err> usb_net: interface disabled
[00:00:00.324,000] <inf> spi_ll_stm32: CS control inhibited (no GPIO device)
[00:00:00.373,000] <wrn> bt_hci_core: ECC HCI commands not available
[00:00:00.374,000] <inf> net_config: Initializing network
[00:00:00.374,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:00.374,000] <wrn> net_if: iface 0x200110e0 is down
[00:00:00.374,000] <err> net_if: Cannot join all nodes address ff02::1 (-1)
[00:00:01.060,000] <err> usb_net: interface disabled
[00:00:03.375,000] <inf> net_echo_server_sample: Run echo server
[00:00:03.375,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv6)...
[00:00:03.375,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv4)...
[00:00:03.375,000] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv6)...
[00:00:03.375,000] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv4)...
uart:~$ 
```

finikorg, 2019-03-20:
@ydamigos How do you run it? I can reproduce the crash building echo_server with netusb overlay (default would be ECM protocol) and connecting it to Windows, not Linux. @Jeepgoing is this correct procedure?

Jeepgoing, 2019-03-20:
@finikorg  yes, i have tested netusb with defconfig.

finikorg, 2019-03-20:
> @finikorg yes, i have tested netusb with defconfig.

with Windows, right? Because in Linux it should work just fine.

Jeepgoing, 2019-03-20:
@finikorg  yep

finikorg, 2019-03-21:
@ydamigos So, have you verified the crash?

finikorg, 2019-03-21:
> > How do you run it? I can reproduce the crash building echo_server with netusb overlay (default would be ECM protocol) and connecting it to Windows, not Linux. @Jeepgoing is this correct procedure?
> 
> @finikorg I built echo_server with netusb overlay and connected it to Linux. I need to try it on Windows.
> It seems similar to #13066 and #7364 crashes. Probably some EP is used before it is configured.

Yes, exactly, Probably Windows does not configure ECM.

ydamigos, 2019-03-21:
> So, have you verified the crash?

Yes, it crashes on Windos. It tries to read from EP01 without configuring it.
```
uart:~$ ***** Booting Zephyr OS v1.14.0-rc1-1291-g94ae61f765d3 *****
[00:00:00.000,000] <dbg> usb_dc_stm32.usb_dc_ep_check_cap: ep 81, mps 16, type 3
[00:00:00.000,000] <dbg> usb_dc_stm32.usb_dc_ep_check_cap: ep 82, mps 64, type 2
[00:00:00.000,000] <dbg> usb_dc_stm32.usb_dc_ep_check_cap: ep 1, mps 64, type 2
[00:00:00.000,000] <dbg> usb_dc_stm32.usb_dc_set_status_callback: 
[00:00:00.000,000] <dbg> usb_dc_stm32.usb_dc_attach: 
[00:00:00.000,000] <dbg> usb_dc_stm32.usb_dc_stm32_init: HAL_PCD_Init
[00:00:00.055,000] <dbg> usb_dc_stm32.usb_dc_stm32_init: HAL_PCD_Start
[00:00:00.059,000] <dbg> usb_dc_stm32.HAL_PCD_ResetCallback: 
[00:00:00.059,000] <dbg> usb_dc_stm32.usb_dc_ep_configure: ep 0x00, ep_mps 64, ep_type 0
[00:00:00.059,000] <dbg> usb_dc_stm32.usb_dc_ep_configure: ep 0x80, ep_mps 64, ep_type 0
[00:00:00.059,000] <dbg> usb_dc_stm32.usb_dc_ep_set_callback: ep 0x00
[00:00:00.059,000] <dbg> usb_dc_stm32.usb_dc_ep_set_callback: ep 0x80
[00:00:00.059,000] <dbg> usb_dc_stm32.usb_dc_ep_set_callback: ep 0x81
[00:00:00.059,000] <dbg> usb_dc_stm32.usb_dc_ep_set_callback: ep 0x01
[00:00:00.061,000] <dbg> usb_dc_stm32.HAL_PCD_SuspendCallback: 
[00:00:00.416,000] <dbg> usb_dc_stm32.HAL_PCD_ResetCallback: 
[00:00:00.848,000] <dbg> usb_dc_stm32.HAL_PCD_SetupStageCallback: 
[00:00:00.848,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 8 bytes, 0+8, 0x200007dc
[00:00:00.848,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x80, len 18
[00:00:00.848,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x80, len 0
[00:00:00.850,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 0 bytes, 0+0, 0x20011f2a
[00:00:00.901,000] <dbg> usb_dc_stm32.HAL_PCD_ResetCallback: 
[00:00:00.933,000] <inf> spi_ll_stm32: CS control inhibited (no GPIO device)
[00:00:00.935,000] <dbg> usb_dc_stm32.HAL_PCD_SetupStageCallback: 
[00:00:00.935,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 8 bytes, 0+8, 0x200007dc
[00:00:00.935,000] <dbg> usb_dc_stm32.usb_dc_set_address: addr 2 (0x02)
[00:00:00.935,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x80, len 0
[00:00:00.936,000] <dbg> usb_dc_stm32.HAL_PCD_DataInStageCallback: epnum 0x00
[00:00:00.948,000] <dbg> usb_dc_stm32.HAL_PCD_SetupStageCallback: 
[00:00:00.948,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 8 bytes, 0+8, 0x200007dc
[00:00:00.948,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x80, len 18
[00:00:00.948,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x80, len 0
[00:00:00.949,000] <dbg> usb_dc_stm32.HAL_PCD_DataInStageCallback: epnum 0x00
[00:00:00.950,000] <dbg> usb_dc_stm32.HAL_PCD_DataOutStageCallback: epnum 0x00, rx_count 0
[00:00:00.950,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 0 bytes, 0+0, 0x20011f2a
[00:00:00.955,000] <dbg> usb_dc_stm32.HAL_PCD_SetupStageCallback: 
[00:00:00.955,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 8 bytes, 0+8, 0x200007dc
[00:00:00.955,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x80, len 80
[00:00:00.955,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x80, len 0
[00:00:00.956,000] <dbg> usb_dc_stm32.HAL_PCD_DataInStageCallback: epnum 0x00
[00:00:00.956,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x80, len 16
[00:00:00.956,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x80, len 0
[00:00:00.957,000] <dbg> usb_dc_stm32.HAL_PCD_DataInStageCallback: epnum 0x00
[00:00:00.958,000] <dbg> usb_dc_stm32.HAL_PCD_DataOutStageCallback: epnum 0x00, rx_count 0
[00:00:00.958,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 0 bytes, 0+0, 0x20011f7a
[00:00:00.961,000] <dbg> usb_dc_stm32.HAL_PCD_SetupStageCallback: 
[00:00:00.961,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 8 bytes, 0+8, 0x200007dc
[00:00:00.961,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x80, len 10
[00:00:00.961,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x80, len 0
[00:00:00.962,000] <dbg> usb_dc_stm32.HAL_PCD_DataInStageCallback: epnum 0x00
[00:00:00.963,000] <dbg> usb_dc_stm32.HAL_PCD_DataOutStageCallback: epnum 0x00, rx_count 0
[00:00:00.963,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 0 bytes, 0+0, 0x20011fa6
[00:00:00.966,000] <dbg> usb_dc_stm32.HAL_PCD_SetupStageCallback: 
[00:00:00.966,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 8 bytes, 0+8, 0x200007dc
[00:00:00.966,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x80, len 4
[00:00:00.966,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x80, len 0
[00:00:00.967,000] <dbg> usb_dc_stm32.HAL_PCD_DataInStageCallback: epnum 0x00
[00:00:00.968,000] <dbg> usb_dc_stm32.HAL_PCD_DataOutStageCallback: epnum 0x00, rx_count 0
[00:00:00.968,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 0 bytes, 0+0, 0x20011f7e
[00:00:00.971,000] <dbg> usb_dc_stm32.HAL_PCD_SetupStageCallback: 
[00:00:00.971,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 8 bytes, 0+8, 0x200007dc
[00:00:00.971,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x80, len 16
[00:00:00.971,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x80, len 0
[00:00:00.972,000] <dbg> usb_dc_stm32.HAL_PCD_DataInStageCallback: epnum 0x00
[00:00:00.973,000] <dbg> usb_dc_stm32.HAL_PCD_DataOutStageCallback: epnum 0x00, rx_count 0
[... 25 lines cut ...]
[00:00:00.991,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x80, len 80
[00:00:00.991,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x80, len 0
[00:00:00.993,000] <dbg> usb_dc_stm32.HAL_PCD_DataInStageCallback: epnum 0x00
[... cut ...]
[00:00:00.999,000] <dbg> usb_dc_stm32.usb_dc_ep_write: ep 0x80, len 0
[00:00:01.000,000] <dbg> usb_dc_stm32.HAL_PCD_DataInStageCallback: epnum 0x00
[00:00:01.002,000] <dbg> usb_dc_stm32.HAL_PCD_SetupStageCallback: 
[00:00:01.002,000] <dbg> usb_dc_stm32.usb_dc_ep_read_wait: ep 0x00, 8 bytes, 0+8, 0x200007dc
[00:00:01.002,000] <dbg> usb_dc_stm32.usb_dc_ep_start_read: ep 0x01, len 64
***** USAGE FAULT *****
  Division by zero
***** Hardware exception *****
Current thread ID = 0x20002e20
Faulting instruction address = 0x8002030
Fatal fault in ISR! Spinning...
```

finikorg, 2019-03-22:
Window set iface 0 alt settings 0, we should not call callback then, I will fix this issue.

closed 2019-04-05 by nashif

Jeepgoing, 2019-06-14:
> I couldn't reproduce the error. I tested echo_server using netusb on my 96b_carbon. It starts without crashing:
>
> ```
> uart:~$ ***** Booting Zephyr OS v1.14.0-rc1-1291-g94ae61f765d3 ***** [...]

Hi, @ydamigos ,  can you tell me how to test this sample?  

there have two unknow devices on my PC,
[image]

ydamigos, 2019-06-14:
Hi @Jeepgoing,

On windows, I had to edit overlay-netusb.conf and change `CONFIG_USB_DEVICE_NETWORK_ECM=y` to `CONFIG_USB_DEVICE_NETWORK_RNDIS=y`.

Then I run:
```
$ cmake -DBOARD=96b_carbon -DOVERLAY_CONFIG=overlay-netusb.conf ../zephyr/samples/net/sockets/echo_server
$ make flash
```
I connected the device to Windows machine. A new `Unknown device` showed up in `Device Manager`. I chose to update drivers with the option `Pick from a list` and then selected `‘Network Adapters’ -> ‘Microsoft Corp’ -> ‘Remote NDIS Compatible device’`.

New adapter showed up in network interfaces.

finikorg, 2019-06-17:
@ydamigos Could you try enabling `USB_DEVICE_OS_DESC`, it should install Windows drivers automatically.

ydamigos, 2019-06-17:
>  Could you try enabling USB_DEVICE_OS_DESC, it should install Windows drivers automatically.

@finikorg I enabled it but Windows drivers didn't install automatically.

finikorg, 2019-06-17:
> > Could you try enabling USB_DEVICE_OS_DESC, it should install Windows drivers automatically.
> 
> @finikorg I enabled it but Windows drivers didn't install automatically.

I think you need to remove old drivers associated with this VID/PID, you can change that pair.

ydamigos, 2019-06-17:
> > > Could you try enabling USB_DEVICE_OS_DESC, it should install Windows drivers automatically.
> > 
> > 
> > @finikorg I enabled it but Windows drivers didn't install automatically.
> 
> I think you need to remove old drivers associated with this VID/PID, you can change that pair.

I changed VID/PID pair and "Remote NDIS Compatible device" driver installed automatically.

Jeepgoing, 2019-07-03:
@ydamigos @finikorg  it works, thanks.

commit 8ed215e0 (2019-04-05): usb: usb_dc_stm32: Fix reading invalid EP

Fix crash when reading invalid endpoint found in harness test.

Fixes #13560
commit 4c9ea1e3 (2019-04-11): usb: netusb: ecm: Enable netusb for the right iface/alt_set

Only enable netusb for the right configuration of interface and
alt_setting.

Fixes #13560