zephyrproject-rtos/zephyr #13567: tests/subsys/settings/fcb/base64 fails when write-block-size is 8

erwango, 2019-02-20:
**Describe the bug**
tests/subsys/settings/fcb/base64 on SoC with flash write-block-size set to 8.
I get same failed result with disco_l475_iot1 (wbs 8) and reel_board (with flash driver modified with write_block_size = 8).

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
fcb/base64 should work with all flash drivers with write-block-size <= 8.

**Impact**
Can't use fcb based subystems (settings, ...) with system with write-block-size = 8.

**Screenshots or console output**
```
***** Booting Zephyr OS zephyr-v1.13.0-4076-gc62ba44 *****
Running test suite test_config_fcb
===================================================================
starting test - test_settings_encode
PASS - test_settings_encode
===================================================================
starting test - test_setting_raw_read
PASS - test_setting_raw_read
===================================================================
starting test - test_setting_val_read
PASS - test_setting_val_read
===================================================================
starting test - config_empty_lookups
PASS - config_empty_lookups
===================================================================
starting test - test_config_insert
PASS - test_config_insert
===================================================================
starting test - test_config_getset_unknown
PASS - test_config_getset_unknown
===================================================================
starting test - test_config_getset_int
PASS - test_config_getset_int
===================================================================
starting test - test_config_getset_int64
PASS - test_config_getset_int64
===================================================================
starting test - test_config_commit
PASS - test_config_commit
===================================================================
starting test - test_config_empty_fcb
PASS - test_config_empty_fcb
===================================================================
starting test - test_config_save_1_fcb
PASS - test_config_save_1_fcb
===================================================================
starting test - test_config_insert2
PASS - test_config_insert2
===================================================================
starting test - test_config_save_2_fcb
PASS - test_config_save_2_fcb
===================================================================
starting test - test_config_insert3
PASS - test_config_insert3
===================================================================
starting test - test_config_save_3_fcb

    Assertion failed at /local/mcu/zephyr/zephyr-stm32wb/tests/subsys/settings/fcb/src/settings_test_fcb.c:285)
bad set-value size
FAIL - test_config_save_3_fcb
===================================================================
starting test - test_config_compress_reset

    Assertion failed at /local/mcu/zephyr/zephyr-stm32wb/tests/subsys/settings/fcb/src/settings_test_fcb.c:285)
bad set-value size
FAIL - test_config_compress_reset
===================================================================
starting test - test_config_save_one_fcb

    Assertion failed at /local/mcu/zephyr/zephyr-stm32wb/tests/subsys/settings/fcb/src/settings_test_fcb.c:285)
bad set-value size
FAIL - test_config_save_one_fcb
===================================================================
starting test - test_config_compress_deleted

    Assertion failed at /local/mcu/zephyr/zephyr-stm32wb/tests/subsys/settings/fcb/src/settings_test_compress_)
The deleted settings shouldn be compressed.
FAIL - test_config_compress_deleted
===================================================================
Test suite test_config_fcb failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Linux,
 - Toolchain (Zephyr SDK)
 - Commit SHA: 020d32dca01b9d8c8278d60e43fd9eb3bf3f2ab0

**Additional context**
Add any other context about the problem here.

erwango, 2019-02-20:
Tbh, I'm not sure setting wbs to 8 in reel_board is supposed to work.
Anyway, it should work on disco_l475_iot1, as flash driver seems to be ok so far (It behaves as expected with samples/drivers/flash_shell or samples/subsys/nvs).

nvlsianpu, 2019-02-21:
Maybe it is another artifact related to https://github.com/zephyrproject-rtos/zephyr/issues/12967.

nvlsianpu, 2019-02-21:
Patch https://github.com/zephyrproject-rtos/zephyr/pull/13625 fixes this issue - can you test it?

nvlsianpu, 2019-02-22:
This bug can be closed - @erwango Pleas confirm or deny.

erwango, 2019-02-25:
@nvlsianpu, txs. I'll test it as soon as I have some time (this week for sure).

erwango, 2019-02-25:
I had a test and it seems I have a configuration issue:
FAIL - test_config_save_fcb_unaligned
Issue happens here:
```
		rc = fcb_init(CONFIG_SETTINGS_FCB_FLASH_AREA, &cf->cf_fcb);
```

I have CONFIG_SETTINGS_FCB_FLASH_AREA=4
while reel_board defines CONFIG_SETTINGS_FCB_FLASH_AREA=3

Not sure it could be the issue. i'll try to spend more time on it tomorrow.

nvlsianpu, 2019-02-26:
I this case exact value you have shouldn't make any difference. However CONFIG_SETTINGS_FCB_FLASH_AREA became a orphan (need to use only its default value `$(dt_int_val,DT_FLASH_AREA_STORAGE_ID)`) - so just keep untouched.

nvlsianpu, 2019-02-26:
I just realized that you have been taring to run test with target modification:
> reel_board (with flash driver modified with write_block_size = 8

I this case unaligned test shouldn't pass as it test 4 B alignment. This test suite was adapted to run on nrf52 targets with 1 B alignment requirement.

In this circumstance, if all other tests pass - this mean that you already used unaligned access with many successes.

nvlsianpu, 2019-02-26:
Target for test is pointed always in testcase.yaml file

erwango, 2019-02-27:
@nvlsianpu 

> I just realized that you have been taring to run test with target modification:

Indeed, 8B might not even be supposed to work. No problem.

> This test suite was adapted to run on nrf52 targets with 1 B alignment requirement.

In that case, how can I check that FCB is actually working with other zephyr flash drivers?
And if there are any contraint put by FCB on the driver (or expectations), are they documented so I can update my flash driver accordingly (if doable)?

nvlsianpu, 2019-02-27:
settings fcb-backend module is supposed to work with 8 B write-block-size.
Change line https://github.com/zephyrproject-rtos/zephyr/blob/master/tests/subsys/settings/fcb/src/settings_test_save_unaligned.c#L25 to `cf.cf_fcb.f_align = 8;` locally and test should pass on your machine.

I relay want to switch testing this to qemu target:
I'm awaiting for flash simulatot PR to be merged, wihich would helps to test this module more widhtlly- but it stall because lack of consensus.

erwango, 2019-02-27:
ok, txs, will have a test right now.

erwango, 2019-02-27:
And is there any link between supported write block size and raw vs base64 setting tests?

nvlsianpu, 2019-02-27:
> And is there any link between supported write block size and raw vs base64 setting tests?

No, both works well. Currently I see no gain in using base64 - I would deprecated it.

erwango, 2019-02-27:
> No, both works well. Currently I see no gain in using base64 - I would deprecated it.

Ok, txs.

Btw, I managed to make it pass with the suggested patch and applying overlay, txs!

nashif, 2019-03-03:
@erwango can we close this? @nvlsianpu maybe we should make this test only run on platforms where it is supposed to run.

erwango, 2019-03-04:
@nashif, I'm closing it, thanks @nvlsianpu for the support.

Regarding:
> maybe we should make this test only run on platforms where it is supposed to run.

This is a subsystem test, so ideally I don't think it should be restricted to few platforms. Though, since it is not generic enough yet, ok to limit but at least it should be stated somewhere what are the expectations/limitations vs flash driver. Or better, filter based on flash driver/ip properties.

nvlsianpu, 2019-03-04:
Tests success depends on what flash driver support in run-time  I can add initial test for assert that - so any user will see the real the reason of failure (and only this - so no misleading messages).

Some of test are checking flash footprint pattern - how them looks like depends on write-block-size, that why test targets are limited for now.

What I really want to do is to test this with qemu and [flash driver simulator](https://github.com/zephyrproject-rtos/zephyr/pull/8254) - Thanks to that it will be possible to verify settings on zephyr CI, with all flash alignment we need.

closed 2019-03-04 by nvlsianpu