zephyrproject-rtos/zephyr #13569: ZTEST: Add optional float/double comparison support

microbuilder, 2019-02-20:
Currently the `zassert_*` functions in the ztest API are based on integer values, which can be accurately compared (`1==1`, etc.). 

There are no options for **single or double precision floating point value comparisons**, which are error-prone to compare againist fixed operands (`x==0.5f`), particularly with single-precision values.

When `CONFIG_FLOAT` is enabled, options should be included to compare single and double precision values for equality within a specific margin.

In previous projects, I've implemented the following helper functions to work with floating point values in unit tests ...

```
bool f_is_equal(float a, float b, float epsilon);
bool f_is_at_least(float a, float b);
bool f_is_less_than(float a, float b);
bool f_is_greater_than(float a, float b);
bool f_is_within(float a, float upper, float lower);
```

... which allows tests like the following code, with control over the degree of prevision involved in the comparison:

```
#define F_EPSILON5  1E-5
TEST_ASSERT(f_is_equal(xyy.x, 0.44422f, F_EPSILON5, "2900K -> xyY"));
```

I understand the allergy to float in an embedded context, but if `CONFIG_FLOAT` has been set to `y` and is being used in the application or library, shouldn't the option be available to test said floating point values in both single and double-precision widths, using 'safer' value comparisons?

I can continue to use my own code to test these, but it seems reasonable to at least consider adding these helpers to ztest and [ztest_assert.h](https://github.com/zephyrproject-rtos/zephyr/blob/master/tests/ztest/include/ztest_assert.h) with an appropriate check for `CONFIG_FLOAT`?

aescolar, 2019-02-20:
@microbuilder : Would you consider preparing a PR with this enhancement?

microbuilder, 2019-02-20:
Sure, I just wanted to check if there was strong resistance to including float support, but I'll put a PR together for this.

microbuilder, 2020-05-21:
Closing for lack of activity. Will reopen as PR when ready.

closed 2020-05-21 by microbuilder