zephyrproject-rtos/zephyr #13574: Missing documentation for fs subsystem

erwango, 2019-02-20:
There is no documentation on fcb or nffs.
Specially how it behaves vs flash supported write-block-size...

ramakrishnapallala, 2019-03-05:
@nvlsianpu  can you look into this issue as fcb and nffs are owned by you.

nashif, 2019-03-05:
@ramakrishnapallala this is less about the specific filesystems and more about the virtual file system interface and fs support in general.

nvlsianpu, 2019-03-05:
- fcb doc is missing - can be take form mynewt.
- NFFS is external project as well - currently not well documented, can consider documentation making but, for sure no now.

@nashif  Can we consider this as lack of documentation, not a bug. I would create separate issue for both subjects.

nashif, 2019-03-05:
lack of documentation is a BUG :)

I would start here: https://docs.zephyrproject.org/latest/reference/file_system/index.html
This is currently the API only need more content to the file system feature in general. Probably a quick into about the supported filesystems with details to be added later.

ramakrishnapallala, 2019-03-12:
I will start on adding more documentation to the API's

ramakrishnapallala, 2019-03-26:
Pushed PR: https://github.com/zephyrproject-rtos/zephyr/pull/14796/

nashif, 2019-03-29:
@nvlsianpu what about documenting fcb and nffs?

nvlsianpu, 2019-03-29:
FCB doc I can copy form MyNewt.
NFFS is under fixing now - expects that It will change a lot. I would only refer to its design.txt file and describe its limitation.

nashif, 2019-04-02:
@nvlsianpu any updates?

nashif, 2019-06-01:
@nvlsianpu ping

nvlsianpu, 2019-07-08:
I'm Preparing FCB doc right now.

nvlsianpu, 2019-07-09:
FCB documentation: https://github.com/zephyrproject-rtos/zephyr/pull/17425

carlescufi, 2019-08-19:
FCB is now documented and NFFS is on the path to deprecation, so closing this.

closed 2019-08-19 by carlescufi