zephyrproject-rtos/zephyr #13594: tests/kernel/mem_protect/mem_protect/kernel.memory_protection build failure on minnowboard with new SDK

nashif, 2019-02-20:
**Describe the bug**
tests/kernel/mem_protect/mem_protect/kernel.memory_protection build failure on minnowboard with new SDK

**To Reproduce**

using the new SDK, build tests/kernel/mem_protect/mem_protect/ for minnowboard

**Expected behavior**
it builds cleanly

**Screenshots or console output**
[100%] Linking C executable zephyr.elf
/home/nashif/Work/SDK/0.10.0-rc2-pr-43/i586-zephyr-elf/bin/../lib/gcc/i586-zephyr-elf/8.2.0/../../../../i586-zephyr-elf/bin/ld:linker_pass_final.cmd:64 cannot move location counter backwards (from 000000000010be4e to 000000000010be4d)

**Environment (please complete the following information):**
 - Toolchain SDK 0.10.0-rc3

nashif, 2019-02-20:
see #13272

galak, 2019-02-20:
I think this might have been fixed by this commit: 5f4683db34b5639c9c908cbf4fbbc68db66defea

galak, 2019-02-20:
> I think this might have been fixed by this commit: [5f4683d](https://github.com/zephyrproject-rtos/zephyr/commit/5f4683db34b5639c9c908cbf4fbbc68db66defea)

Never mind, this commit fixes the build issue on minnowboard for   tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout.poll

andrewboie, 2019-02-20:
The root cause is that for some reason, on minnowboard for this test only, CONFIG_KOBJECT_TEXT_AREA (128) is too small.
I don't know why this is the case yet.

PR #13600 (2019-02-21): userspace: increase gperf text areas
128 already gets exceeded in a lot of cases, and the binaries
with the new SDK are very slightly larger. Just kick this up
to 256 and get rid of all the exceptions.

Fixes: #13594
commit 794d382b (2019-02-21): userspace: increase gperf text areas

128 already gets exceeded in a lot of cases, and the binaries
with the new SDK are very slightly larger. Just kick this up
to 256 and get rid of all the exceptions.

Fixes: #13594
closed 2019-02-21 by nashif