zephyrproject-rtos/zephyr #13610: kernel: Non-deterministic and very high ISR latencies

cvinayak, 2019-02-21:
**Describe the bug**
Non-deterministic and very high ISR latencies. Bluetooth controller asserts when running samples/bluetooth/central_hr on nrf51_pca10028 board.

```
[2019-02-21 10:54:43.408] ***** Booting Zephyr OS v1.14.0-rc1-268-g8bd360fa14 *****
[2019-02-21 10:54:43.426] Bluetooth initialized
[2019-02-21 10:54:43.428] [00:00Scanning successfully started
[2019-02-21 10:54:43.434] :00.010,650] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[2019-02-21 10:54:43.443] [00:00:00.010,681] <inf> bt_hci_core: HW Variant: nRF51x (0x0001)
[2019-02-21 10:54:43.453] [00:00:00.010,742] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 1.14 Build 0
[2019-02-21 10:54:43.467] [00:00:00.013,031] <inf> bt_h[DEVICE]: 61:bb:ca:fe:4b:d5 (random), AD evt type 0, AD data len 14, RSSI -46
[2019-02-21 10:54:43.496] [AD]: 1 data_len 1
[2019-02-21 10:54:43.499] [AD]: 255 data_len 9
[2019-02-21 10:54:44.029] [DEVICE]: 61:bb:ca:fe:4b:d5 (random), AD evt type 4, AD data len 0, RSSI -46
[2019-02-21 10:54:46.277] ci_core: Identity: d2:06:aa:d5:f3:9a (random)
[2019-02-21 10:54:46.281] [00:00:00.013,092] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0000, manufacturer 0x05f1
[2019-02-21 10:54:46.292] [00:00:00.013,122] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0xffff
[2019-02-21 10:54:47.015] [00:00:03.627,593] <err> bt_ctlr_llsw_ctrl: assert: '!radio_is_ready()' failed
[2019-02-21 10:54:47.023] ***** HARD FAULT *****
[2019-02-21 10:54:47.025] ***** Hardware exception *****
[2019-02-21 10:54:47.028] Current thread ID = 0x200015dc
[2019-02-21 10:54:47.031] Faulting instruction address = 0x12cbe
[2019-02-21 10:54:47.034] Fatal fault in ISR! Spinning...
```

If a connection establishes with a peripheral in vicinity, the Max. ISR latencies measured is very high, over 100 us.

Legend for the log message:
```
<inf> bt_ctlr_hci: l: 9, 8, 111; t: 46, 46, 95.
```
l: Radio peripheral ISR latency Current value of 9us, Minimum of 8us, Maximum of 111us.
t: Radio peripheral ISR CPU usage Current value of 46, Minimum of 46us, Maximum of 95us.

```
[2019-02-21 10:47:17.917] ***** Booting Zephyr OS v1.14.0-rc1-268-g8bd360fa14 *****
[2019-02-21 10:47:17.933] Bluetooth �***** Booting Zephyr OS v1.14.0-rc1-268-g8bd360fa14 *****
[2019-02-21 10:47:17.983] Bluetooth initialized
[2019-02-21 10:47:17.986] [00:00:0Scanning successfully started
[2019-02-21 10:47:17.991] 0.010,650] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[2019-02-21 10:47:18.001] [00:00:00.010,681] <inf> bt_hci_[DEVICE]: 61:bb:ca:fe:4b:d5 (random), AD evt type 0, AD data len 21, RSSI -49
[2019-02-21 10:47:18.014] [AD]: 1 data_len 1
[2019-02-21 10:47:18.016] [AD]: 3 data_len 4
[2019-02-21 10:47:18.023] core: HW Variant: nRF51x (0x0001)
[2019-02-21 10:47:18.222] [00:00:00.010,711] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0Connected: 61:bb:ca:fe:4b:d5 (random)
[2019-02-21 10:47:18.488] [ATTRIBUTE] handle 39
[2019-02-21 10:47:18.838] [ATTRIBUTE] handle 45
[2019-02-21 10:47:18.938] [ATTRIBUTE] handle 47
[2019-02-21 10:47:18.938] [SUBSCRIBED]
[2019-02-21 10:47:21.047] x00) Version 1.14 Build 0
[2019-02-21 10:47:21.051] --- 18 messages dropped ---
[2019-02-21 10:47:21.052] [00:00:00.996,032] <inf> bt_ctlr_hci: l: 9, 8, 111; t: 46, 46, 95.
[2019-02-21 10:47:21.061] [00:00:01.041,198] <inf> bt_ctlr_hci: l: 8, 8, 111; t: 62, 46, 95.
[2019-02-21 10:47:21.068] [00:00:01.091,339] <inf> bt_ctlr_hci: l: 8, 8, 111; t: 52, 46, 95.
[2019-02-21 10:47:21.076] [00:00:01.141,357] <inf> bt_ctlr_hci: l: 8, 8, 111; t: 46, 46, 95.
[2019-02-21 10:47:21.085] [00:00:01.191,009] <inf> bt_ctlr_hci: l: 8, 8, 111; t: 49, 46, 95.
[2019-02-21 10:47:21.093] [00:00:01.241,333] <inf> bt_ctlr_hci: l: 8, 8, 111; t: 46, 46, 95.
[2019-02-21 10:47:21.101] [00:00:01.290,954] <inf> bt_ctlr_hci: l: 8, 8, 111; t: 50, 46, 95.
[2019-02-21 10:47:21.109] [00:00:01.341,308] <inf> bt_ctlr_hci: l: 8, 8, 111; t: 46, 46, 95.
[2019-02-21 10:47:21.116] [00:00:01.441,375] <inf> bt_ctlr_hci: l: 8, 8, 111; t: 62, 46, 95.
[2019-02-21 10:47:21.124] [00:00:01.442,291] <inf> bt_ctlr_hci: l: 8, 8, 111; t: 46, 46, 95.
[2019-02-21 10:47:21.133] [00:00:01.490,875] <inf> bt_ctlr_hci: l: 8, 8, 111; t: 50, 46, 95.
[2019-02-21 10:47:21.141] [00:00:01.540,496] <inf> bt_ctlr_hci: l: 8, 8, 111; t: 44, 44, 95.
[2019-02-21 10:47:21.149] [00:00:01.941,345] <inf> bt_ctlr_hci: l: 8, 8, 111; t: 46, 44, 95.
[2019-02-21 10:47:21.157] [00:00:01.990,966] <inf> bt_ctlr_hci: l: 9, 8, 111; t: 50, 44, 95.
[2019-02-21 10:47:21.165] [00:00:02.090,454] <inf> bt_ctlr_hci: l: 9, 8, 111; t: 44, 44, 95.
```

**To Reproduce**
Checkout 8bd360fa14a6fff0f09cdf574937a9493b39e095
[... 49 lines cut ...]
[2019-02-21 10:31:57.706] [bt] [INF] encode_control: l: 6, 6, 16; t: 66, 40, 110.
[2019-02-21 10:31:57.712] [bt] [INF] encode_control: l: 11, 6, 16; t: 41, 40, 110.
[2019-02-21 10:31:57.755] [bt] [INF] encode_control: l: 6, 6, 16; t: 47, 40, 110.
[2019-02-21 10:31:57.805] [bt] [INF] encode_control: l: 6, 6, 16; t: 51, 40, 110.
[2019-02-21 10:31:57.856] [bt] [INF] encode_control: l: 6, 6, 16; t: 48, 40, 110.
[2019-02-21 10:31:57.905] [bt] [INF] encode_control: l: 6, 6, 16; t: 51, 40, 110.
[2019-02-21 10:31:58.006] [bt] [INF] encode_control: l: 6, 6, 16; t: 48, 40, 110.
[2019-02-21 10:31:58.056] [bt] [INF] encode_control: l: 6, 6, 16; t: 51, 40, 110.
[2019-02-21 10:31:58.155] [bt] [INF] encode_control: l: 6, 6, 16; t: 46, 40, 110.
[2019-02-21 10:31:58.205] [bt] [INF] encode_control: l: 6, 6, 16; t: 45, 40, 110
```

**Impact**
Showstopper, Blocker for supporting Bluetooth on nRF51 Series and critical for nRF52 series due to bad ISR latencies.

**Screenshots or console output**
Refer to the ones added in the description of the bug.

**Environment (please complete the following information):**
 - OS: Linux (Archlinux distribution) and  MacOS with HomeBrew
 - Toolchain: Native Archlinux distributed arm-none-eabi- toolchain, gcc version 8.2.0 (Arch repository)
 - Commit SHA or Version used: 8bd360fa14a6fff0f09cdf574937a9493b39e095

**Additional context**
None.

cvinayak, 2019-02-21:
Attention: @andyross  and @pabigot towards kernel scheduling/timeout and nrf_rtc_timer.c changes respectively, since v1.13.0.

cvinayak, 2019-02-21:
@pabigot reverting nrf_rtc_timer.c changes to 37fbff6179e8bca7e780a83771b974c5a71a647e reduces the latencies by ~50 us

[image]

cvinayak, 2019-02-21:
@andyross and @pabigot, the nrf_rtc_timer.c  was lockless in v1.13.0. Do you think it can be made lockless for v1.14.0 too?

pabigot, 2019-02-21:
> @pabigot reverting nrf_rtc_timer.c changes to [37fbff6](https://github.com/zephyrproject-rtos/zephyr/commit/t) reduces the latencies by ~50 us

@cvinayak To clarify: you're saying that reverting my changes made in #12605 reduces latencies?

Odd.  My own experiments on the ISR timing itself showed significant decreases in locking with that rework, especially on nRF51.  It's not immediately obvious how the timer ISR could be at fault.

There are a lot more `irq_lock()` calls (hidden as `spinlock` uses; see #12537) throughout Zephyr now so it may be that the direct cause is something else that's affected by the duration of the clock maintenance code.

I'm not working Zephyr at the moment so will have to leave this to @anangl.

nashif, 2019-02-21:
@cvinayak any chance you can add a testcase for this to be able to catch such regressions in the future?

carlescufi, 2019-02-21:
@cvinayak are you sure about your commit? The one you reference is from @anangl (37fbff6) and doesn't seem to be related.

andyross, 2019-02-21:
@cvinayak A timer ISR was never really lockless -- it has to call into the timeout system and out into arbitrary callbacks, all of which need synchronization.  Someone should definitely check the bisection though -- I agree that the referenced commit doesn't seem particularly relevant.

andyross, 2019-02-21:
Oh, and to be clear:

> There are a lot more irq_lock() calls (hidden as spinlock uses; see #12537) throughout Zephyr now

There really are not, and I'm not sure why you'd think so.  Synchronization in the spinlockification patch was almost entirely 1:1, and where it differed it was uniformly in the direction of reducing latency by breaking existing recursive irq_lock usage.  If we've regressed it's a bug and we should fix it.

pabigot, 2019-02-21:
> > There are a lot more irq_lock() calls (hidden as spinlock uses; see #12537) throughout Zephyr now
> 
> There really are not, and I'm not sure why you'd think so.

I was mixing two things, one being an unsupported assumption that since a lot of material was added/changed since 1.13.0, there would be more locks of any sort.  This command:
```
for d in kernel drivers ; do 
  for h in v1.13.0 HEAD ; do  
    echo -n "$d $h " ; git grep -P 'k_spin_lock|irq_lock' ${h} ${d} | wc ;
   done ;
done
```
shows the rough number of lock operations has decreased from 106 to 87 in in `kernel`, and increased from 68 to 98 in `drivers`.  That's consistent with your expectation that the total number has not increased since 1.13.0 (I think `kernel` is more representative than `driver` here), so thanks for calling that out.

But I also meant to indicate that if you're looking for an `irq_lock` you'll undercount them if you don't also look for `spin_lock` which didn't exist in 1.13.0.  Whatever's causing this regression might be due to changes under a lock that's now spelled `k_spin_lock()`.

andyross, 2019-02-21:
I'm sorry, I misread @cvinayak bisection report.  If reverting *only* the changes to nrf_rtc_timer.c between 37fbf fixes your latency problems, then that points to a pretty tight list of patches mostly from @pabigot (FWIW: these all look fine to me):

```
75f77db432 include: misc: util.h: Rename min/max to MIN/MAX
ac36886e62 drivers: nrf: timer: add inline qualifier where inlining is intended
084363a0dc drivers: timer: nrf: refactor for speed and correctness
71882ff8c4 drivers: timer: nrf: drop unnecessary counter mask
4b24e88fa4 drivers: timer: nrf: use irq_lock instead of spinlock
```

If you're reverting your whole **tree** to that, then I'd guess the more likely culprit is the spinlockification series.  Note that most of it should be bisectable independently (though IIRC there's a header dependency that @carlescufi discovered that has to be manually fixed) so you can zero in on the specific subsystem at fault.  That series "shouldn't" have changed latencies at all, it's just refactoring.

pabigot, 2019-02-22:
@cvinayak @andyross Some of the wakeups are explained in 
https://github.com/zephyrproject-rtos/zephyr/issues/12542#issuecomment-455872935; in short even with default tickless the timeslicing infrastructure is default enabled.  Which is a mess because drivers like I2C are neither re-entrant nor threadsafe, so a timeslice that suspends a thread with a transfer in progress could introduce some very difficult to find.  A tickless system by default should have run-to-completion threads.  IMO.

Without knowing exactly what channel 0 measures I can't explain how you see 21 us lock durations, but I never saw anything greater than 2.88 us on an nRF51, nor did I see anything greater than 11 us before my changes.  Information about timing related to my changes to `nrf_rtc_timer` is in #12605.

andyross, 2019-02-22:
> even with default tickless the timeslicing infrastructure is default enabled

Timeslicing has always been "default y" in kconfig.  But it only affects threads that are at preemptible priorities, which aren't the default.  It did indeed get reworked along with the rest of the timing system during the churn last fall, but it's almost entirely orthogonal to tickless.

I'll admit I remain confused, though.  There really doesn't seem to be anything in that patch series that should impact latencies.  The locking strategy is basically the same.

andrewboie, 2019-02-22:
> Which is a mess because drivers like I2C are neither re-entrant nor threadsafe, so a timeslice that suspends a thread with a transfer in progress could introduce some very difficult to find misbehavior. 

If a driver isn't reentrant or thread-safe, that is a serious bug in the driver and not a problem in the kernel...what I2C drivers are like this?

andrewboie, 2019-02-22:
> A tickless system by default should have run-to-completion threads. IMO.

I'm confused here.
Tickless, and timeslicing really don't have much to do with each other, except of course if you enable timeslicing you're going to get more timer IRQs than if you didn't.

cvinayak, 2019-02-25:
@pabigot attached is the patch I used to profile the irq_locking in nrf_rtc_timer.c
```
From 44a8ede5526f011a9443aaef3aead53853ea5790 Mon Sep 17 00:00:00 2001
From: Vinayak Kariappa Chettimada <vich@nordicsemi.no>
Date: Mon, 25 Feb 2019 16:24:39 +0530
Subject: [PATCH] drivers: timer: profile the irq_locking

Profile IRQ locking in nrf_rtc_timer.

---
 drivers/timer/nrf_rtc_timer.c         | 2 ++
 samples/bluetooth/central_hr/prj.conf | 2 ++
 2 files changed, 4 insertions(+)

diff --git a/drivers/timer/nrf_rtc_timer.c b/drivers/timer/nrf_rtc_timer.c
index ef202da355..b27075191e 100644
--- a/drivers/timer/nrf_rtc_timer.c
+++ b/drivers/timer/nrf_rtc_timer.c
@@ -139,6 +139,7 @@ void z_clock_set_timeout(s32_t ticks, bool idle)
 	u32_t cyc = MIN((1 + ticks) * CYC_PER_TICK, MAX_DELAY);
 
 	u32_t key = irq_lock();
+	NRF_GPIO->OUTSET = BIT(12);
 	u32_t d = counter_sub(counter(), last_count);
 
 	/*
@@ -176,6 +177,7 @@ void z_clock_set_timeout(s32_t ticks, bool idle)
 	}
 	set_comparator(last_count + cyc);
 
+	NRF_GPIO->OUTCLR = BIT(12);
 	irq_unlock(key);
 #endif
 }
diff --git a/samples/bluetooth/central_hr/prj.conf b/samples/bluetooth/central_hr/prj.conf
index e937bf405d..54f6215497 100644
--- a/samples/bluetooth/central_hr/prj.conf
+++ b/samples/bluetooth/central_hr/prj.conf
@@ -3,3 +3,5 @@ CONFIG_BT_DEBUG_LOG=y
 CONFIG_BT_CENTRAL=y
 CONFIG_BT_SMP=y
 CONFIG_BT_GATT_CLIENT=y
+
+CONFIG_BT_CTLR_DEBUG_PINS=y
-- 
2.20.1

```

Also attached are the logic16 data files with the measurements.
[nrf_rtc_timer_profile.zip](https://github.com/zephyrproject-rtos/zephyr/files/2900239/nrf_rtc_timer_profile.zip)

pabigot, 2019-02-25:
I'm not sure who's working this (it isn't me; I'm not working Zephyr at this time), but I haven't seen progress and was curious so I spent a few hours looking at it.  Note that I can't replicate the environment: I don't have an iPhone, and I'm not seeing the `[bt][INF]` output.  So this is based on instrumentation alone.

IMO this instability is due to the requirement to align timeout deadlines to the 100 Hz system (tick) clock rather than use the native 32 KiHz RTC clock.  This alignment is not a cheap operation on the Cortex-M0.

The solution from #12605 was designed with the expectation that most timeouts were set through `z_clock_announce()`, so that the common case would be that `z_clock_set_timeout` would be invoked within one cycle of the last timer interrupt.  That is not the case with this application: the branch that is required to adjust the pre-calculated deadline is almost always entered, and the time it requires is apparently not always less than the previous solution as I had thought.  (There may be toolchain version impacts as well; I'm using `gcc-arm-none-eabi-8-2018-q4-major`.)

This is a problem because the Cortex-M0 needs to call `__udivsi3` to perform the adjustment because `CYC_PER_TICK` is 327.  This call/return takes at least 3 us.  If `SYS_CLOCK_TICKS_PER_SEC` is set to 128 it drops to 0.18 us because the division is a simple shift.  It would be interesting to see if that change makes the real application behave better.

The application also frequently sets six or more timeouts in less than 1 ms, all of which require recalculating the deadline.  I've seen cases where four calls to `z_clock_set_timeout()` occur within 148 us.  Of those, two were invocations with `ticks` set to zero, so there might be an opportunity for optimization there (though it's not trivial).  There's also some potential for pre-calculating a fallback adjustment based on the assumption `z_clock_set_timeout()` is not interrupted before it reaches the locked region.  In the general case, though, the only way I see to reliably fix the delay is to eliminate the costly alignment operation.

If changing ticks-per-second to ensure `CYC_PER_TICK` is a power of two is not an acceptable solution, it may be appropriate to revert 084363a0dc2a4d339a16c63c5d817400ae469605 in whole or in part to make this work, with the understanding that other instabilities will reappear and that #12541 must be re-opened.

andyross, 2019-02-25:
Is that consistent with this being a regression, though?  That division was in the old code too, and the clock settings were the same.

But the advice sounds sane to me: choosing the tick rate (which in tickless mode is an entirely arbitrary thing that just defines the units of timekeeping) to match the underlying clock for performance reasons seems like a good idea.

carlescufi, 2019-03-06:
@andyross @nashif as requested @cvinayak has added a "To Reproduce" section in the main description

carlescufi, 2019-03-06:
@cvinayak funnily I cannot reproduce the high numbers once connected, though I do see the assert if scanning without connection.

Scanning (after a while if not remote device found):
`[00:00:20.015,563] <err> bt_ctlr_llsw_ctrl: assert: '!radio_is_ready()' failed`

Connecting to another board running `peripheral_hr`:
`[00:00:42.721,038] <inf> bt_ctlr_hci: l: 6, 6, 39; t: 44, 43, 109.`

Connecting to iPhone:
`[00:00:01.056,060] <inf> bt_ctlr_hci: l: 6, 6, 25; t: 49, 39, 104.`

So worst case scenario I'm seeing is 39uS.

rljordan, 2019-03-07:
@SavinayDharmappa Please take a look at this.

andyross, 2019-03-07:
Let's at least start with a reversion per the comments above, as that seems lowest risk.  Can you guys try the linked patch to see if that resolve the issue as @cvinayak reported?  If so, then that seems simple.  If not we'll have to try something else.

Unfortunately I have neither a nrf51 board or iPhone (though I can probably scrounge up the former) so local reproduction is going to take some work.

pabigot, 2019-03-07:
FWIW I have a very hard time believing that the `irq_lock` around that section of nrf_rtc_timer could be held for 105 us.  That's an *extremely* long time; the longest operation in that sequence is an integer division that's less than 5 us, and the longest I saw the lock held was less than 20 us.  So I don't think reverting just the changes to that file will resolve the issue completely.

I would believe there are other `irq_lock` occurrences that could be that long.

andyross, 2019-03-07:
I don't disagree, but at this point we need to take what we can get.  @cvinayak can confirm, if it works we can merge and then fix it at our leisure.

andyross, 2019-03-11:
Recent discussion in PR #14176 seems to indicate that it addresses this issue.  For anyone here who might have missed that, please reveiw.

andyross, 2019-03-15:
Think we can close this one.  I opened #14577 to track the need for performance work in this ISR.  Please add notes there as needed.

closed 2019-03-15 by andyross

lopsided98, 2019-06-30:
I am running into this issue on Zephyr master (d045bd7673348db395e53b3d14c61d6742a25a06) in my own application on the `nrf51_ble400` board. This application is set up as a BLE peripheral and has quite a few features enabled, including BLE security and and custom sensor driver than uses a GPIO interrupt. I have not yet gotten a chance to try to create a minimal example that reproduces the problem. The crash only occurs when a device is connected.

This is the log just before the crash, with `CONFIG_BT_CTLR_PROFILE_ISR=y`. The latencies do not seem unreasonable compared to https://github.com/zephyrproject-rtos/zephyr/issues/13610#issuecomment-465949603.
```
[00:07:16.023,956] <inf> bt_ctlr_hci: l: 31, 7, 55; t: 74, 58, 111.
[00:07:16.033,355] <inf> bt_ctlr_hci: l: 7, 7, 55; t: 74, 58, 111.
[00:07:20.070,220] <inf> bt_ctlr_hci: l: 7, 7, 55; t: 80, 58, 111.
[00:07:20.083,526] <inf> bt_ctlr_hci: l: 7, 7, 55; t: 74, 58, 111.
[00:07:20.119,049] <inf> bt_ctlr_hci: l: 7, 7, 55; t: 103, 58, 111.
[00:07:20.133,361] <inf> bt_ctlr_hci: l: 7, 7, 55; t: 74, 58, 111.
[00:07:23.283,477] <inf> bt_ctlr_hci: l: 7, 7, 55; t: 95, 58, 111.
[00:07:23.287,780] <inf> bt_ctlr_hci: l: 8, 7, 55; t: 74, 58, 111.
[00:07:24.994,018] <inf> bt_ctlr_hci: l: 7, 7, 55; t: 61, 58, 111.
[00:07:25.033,386] <inf> bt_ctlr_hci: l: 7, 7, 55; t: 74, 58, 111.
[00:07:28.333,465] <inf> bt_ctlr_hci: l: 7, 7, 55; t: 95, 58, 111.
[00:07:28.357,818] <inf> bt_ctlr_hci: l: 7, 7, 55; t: 74, 58, 111.
[00:07:41.812,988] <inf> bt_ctlr_hci: l: 33, 7, 55; t: 74, 58, 111.
[00:07:41.833,343] <inf> bt_ctlr_hci: l: 7, 7, 55; t: 74, 58, 111.
[00:07:45.078,948] <err> bt_ctlr_llsw_ctrl: assert: '!radio_is_ready()' failed
***** HARD FAULT *****
***** Hardware exception *****
Current thread ID = 0x20000010
Faulting instruction address = 0x12c68
Fatal fault in ISR! Spinning...
```

cvinayak, 2019-07-01:
@lopsided98 do you have a master commit which was fine? could you check if v1.14-branch works fine for your application? I would try to bisect to find the commit beyond which the controller is seeing increased latencies. Is this reproducible in the vanilla samples/bluetooth/peripheral sample?

Thinking aloud, assuming the assert happened in an encrypted connection, the Radio ISR CPU time too seems to have increased, maybe changes in controller code too may be contributing to the increased overall CPU use. I should do some GPIO debug pin profiling to confirm the new numbers seem above.