zephyrproject-rtos/zephyr #13615: How can I configure the maximum and minimum connect event length when create a new le connection?

lylezhu2014, 2019-02-21:
I am trying to change the maximum and minimum connect event length. They cannot be set when calling the function bt_conn_create_le (max and min ce length are zero by default). 
I did not find a way to change the default value. Aslo there is not any description found related to these two values. Did I miss something? Is there a way to configure these two values when create a new le connection?

jhedberg, 2019-02-21:
I don't think you're missing anything. There's never been a need for these, so they haven't been exposed in the API. Note that these are informational parameters to the controller, i.e. they don't necessarily do anything at all, and if they do it's controller-dependent on how they're used. @cvinayak @carlescufi do you know if our controller does anything with this info?

cvinayak, 2019-07-16:
Just for the record, as related to the issue here, a related commit towards using connection event length: 32c2f850ba2b89449b4571a20b53fc288158b460

closed 2020-09-19 by nashif