zephyrproject-rtos/zephyr #13629: bluetooth: controller: allows to set random address during scanning

PavelVPV, 2019-02-21:
**Describe the bug**
When running the bluetooth mesh example (samples/bluetooth/mesh), the passive is scanning is enabled at the beginning. At some point the mesh example starts advertising by calling `bt_le_adv_start()`.
In the `bt_le_adv_start()` function, HCI Core tries to set random address. And the controller allows to do that even if it is in passive scanning. Here is the line of code that allows to set random address during passive scanning: https://github.com/zephyrproject-rtos/zephyr/blob/5e83e919e3805cd684b5bb610e1c85dc9aa42a7c/subsys/bluetooth/controller/ll_sw/ll_addr.c#L64 The issue is that by BT specification "Vol2, Part E, 7.8.4 LE Set Random Address Command" the command shall return an error.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -s sample/bluetooth/mesh -d sample/bluetooth/mesh/build -b nrf52_pca10040
2. west debug
3. Put a break point at subsys/bluetooth/controller/ll_sw/ll_addr.c:64
4. Check scanner state.

**Expected behavior**
Follow the BT Spec clause "Vol2, Part E, 7.8.4 LE Set Random Address Command": "If the Host issues this command when scanning or legacy advertising is enabled, the Controller shall return the error code Command Disallowed (0x0C) ".

cvinayak, 2019-03-01:
@jhe the relaxation in Zephyr was introduced in #5918. If I remember correct, this was Zephyr mesh stack requirement to be able to set random address to be able to start advertising while continuing to passive scan.

carlescufi, 2019-03-01:
Downgrading from bug to question because this is an intentional feature that was introduced willingly.

jhedberg, 2019-03-01:
@cvinayak yes, that was the reasoning I remember as well. If there is no clear reason why the core spec should be requiring this, I think it'd be good to raise an erratum for it.

carlescufi, 2019-03-05:
Closing this since it's been explained why the current controller acts like it does.

closed 2019-03-05 by carlescufi

jhedberg, 2019-03-05:
I was asked to provide some explanation why we want/need to do this for Bluetooth Mesh. Basically, the specification has gone through a lot of effort to ensure privacy of the traffic, i.e. who is sending, to whom, how many nodes in the network etc. Since mesh messaging is done using normal advertising special HW is not required for monitoring the traffic. However, if the bdaddr of the advertising packets is fixed or predictable, a lot of these security features are lost: you'll be able to make inferences not otherwise possible of the traffic in the mesh network. A simple way to protect against this is to ensure that every packet goes out with a new Non-Resolvable Private Address.

At the same time, it's essential that all Mesh nodes strive for as close as possible to 100% reception duty cycle, which means scanning as close as possible to 100% of the time, hence we do not want to ask the controller to stop scanning when sending out packets.

joerchan, 2019-03-06:
The only question that we need to agree on is how the passive scanner reports directed advertisement.
To make it work it seems to me that the random address that was set when the advertiser was started, is the one the scanner should try to match. And not the latest random address set.
Can we agree that that is the behaviour we want, and that that is the current implementation of the controller?

jhedberg, 2019-03-06:
I'll leave it up to the controller devs to answer the question from @joerchan (I don't really have any opinion about it), however I just wanted to point out one thing that came to mind: the apparent conflict with the HCI spec should go away once we move to vendor HCI commands, since with them the random address is explicitly bound to each advertising packet, since it's either given in the Mesh_Advertise HCI command, or it's left up to the controller to generate an NRPA for that advertising packet.

cvinayak, 2019-03-06:
@joerchan in the current controller, the scanner caches the init_addr at the time of the scanner being enabled and does not use latest random address being set by advertiser while the scanner is already running.

rugeGerritsen, 2019-03-07:
@cvinayak, if I understand you correctly, this means that the current controller will only report advertisements that are directed to the address used when enabling the scanner. This kind of behavior is not really intuitive to the users of the controller. Do we really want to settle on this behavior?

jhedberg, 2019-03-07:
@rugeGerritsen are you aware of any other use case besides Mesh that uses scanning & advertising in this manner? The way the mesh stack is implemented, it completely "owns" the advertising & scanning, i.e. doesn't allow any other usage those functions, and since directed advertising is not part of mesh the issue there becomes mostly academic.

rugeGerritsen, 2019-03-07:
I agree that you may treat this as an academic issue, per now I do not see any other use case.
My concern is that the behaviour is not intuitive to an application developer, and that it may change if the controller implementation changes.