zephyrproject-rtos/zephyr #13633: Execute native_posix using linux-arm SBC (e.g. raspi, beaglebone)

nikooiko, 2019-02-21:
**Is your feature request related to a problem?**
We are using boards that contain a host MCU that runs zephyr and also an uart radio peripheral for creating networks. For testing purposes, we often need to create large networks (> 100 devices) that are very expensive and hard to manage.

**Describe the solution you'd like**
A possible solution to do so is instead of using our production boards, we could use single board computers with multiple radios connected to their uarts and be able to execute multiple instances of the same app in one host CPU. This Linux setup will be easily managed and will allow us to reduce the cost of such tests.

Native posix abstraction layers seems to be able to do the job, but currently we are unable to compile/run such application using Arm-SBC. We are unable to use Zephyr SDK (because it doesn't support such devices) or host compilation tools (because the current native_posix compiler flags are built only for 64bit x86 archictures).

**Describe alternatives you've considered**
In our system, the zephyr os is just a middleware that provides an implementation of our Software Abstraction Layer. To solve the problem described above we could create another middleware that will allow us to drop everything zephyr-specific and enable us to run our apps on any platform.

**Additional context**
In the first image we can see the setup we are trying to test/emulate.
[image]

In the second image we can see how we can do it using a SBC.
[image]

aescolar, 2019-02-21:
Background: Today, upstream Zephyr only has the option to build native_posix on a x64 host just because that's what most people out there has as their development environments.
But I can tell you that, in our company, we are also compiling both native_posix and the nrf52_bsim targeting MIPS, and it works fine.
I don't think it would be too difficult to support building for ARM (either cross-compiling, or building "natively" in arm), just a few cmake changes. You are welcome to propose a patch in a PR.

Anyhow, depending on the radio you are using, and the purpose of your setup, you may want to consider [BabbleSim](https://babblesim.github.io), and be fully HW free (we are using it in Zephyr).

nikooiko, 2019-02-22:
Thanks for the feedback.
I'll look into it and I will return with the results (and possibly with a pull request).

aescolar, 2019-02-22:
@nikooiko : Question: I guess your radio is a BLE radio, and plan to expose it thru the Linux kernel to native_posix ? (not connecting directly via native_posix's UART)

nikooiko, 2019-02-22:
@aescolar : It's a proprietary 802.15.4 UART radio connected with an FTDI to the Linux SBC USB port and the Linux FS API is used to communicate with the radio.

nikooiko, 2019-02-25:
I've tracked the changes that must be done in order to be able to compile for raspberry pi 32bit linux (64bit need more work because pointer size assertions cause errors).

### Lack of multilib support (e.g. raspi)
#### Problem
The first problem is the `-m32` flag for the gcc, because the raspi images lack **GCC_MULTILIB** support:
https://github.com/zephyrproject-rtos/zephyr/blob/23ba884b2bb5af8801a4874804800cc93ece7930/cmake/compiler/host-gcc/target.cmake#L19-L23

https://github.com/zephyrproject-rtos/zephyr/blob/23ba884b2bb5af8801a4874804800cc93ece7930/arch/posix/CMakeLists.txt#L1-L3

https://github.com/zephyrproject-rtos/zephyr/blob/23ba884b2bb5af8801a4874804800cc93ece7930/arch/posix/CMakeLists.txt#L28-L32

#### Possible Solution
Add a new **CONFIG_NO_MULTILIB_SUPPORT** definition and use in a similar way as **CONFIG_X86_64** to disable -m32 flags

### Invalid Assembly Declarations
#### Problem
The second problem is that by default native_posix compilation results into x86 assembly which uses *@* symbol for function/object declarations:
https://github.com/zephyrproject-rtos/zephyr/blob/23ba884b2bb5af8801a4874804800cc93ece7930/include/toolchain/gcc.h#L243-L246

https://github.com/zephyrproject-rtos/zephyr/blob/23ba884b2bb5af8801a4874804800cc93ece7930/include/toolchain/gcc.h#L368-L373

#### Possible Solution
Add a new **CONFIG_ARM_POSIX** definition and use in a similar way as **CONFIG_ARM** to create the correct assembly declarations

aescolar, 2019-03-04:
@nikooiko : In my mind, both issues would seem to require a similar solution:
Differentiate on the POSIX ARCH underlying architecture (m32 being just a "please even that we are in x86-64, compile for x86 instead of the default" compiler switch which does not make sense on other than x86-64 environments).

mped-oticon, 2019-03-04:
@nikooiko It seems you are on the right track. I'll just add a few notes. 

Docker image of Debians of many different architectures:
https://github.com/mped-oticon/debian_foreign
You can simulate most/all effects of compling for raspberry pi, by adding arm as a target to debian_foreign.

Zephyr hack for MIPS:
https://github.com/mped-oticon/zephyr/commit/f11b3e1021958ef7e8c579fc2c5f669093a3985f

To avoid changes in CMake I wrote a gcc wrapper to add/subtract the flags I wanted:

```
#!/bin/bash
# Author: Mark Ruvald Pedersen (MPED)
# Description: This script wraps gcc, adapting command line options for Zephyr CMake

function gccwrap
{
    local num_params=$#
    local -a parms=( "$@" )

    for (( i=(($num_params - 1)); i>=0; i-- ))
    do
        local ith_arg="${parms[$i]}"
        # echo "$i : $ith_arg" 1>&2

        case "$ith_arg" in
            -m32)                               unset parms[$i] ;;
            -fno-asynchronous-unwind-tables)    unset parms[$i] ;;
            "-D_FORTIFY_SOURCE=2")              unset parms[$i] ;;
        esac
    done

    if [[ "$PWD" == *"/bsim/"* ]]; then
        prefix_params="-g -ggdb"
    fi

    if [[ "$PWD" == *"/zephyr/"* ]]; then
        prefix_params="-g -ggdb -static" # Statically compile + enable debug to permit GDB symbol resolution
    fi

    if [[ "$PWD" == *"/workspace/git/build/"* ]]; then
        prefix_params="-g -ggdb -static" # Statically compile + enable debug to permit GDB symbol resolution
    fi

    set -- "${prefix_params} ${parms[@]}"

    echo "(cd $PWD && /usr/bin/gcc.real $@)" 1>&2
    exec /usr/bin/gcc.real $@
}

gccwrap $@

```

aescolar, 2019-03-04:
I guess you could add an kconfig choice as part of POSIX_ARCH menu, something like `POSIX_ARCH_TARGET_{X86|ARM|MIPS}` which defaults to X86

aescolar, 2023-07-20:
AFAIR this has been working for a while now. So closing as done.