zephyrproject-rtos/zephyr #13637: Introduce supervisor-only stack declaration macros

andrewboie, 2019-02-21:
We currently assume that any given stack object may be used by a thread which can enter user mode. However, this assumption wastes memory for threads which always exist in supervisor mode.

Introduce some new stack declaration macros similar to K_THREAD_STACK_DEFINE which declare supervisor-only thread stacks:

- These stacks will not reserve memory for a privileged mode stack for syscalls, either by adding extra memory to the stack object (x86, ARC) or reservation of privileged stack buffer elsewhere (ARM)
- These stacks will be registered as a distinct kernel object type, such that if a thread with one of these stacks attempts to drop to user mode, or be created with the K_USER flag, will result in an error.
- These stacks will not have their sizes adjusted so that memory protection hardware regions can be set on them; for example there will be no rounding up to the nearest power of 2 (ARM) or page boundary (x86)
- Supervisor thread stack overflow guard regions will still be supported

ioannisg, 2019-06-11:
@andrewboie do you have plans to pick this up for 2.0 release?

andrewboie, 2019-06-11:
@ioannisg no I won't have the bandwidth, my time lately is all eaten up with code coverage issues

andrewboie, 2019-10-18:
This is on my plate for 2.1, will send a PR in the next week or so

ioannisg, 2019-10-19:
Thanks @andrewboie !

dleach02, 2019-12-10:
> This is on my plate for 2.1, will send a PR in the next week or so

@andrewboie did a PR get posted for this?

dleach02, 2019-12-10:
@ioannisg , since you have a tracking ticket that includes this I don't think we need to keep incrementing the milestone with each release

andrewboie, 2019-12-10:
@dleach02 it didn't make 2.1, I got preempted taking over some responsibilities for a colleague who left the company, but I'm looking at it this month.

bbolen, 2020-03-05:
@andrewboie Is there anything I could do to help move this forward?  I'm willing to go take the information in the description and work on a solution, or continue any work you already have started if it would help.

andrewboie, 2020-03-13:
Working on this now. It's a heavy lift for a few reasons but this is my primary focus for the next few weeks.

There's a related PR up already, see #23431

maksimmasalski, 2020-03-23:
@andrewboie Do you plan to develop test case for that macros?

andrewboie, 2020-04-26:
Draft PR submitted. See #24714

closed 2020-07-31 by nashif