zephyrproject-rtos/zephyr #13639: Use dirsync for doxygen directory syncing

carlescufi, 2019-02-21:
This [commit](https://github.com/zephyrproject-rtos/zephyr/commit/52842317505fc28354689011b649b3e3c9914362#diff-9c67a8cea4c574e6c88fda310f85c0e1) introduced a mechanism to preserve `mtime` in Doxygen-generated XML files in order to avoid having to process unchanged files every time we incrementally build the doc. More info can be found in [this thread](https://sourceforge.net/p/doxygen/mailman/doxygen-users/thread/CA%2BrkZbh9M80Suf%3DF9mohyxskcYGH%2BB04H8fvO3cz_yWY-RUgYA%40mail.gmail.com/#msg36580807).

The [dirsync](https://pypi.org/project/dirsync/) library seems up to the task, so potentially it could be used to replace much of the code in the script.

marc-hb, 2019-02-21:
Thanks @carlescufi for filing this. I'm the one to blame for the additional 100 lines of `restore_modification_times.py` and I agree it would be nice to get rid of most of them and replace them by a rsync-like python library as very well explained in the doxygen mailing-list thread above. 
Summarizing a private discussion with @carlescufi : the most critical requirement is obviously the ability NOT to copy/update the modification timestamp when it is _newer but the content has not changed_. This is more or less the `--checksum` option of `rsync`.
Also important is the equivalent of rsync's `--delete` option to remove renamed or deleted XML files.

As usual, the library replacement needs to be evaluated with respect to quality, maintainer responsiveness etc.

marc-hb, 2021-04-07:
> This commit introduced...

Ouch, I spent ages last week looking for this old issue and gave up eventually. Mentioning a commit links only one way. Mentioning PR #13159 here links both ways and would have let me find this issue :-)

gmarull, 2022-08-11:
doxyrunner replaced the mechanism described here, closing.

closed 2022-08-11 by gmarull