zephyrproject-rtos/zephyr #13651: ARC does not set thread->stack_info correctly

andrewboie, 2019-02-22:
**Describe the bug**
For any given thread, thread->stack_info.start / thread->stack_info.size represents the area within the stack that is read/writable by the user.

However, I've found that on ARC, this is not the case.

**To Reproduce**
Pull #13650 and run on nsim_em, observe crash.

**Expected behavior**
Test should pass, the user thread should have read/write access to this region in its own stack.

andrewboie, 2019-03-07:
Scope of this fix is for configurations that use a guard.
For configurations which set hardware registers to enforce stack bounds, we may need to revise how stack_analyze works.
I think we need a new nsim_* build target that exercises MPU stack overflow guards, I do not thing this is currently being tested.

nashif, 2020-03-11:
@vonhust  can you please take a look at this?

andrewboie, 2020-03-24:
This is still broken.

nashif, 2020-03-24:
@ruuddw  @abrodkin fyi

andrewboie, 2020-04-19:
I'm doing some work in this area and will take this one.

closed 2020-07-31 by nashif