zephyrproject-rtos/zephyr #13654: mimxrt1050_evk test/kernel/mem_protect/stack_random fails on stack fault

hakehuang, 2019-02-22:
**Describe the bug**
there are many fatal fault when execution on board mimxrt1050

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1050_evk ..
3. make
4. See error

**Expected behavior**
test case can return correct result

**Impact**
looks like the stack size setting for this case is not correct

**Screenshots or console output**
```
AASSRRIINNFFII  %]]@@%::dd***** USAGE FAULT *****

  Illegal load of EXC_RETURN into PC

***** Hardware exception *****

Current thread ID = 0x8000030c

Faulting instruction address = 0x72e003f6

Fatal fault in thread 0x8000030c! Aborting.

***** USAGE FAULT *****

  Illegal load of EXC_RETURN into PC

***** Hardware exception *****

Current thread ID = 0x8000030c

Faulting instruction address = 0x72e003f6

Fatal fault in thread 0x8000030c! Aborting.

***** USAGE FAULT *****

  Illegal load of EXC_RETURN into PC

***** Hardware exception *****

Current thread ID = 0x8000030c

Faulting instruction address = 0x72e003f6

Fatal fault in thread 0x8000030c! Aborting.
...
```

check attached file for more details
[serial_2019-02-21-23-42-27.txt](https://github.com/zephyrproject-rtos/zephyr/files/2892198/serial_2019-02-21-23-42-27.txt)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Version v1.14.0_rc1

hakehuang, 2019-02-22:
@MaureenHelm

agansari, 2019-02-22:
Tested on commit: c2d5e7b01ccff8be61ca139f578fdbce84b08713 

> ***** Booting Zephyr OS v1.14.0-rc1-444-gb1a7e71436 *****
> Running test suite stack_pointer_randomness
> ===================================================================
> starting test - test_stack_pt_randomization
> Test Stack pointer randomization
> stack pointer changed 55 times out of 64 tests
> PASS - test_stack_pt_randomization
> ===================================================================
> Test suite stack_pointer_randomness succeeded
> ===================================================================
> PROJECT EXECUTION SUCCESSFUL
> ***** Booting Zephyr OS v1.14.0-rc1-445-gc2d5e7b01c *****
> Running test suite stack_pointer_randomness
> ===================================================================
> starting test - test_stack_pt_randomization
> Test Stack pointer randomization
> stack pointer changed 58 times out of 64 tests
> PASS - test_stack_pt_randomization
> ===================================================================
> Test suite stack_pointer_randomness succeeded
> ===================================================================
> PROJECT EXECUTION SUCCESSFUL

Test passes.

dleach02, 2019-02-22:
not able to reproduce this either...

hakehuang, 2019-02-25:
I test RC1 based on https://github.com/zephyrproject-rtos/zephyr/commit/588d7b068c9166f043ccb0f9bf50bf39d0173cee, not c2d5e7b

hakehuang, 2019-02-25:
and if I run this case for 2 times, then Ican reproduce this issue, but I think this is not the only reproduce way.

ioannisg, 2019-02-25:
Unfortunately, I do not have hardware to test this on target.

hakehuang, 2019-02-27:
after download the application reboot the board can not reproduce this issue, looks like it is related my program method, as I am using the on board cmsis-dap to program a bootloader first. I will check my bootloader fisrt. Thanks. @dleach02 @galak @ioannisg  thanks for your checking

closed 2019-02-27 by hakehuang

hakehuang, 2019-02-28:
this is the same root cause as #13727