zephyrproject-rtos/zephyr #13655: mimxrt1050_evk test/crypto/rand32 meets Kernel Panic

hakehuang, 2019-02-22:
**Describe the bug**
kernel panic and code assert in entropy_mcux_trng.c:27

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1050_evk  ..
3. make
4. See error

**Expected behavior**
test runs OK

**Impact**
rng function can not work well

**Screenshots or console output**
```
***** Booting Zephyr OS 1.14.0-rc1 *****

Running test suite common_test

===================================================================

starting test - test_rand32

ASSERTION FAIL [!status] @ /build/src/workspace/rt1050_master-OI33AU5JUM5KTYHISBQ5NMALLT3XI63JEDNQBAUB7MQPDOVV6BBA@2/drivers/entropy/entropy_mcux_trng.c:27

***** Kernel Panic! *****

Current thread ID = 0x80000298

Faulting instruction address = 0x600035a6
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Version: v1.14.0_rc1

hakehuang, 2019-02-22:
@MaureenHelm

agansari, 2019-02-22:
Tested on commit: [c2d5e7b](https://github.com/zephyrproject-rtos/zephyr/commit/c2d5e7b01ccff8be61ca139f578fdbce84b08713)

```
***** Booting Zephyr OS v1.14.0-rc1-445-gc2d5e7b01c *****
Running test suite common_test
===================================================================
starting test - test_rand32
Generating random numbers
PASS - test_rand32
===================================================================
Test suite common_test succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

Test passes.

MaureenHelm, 2019-02-22:
@agansari can you rerun the test on RC1 to make sure you could reproduce the failure?

agansari, 2019-02-22:
@MaureenHelm I've tested this and #13654 on rc1, but I couldn't reproduce these issues.
@hakehuang can you show me a way to reproduce this issue? Maybe run multiple tests at once?

dleach02, 2019-02-22:
I'm not able to reproduce either

hakehuang, 2019-02-25:
I test RC1 based on [588d7b0](https://github.com/zephyrproject-rtos/zephyr/commit/588d7b068c9166f043ccb0f9bf50bf39d0173cee), not [c2d5e7b](https://github.com/zephyrproject-rtos/zephyr/commit/c2d5e7b01ccff8be61ca139f578fdbce84b08713)

hakehuang, 2019-02-25:
@agansari , yes I run it for multiply time then it will be reproduces

```
***** Booting Zephyr OS 1.14.0-rc1 *****

Running test suite common_test

===================================================================

starting test - test_rand32

ASSERTION FAIL [!status] @ /build/src/workspace/rt1050_master-OI33AU5JUM5KTYHISBQ5NMALLT3XI63JEDNQBAUB7MQPDOVV6BBA@2/drivers/entropy/entropy_mcux_trng.c:27

***** Kernel Panic! *****

Current thread ID = 0x80000278

Faulting instruction address = 0x600035a6

======== Target Main Serial Content:   ========
start application at 0x600062c9 !!!

***** Booting Zephyr OS 1.14.0-rc1 *****

Running test suite common_test

===================================================================

starting test - test_rand32

Generating random numbers

PASS - test_rand32

===================================================================

Test suite common_test succeeded

===================================================================

PROJECT EXECUTION SUCCESSFUL
```

agansari, 2019-02-26:
@hakehuang can't reproduce with 588d7b068c9166f043ccb0f9bf50bf39d0173cee
After compiling and using make flash board behaves okay, I've restarted (via button SW4) it at least 10 times for each test, but  test passes every time.

agansari, 2019-02-27:
Is this the same issue as https://github.com/zephyrproject-rtos/zephyr/issues/13654#issuecomment-467702805 ?

hakehuang, 2019-02-28:
@agansari  yes, close it

closed 2019-02-28 by hakehuang

hakehuang, 2019-03-07:
this error is because the FPRC of arm control register is set, then the stack is overflow as there are extra FP pushed. @agansari @nashif