zephyrproject-rtos/zephyr #13657: how to init uart3 in stm32_dev_min

fghjhuang, 2019-02-22:
### i learn the /sample and api refrense, but i didn't see how to init a uart port in codes, can anybody help me?
* i add->"CONFIG_UART_STM32_PORT_1=y" in prj.conf
* `
#include <uart.h>
`
* 
```c
void initUSART(void){ 
	struct uart_config config={
			.baudrate=115200
	};
	usart_dev = device_get_binding("UART1");
	uart_configure(usart_dev,&config); 
} 
void sendUartData(void){
	uart_poll_out(usart_dev,0xff);
}

### i init like above, but it turned wrong after i flash into stm32_min_dev board

explora26, 2019-02-22:
Create stm32_min_dev.overlay
```
&usart3 {
    status = "ok";
};
```
Add "CONFIG_UART_3=y" in prj.conf

fghjhuang, 2019-02-24:
> Create stm32_min_dev.overlay
> 
> ```
> &usart3 {
>     status = "ok";
> };
> ```
> Add "CONFIG_UART_3=y" in prj.conf

i had add this config, what i mean is that how to init receiver from uart3 & how to send data by uart3? can you help me?

explora26, 2019-02-24:
Try this, it's sould be work.
```
#include <zephyr.h>
#include <misc/printk.h>
#include <uart.h>

static struct device *uart_dev;

void main(void)
{
	u8_t c;
	struct uart_config config;

	uart_dev = device_get_binding("UART_3");

	uart_config_get(uart_dev, &config);

	config.baudrate = 115200;

	uart_configure(uart_dev, &config);

	while (true) {
		if (uart_poll_in(uart_dev, &c) == 0)
			uart_poll_out(uart_dev, c);
	}
}
```

fghjhuang, 2019-02-25:
> Try this, it's should be work.
> 
> ```
> #include <zephyr.h>
> #include <misc/printk.h>
> #include <uart.h>
> 
> static struct device *uart_dev;
> 
> void main(void)
> {
> 	u8_t c;
> 	struct uart_config config;
> 
> 	uart_dev = device_get_binding("UART_3");
> 
> 	uart_config_get(uart_dev, &config);
> 
> 	config.baudrate = 115200;
> 
> 	uart_configure(uart_dev, &config);
> 
> 	while (true) {
> 		if (uart_poll_in(uart_dev, &c) == 0)
> 			uart_poll_out(uart_dev, c);
> 	}
> }
> ```

### great! it does work,ths!

erwango, 2019-02-25:
@fghjhuang , great you managed to work it out. Please tell us when we can close this point.

closed 2019-02-27 by fghjhuang