zephyrproject-rtos/zephyr #13662: samples/subsys/usb/cdc_acm: Stuck at "Wait for DTR"

cinlyooi-intel, 2019-02-22:
**Describe the bug**
The last message is "Wait for DTR". Next message "DTR set, start test" not displayed.

Looks like the program is stuck at uart_line_ctrl_get() command

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=sam_e70_xplained -DARCH=arm  -D-DOVERLAY_CONFIG=overlay-composite-cdc-msc.conf   .. #sam_e70_xplained:arm
3. make
4. See error

Other targets:
-     -DBOARD=quark_se_c1000_devboard -DARCH=x86
-     -DBOARD=nrf52840_pca10056 -DARCH=arm 
-     -DBOARD=quark_se_c1000_devboard -DARCH=x86 

**Screenshots or console output**
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-444-gb1a7e71436 (delayed boot 1000ms) *****
Wait for DTR
```

**Environment (please complete the following information):**
 - OS: Fedora
 - Toolchain: Zephyr
 - Commit SHA or Version used: b1a7e71436e8fa74e26613e5d1d3bfd1214c27ae

**Additional context**

jfischer-phytec-iot, 2019-02-22:
Can not reproduce.

Tested on nrf52840_pca10056 and frdm_k64f

`cmake -DBOARD=nrf52840_pca10056 -DCONF_FILE="prj.conf overlay-composite-cdc-msc.conf" ../
`
```
[40761.348031] usb 9-1: new full-speed USB device number 6 using uhci_hcd                               
[40761.530652] usb 9-1: New USB device found, idVendor=2fe3, idProduct=0100, bcdDevice= 0.11            
[40761.530657] usb 9-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3                        
[40761.530660] usb 9-1: Product: Zephyr CDC ACM sample                                                  
[40761.530663] usb 9-1: Manufacturer: ZEPHYR
[40761.530666] usb 9-1: SerialNumber: ABCDEF012345
[40761.533702] cdc_acm 9-1:1.0: ttyACM1: USB ACM device                                                 
[40761.536051] usb-storage 9-1:1.2: USB Mass Storage device detected                                    
[40761.536711] scsi host9: usb-storage 9-1:1.2
[40762.559678] scsi 9:0:0:0: Direct-Access     ZEPHYR   ZEPHYR USB DISK  0.01 PQ: 0 ANSI: 0 CCS         
[40762.560304] sd 9:0:0:0: Attached scsi generic sg4 type 0                                             
[40762.565669] sd 9:0:0:0: [sdd] 32 512-byte logical blocks: (16.4 kB/16.0 KiB)                         
[40762.568674] sd 9:0:0:0: [sdd] Write Protect is off                                                   
[40762.568677] sd 9:0:0:0: [sdd] Mode Sense: 03 00 00 00                                                
[40762.571599] sd 9:0:0:0: [sdd] No Caching mode page found                                             
[40762.571605] sd 9:0:0:0: [sdd] Assuming drive cache: write through                                    
[40762.599676]  sdd:
[40762.618616] sd 9:0:0:0: [sdd] Attached SCSI removable disk     
```          

```
***** Booting Zephyr OS v1.14.0-rc1-445-gc2d5e7b01c *****
Wait for DTR
DTR set, start test
Baudrate detected: 115200

```

qianfan-Zhao, 2019-02-24:
'DTR set' are printed after usb plugin, did you forget this?

cinlyooi-intel, 2019-03-22:
@qianfan-Zhao After talking to @finikorg and looking at your comment, I realized that I try this sample wrongly.

Sorry for the trouble I had caused and thank you all for your patience

closed 2019-03-22 by cinlyooi-intel

NancyChen0526, 2022-09-05:
worked on nRF52840-DK