zephyrproject-rtos/zephyr #13664: samples/subsys/usb/cdc_acm_composite: No output beyond "***** Booting Zephyr OS v1.14.0-rc- ....****** banner

cinlyooi-intel, 2019-02-22:
**Describe the bug**
The console only outputs the "Delay Boot" and "Booting Zephyr" banners three Times.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=quark_se_c1000_ss_devboard -DARCH=arc
3. make
4. See error

Other configuration:
1. -DBOARD=frdm_k64f -DARCH=arm

**Screenshots or console output**
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-444-gb1a7e71436 (delayed boot 1000ms) *****
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-444-gb1a7e71436 (delayed boot 1000ms) *****
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-444-gb1a7e71436 (delayed boot 1000ms) *****
```

**Environment (please complete the following information):**
 - OS: Fedora29
 - Toolchain: zephyr
 - Commit SHA or Version used : -DBOARD=frdm_k64f -DARCH=arm

**Additional context**
I put a printf("START") command at the start of main() but failed to see it when running it.
Then I replaced the whole main.c file with the said command but still failed to see it.
Based on this, I do not think zephyr had booted.

jfischer-phytec-iot, 2019-02-22:
what is the difference to #13662?

cinlyooi-intel, 2019-02-22:
Different test case. This one has two devices to connect to. I was not 100% sure it is the same as #13662 so filed it separately

jfischer-phytec-iot, 2019-02-22:
>cmake -DBOARD=quark_se_c1000_ss_devboard -DARCH=arc

Does this architecture have usb? :angry:

cinlyooi-intel, 2019-02-22:
@jfischer-phytec-iot  OOOUCH! According to quark_se_c1000_ss_devboard.yaml, "no usb_device" ;-(

Very very sorry. Please pardon this new boy who still have to learn his rope.

closed 2019-02-22 by jfischer-phytec-iot