zephyrproject-rtos/zephyr #13665: amples/subsys/usb/cdc_acm_composite: Stuck at "Wait for DTR"

cinlyooi-intel, 2019-02-22:
**Describe the bug**
The last message is "Wait for DTR". Next message "DTR set, start test" not displayed.

Looks like the program is stuck at uart_line_ctrl_get() command

**To Reproduce**
Steps to reproduce the behavior:

1. mkdir build; cd build
2. cmake -DBOARD=sam_e70_xplained -DARCH=arm  -D-DOVERLAY_CONFIG=overlay-composite-cdc-msc.conf   .. #sam_e70_xplained:arm
3. make
4. See error

Other targets:

* ```
  -DBOARD=nrf52840_pca10056 -DARCH=arm 
  ```
* ```
  -DBOARD=frdm_k64f -DARCH=arm
  ```

**Screenshots or console output**

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-444-gb1a7e71436 (delayed boot 1000ms) *****
Wait for DTR
```
**Environment (please complete the following information):**

* OS: Fedora
* Toolchain: Zephyr
* Commit SHA or Version used: [b1a7e71](https://github.com/zephyrproject-rtos/zephyr/commit/b1a7e71436e8fa74e26613e5d1d3bfd1214c27ae)

cinlyooi-intel, 2019-02-22:
Most likely we will find it is the same reason as https://github.com/zephyrproject-rtos/zephyr/issues/13662

jfischer-phytec-iot, 2019-02-22:
what is the difference to #13662?

nashif, 2019-02-22:
duplicate

closed 2019-02-22 by nashif